esphome/issues #6393: Grove Multichannel Gas Sensor V2  NO2 sensor not recognized failed to compile

orbital253, 2024-11-01:
### The problem

I got Grove Multichannel Gas Sensor V2 and i used the example code

```
sensor:
  - platform: grove_gas_mc_v2
    no2:
      id: precal_no2
      name: "Nitrogen Dioxide"
    ethanol:
      name: "Ethanol"
    carbon_monoxide:
      name: "Carbon Monoxide"
    tvoc:
      name: "Volatile Organic Compounds"
```

but when i try to compile it i get the error

```
INFO ESPHome 2024.10.2
INFO Reading configuration /config/esphome/airquality.yaml...
Failed config

sensor.grove_gas_mc_v2: [source /config/esphome/airquality.yaml:57]
  platform: grove_gas_mc_v2
  
  [no2] is an invalid option for [sensor.grove_gas_mc_v2]. Please check the indentation.
  no2: 
    id: precal_no2
    name: Nitrogen Dioxide
  ethanol: 
    id: precal_ethanol
    name: Grove Ethanol
  carbon_monoxide: 
    id: precal_co
    name: Grove Carbon Monoxide
  tvoc: 
    id: precal_tvoc
    name: Grove Volatile Organic Compounds
```

i have another i2c device with the same indentation that works ok. when i comment out the no2 code compiles correctly

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.4

### What platform are you using?

ESP32

### Board

Lillygo T7

### Component causing the issue

Grove Multichannel Gas Sensor V2

### Example YAML snippet

```yaml
sensor:
  - platform: grove_gas_mc_v2
    no2:
      id: precal_no2
      name: "Nitrogen Dioxide"
    ethanol:
      id: precal_ethanol
      name: "Grove Ethanol"
    carbon_monoxide:
      id: precal_co
      name: "Grove Carbon Monoxide"
    tvoc:
      id: precal_tvoc
      name: "Grove Volatile Organic Compounds"
```

ppussar, 2024-12-03:
This configuration works for me:

`sensor:
  - platform: grove_gas_mc_v2
    nitrogen_dioxide:
      name: "Nitrogen Dioxide"
    ethanol:
      name: "Ethanol"
    carbon_monoxide:
      name: "Carbon Monoxide"
    tvoc:
      name: "Volatile Organic Compounds"`

The documentation seems to be outdated.

orbital253, 2024-12-19:
@ppussar You are correct the documentation is outdated. it works ok with your settings

closed 2025-01-07 by nagyrobi