esphome/issues #6394: Ultrasonic A02YYUW errors when no object in front of sensor

ntompson, 2024-11-02:
### The problem

## My use case

I am adding ultrasonic distance sensors to my automated swing gate controller to detect cars in the path of the gate, allowing me to prevent the gate opening or closing to avoid expensive mistakes. I've chosen the DFRobot waterproof ultrasonic sensor ([A02YYUW](https://esphome.io/components/sensor/a02yyuw)], which works very nicely.

## The issue

Everything works perfectly well when there is an object in front of the sensor within its range. I see distance measurements in the ESPHome log, and the HA sensor reports distance accurately. However, if there is no object in front of the sensor (no car in the way of the the gate), rather than update the sensor or report a maximum distance, the log fills with error messages (`Invalid data read from sensor`) - see ESPHome log excerpt below.

## Problems created by this issue

This causes two problems:

* When there is nothing in front of the sensor, I get error messages instead of a measurement, meaning that I need a workaround to determine that driveway is clear (I have one using filters, see below)
* At ESP boot, if there is no car in the way of the sensor (which is rather common), the sensor never initialises with a value, so my workaround fails (until something moves in front of the sensor)

## Possible solutions

It would be nice if the component allowed one to set a default value to report when the sensor cannot find an object. One could set the default to a value greater than the sensor's measurement range, for example.

## Workaround

As noted above, I've implemented a workaround, using a timeout filter (see config below). This solves the first problem, but I can't find a way to solve the second problem.

I'm open to suggestions for alternative workarounds. One requirement I have is that this can be solved in ESPHome (without the help of automations implemented in Home Assistant). The solution needs to work if my HA server is down, or the power is out (the gate controller has battery backup).

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.4

### What platform are you using?

ESP32

### Board

  ESP-WROOM-32 Development Board

### Component causing the issue

A02YYUW

### Example YAML snippet

```yaml
logger:
  level: DEBUG
  logs:
    a02yyuw.sensor: ERROR

sensor:
  - platform: "a02yyuw"
    name: "Distance"
    uart_id: uart_1
    filters:
      - throttle_average: 0.5s
      - timeout:
          timeout: 1.5s
          value: 5000
```

### Anything in the logs that might be useful for us?

```txt
[17:00:20][D][sensor:094]: 'Distance': Sending state 1152.00000 mm with 0 decimals of accuracy
[17:00:21][D][sensor:094]: 'Distance': Sending state 1195.66663 mm with 0 decimals of accuracy
[17:00:21][D][sensor:094]: 'Distance': Sending state 1230.80005 mm with 0 decimals of accuracy
[17:00:22][W][a02yyuw.sensor:032]: Invalid data read from sensor: FF.00.00.FF
[17:00:22][D][sensor:094]: 'Distance': Sending state 1244.25000 mm with 0 decimals of accuracy
[17:00:22][W][a02yyuw.sensor:032]: Invalid data read from sensor: FF.00.00.FF
[17:00:22][W][a02yyuw.sensor:032]: Invalid data read from sensor: FF.00.00.FF
[17:00:22][W][a02yyuw.sensor:032]: Invalid data read from sensor: FF.00.00.FF
[17:00:22][W][a02yyuw.sensor:032]: Invalid data read from sensor: FF.00.00.FF
```

phillipAB47, 2025-01-16:
I was having the same issue as you were.  I changed to a JSN-SRO4T sensor and that solved my 'no object' problem.  Hope you got it all worked out.

[image]

ntompson, 2025-01-19:
That's a good reminder that I never came back to this. In the end, I modified the source code for the A02YYUW sensor in ESPHome so that it would do what I want. If I knew how to use GitHub, I'd submit it as a PR. For now, I'm just using the modified code as an [`external_component`](https://esphome.io/components/external_components.html).

I've changed `a02yyuw.cpp` as below. There are really only three changes:

* When the sensor measures zero, report `MAX_DISTANCE` as the distance. I've configured this to 4.5 m, which is the highest the sensor can read. While we can't be sure whether zero is actually zero or a distance greater than 4.5 m, in my application, there is no possibility it is zero
* Although the sensor specification says that distances closer than 0.03 m are unreliable, I don't care. So I report all measured non-zero distances
* Prime the sensor with `MAX_DISTANCE` as the last good distance - so at power on, if there is nothing in front of the sensor, it will report `MAX_DISTANCE` until something comes in range of the sensor

With these changes in place, I've found these sensors to be extremely reliable - my system is working very well.

```cpp
// Datasheet https://wiki.dfrobot.com/_A02YYUW_Waterproof_Ultrasonic_Sensor_SKU_SEN0311

#include "a02yyuw.h"
#include "esphome/core/helpers.h"
#include "esphome/core/log.h"

namespace esphome {
namespace my_a02yyuw {

static const char *const TAG = "a02yyuw.sensor";
static const int MAX_DISTANCE = 4500;
static int last_good = MAX_DISTANCE;

void A02yyuwComponent::loop() {
  uint8_t data;
  while (this->available() > 0) {
    this->read_byte(&data);
    if (this->buffer_.empty() && (data != 0xff)) {
      // ESP_LOGV(TAG, "Sync");
      continue;
    }
    buffer_.push_back(data);
    if (this->buffer_.size() == 4)
      this->check_buffer_();
  }
}

void A02yyuwComponent::check_buffer_() {
  uint8_t checksum = this->buffer_[0] + this->buffer_[1] + this->buffer_[2];
  if (this->buffer_[3] == checksum) {
    float distance = (this->buffer_[1] << 8) + this->buffer_[2];
    if (distance > 0) {
      ESP_LOGV(TAG, "Distance from sensor: %f mm", distance);
      this->publish_state(distance);
      last_good = distance;
    } else {
      float zero_distance = MAX_DISTANCE;
      this->publish_state(zero_distance);
    }
  } else {
    ESP_LOGV(TAG, "checksum failed: %02x != %02x", checksum, this->buffer_[3]);
  }
  this->buffer_.clear();
}

void A02yyuwComponent::dump_config() {
  LOG_SENSOR("", "A02yyuw Sensor", this); 
}

}  // namespace a02yyuw
}  // namespace esphome
```

ntompson, 2025-01-19:
I should add: @TH-Braemer - if you're interested / willing, it would be great to see these changes go into the main code base.

TH-Braemer, 2025-01-24:
Sorry for being so slow...
Of course I am interested, but it's been a while since i messed with this sensor, and I don't have a spare for testing (it just works in my application).
I do though need you to explain one thing. You say you don't care about the value below 0.03m. Can you tell, what your solution reports in that case - I hope that it's not MAX-DISTANCE.
In my application it's very important in case it fails to read (for any reason), that it reports the last measured value, not something else, otherwise my statistics gets blown...
If you can confirm that's the case, We can get it into a PR, if I can remember how ;)

Grisly00, 2025-03-28:
Hi, 
I just stumbled upon this issue as I am struggling with the ultrasonic sensor myself.  It would be great if the no value state gets handled in the future, but in my opinion it should clearly return NaN (or something like -999) and not just MAX_DISTANCE or MIN_DISTANCE to be more generic and cover all use cases. 
The rest can then be handled in the yaml configuration.

ntompson, 2025-03-30:
A sensible solution (which I have to confess is beyond my capabilities) would be to expose a couple of options in the config for the sensor.

* By default, the sensor behaves as it does today - that is - throw errors if the sensor measures out of range. That way, no existing deployments get broken
* Add a `min_distance` option. If set, sensor reports this value if the sensor measures below `min_distance`
* Add a `max_distance` option. Similarly, report `max_distance` if the sensor measures over `max_distance`
* If both are set, will requiring inferencing based on last good measurement to decide whether to report `min_distance` or `max_distance` (since an out of range measurement is reported by the sensor as 0).
* Add a `report_last_good` option. If set to true, report the last in range measurement whenever an out of range measurement is made. Could be complementary with `min_distance` and `max_distance`, so that the last good option is only used in the case `min_distance` and / or `max_distance` is not set

Probably needs some messing about to figure out what works. Also need to figure out how to "prime" the sensor - at power up, the sensor will report 0 if there is no obstacle in range. In my use case, this could be hours.

Grisly00, 2025-03-31:
Yes, this sounds like a good approach, although I do not know if it has to be this elaborate. If we simply add a NaN return the rest of the logic could also be handled from the  yaml config with something like 
`if (isnan(id(distance_sensor).state)) {...}`. At the moment though, invalid values cannot  be handled at all, as far as I understand.