esphome/issues #6395: Component NAU7802 blocks the operation of other components

fran6120, 2024-11-02:
### The problem

If for some reason the NAU7802 component is marked as failed during setup, the rest of the components will either not be initialized or simply not function correctly.

This even affects the WiFi component, never connecting to the configured network.

By changing the setup priority of the NAU7802 component to -100 to ensure that it is the last component to be configured, the wifi component works correctly.

It can also be verified by changing the setup priority of the NAU7802 component to a very specific value, and adding a looger.log message to the priority boot sequence to see where it gets stuck.

I have verified that the error occurs with both the Arduino framework and ESP-IDF

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32c3

### Component causing the issue

NAU7802

### Example YAML snippet

```yaml
esphome:
  name: esp32c3
  friendly_name: esp32c3

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  level: VERY_VERBOSE
  setup_priority: 1000

# Enable Home Assistant API
api:
  encryption:
    key: "2AQVS5IW+jI1/nKrzG9v+2w/3SCSAeG0WwRazsVCszE="

ota:
  - platform: esphome
    password: "0fa23ee9bab49609b3626ef20cba7778"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32C3 Fallback Hotspot"
    password: "5QY6yOGYiWoU"

captive_portal:

i2c:
[... 950 lines cut ...]
[12:48:46][VV][i2c.arduino:176]: 0x71 TX 
[12:48:46][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[12:48:46][VV][i2c.arduino:176]: 0x72 TX 
[12:48:46][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[12:48:46][VV][i2c.arduino:176]: 0x73 TX 
[12:48:46][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[12:48:46][VV][i2c.arduino:176]: 0x74 TX 
[12:48:46][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[12:48:46][VV][i2c.arduino:176]: 0x75 TX 
[12:48:46][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[12:48:46][VV][i2c.arduino:176]: 0x76 TX 
[12:48:46][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[12:48:46][VV][i2c.arduino:176]: 0x77 TX 
[12:48:46][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[12:48:46][VV][scheduler:063]: set_interval(name='', interval=60000, offset=20565)
[12:48:46][I][Pririty:189]: priority 435 success
[12:48:46][C][nau7802:056]: Setting up NAU7802 'NAU7802 Value'...
[12:48:46][VV][i2c.arduino:176]: 0x2A TX 1F
[12:48:46][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[12:48:46][E][nau7802:060]: Failed I2C read during setup()
[12:48:46][E][component:119]: Component nau7802.sensor was marked as failed.
[12:48:46][E][component:164]: Component nau7802.sensor set Error flag: unspecified
[12:48:46][VV][scheduler:063]: set_interval(name='update', interval=60000, offset=3890)
[12:48:46][VV][scheduler:226]: Running interval '' with interval=60000 last_execution=4294887025 (now=295)
```

fran6120, 2024-11-04:
@cujomalainey 

Any ideas?

Until now I was using this component, as an external component, with the code from the first version, before it was integrated into esphome, and there were no problems when starting the esp32 without the sensor connected.

Seeing that the sensor has been fully integrated, I updated to the latest version replacing the external component containing the old code, with the native component, and as you can see in the logs I lost the wifi connection.

Anyway, thank you very much for bringing the NAU7802 component to esphome.

cujomalainey, 2024-11-04:
Try deleting this line

https://github.com/esphome/esphome/blob/dev/esphome/components/nau7802/nau7802.cpp#L320

ssieb, 2024-11-04:
That will fix this issue, but I have some other concerns about the component.  I can finish the PR I started or we can discuss it on discord if you want.

fran6120, 2024-11-04:
It works fine now. It was also necessary to remove `bool can proceed() override;` from `nau7802.h`
```

[21:02:42][I][logger:156]: Log initialized
[21:02:42][V][esp32.preferences:059]: nvs_get_blob('233825507'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[21:02:42][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[21:02:42]0;38m[VV][i2c.arduino:202]: TX failed: not acknowledged: 2ESP-ROM:esp32c3-api1-20210207
[21:02:42]Build:Feb  7 2021
[21:02:42]rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
[21:02:42]Saved PC:0x4038bd68
[21:02:42]SPIWP:0xee
[21:02:42]mode:DIO, clock div:1
[21:02:42]load:0x3fcd6100,len:0x438
[21:02:42]load:0x403ce000,len:0x918
[21:02:42]load:0x403d0000,len:0x24e4
[21:02:42]entry 0x403ce000
[21:02:42][I][logger:156]: Log initialized
[21:02:42][VV][esp32.preferences:071]: nvs_get_blob: key: 233825507, len: 4
[21:02:42][C][safe_mode:079]: There have been 1 suspected unsuccessful boot attempts
[21:02:42][VV][esp32.preferences:040]: s_pending_save: key: 233825507, len: 4
[21:02:42][D][esp32.preferences:114]: Saving 1 preferences to flash...
[21:02:42][VV][esp32.preferences:123]: Checking if NVS data 233825507 has changed
[21:02:42][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[21:02:42][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[21:02:42][I][app:029]: Running through setup()...
[21:02:42][V][app:030]: Sorting components by setup priority...
[21:02:42][I][i2c.arduino:218]: Performing I2C bus recovery
[21:02:42][   182][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=7 scl=6 freq=100000
[21:02:42][V][i2c.arduino:042]: Scanning i2c bus for active devices...
[21:02:42][VV][i2c.arduino:176]: 0x08 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x09 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x0A TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x0B TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x0C TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x0D TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x0E TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x0F TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x10 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x11 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x12 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x13 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x14 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x15 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x16 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x17 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x18 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x19 TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x1A TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x1B TX 
[21:02:42][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[21:02:42][VV][i2c.arduino:176]: 0x1C TX 
[... 281 lines cut ...]
[21:02:51][C][captive_portal:089]: Captive Portal:
[21:02:51][C][mdns:116]: mDNS:
[21:02:51][C][mdns:117]:   Hostname: esp32c3
[21:02:51][V][mdns:118]:   Services:
[21:02:51][V][mdns:120]:   - _esphomelib, _tcp, 6053
[21:02:51][V][mdns:122]:     TXT: friendly_name = esp32c3
[21:02:51][V][mdns:122]:     TXT: version = 2024.10.2
[21:02:51][V][mdns:122]:     TXT: mac = 70041d770864
[21:02:51][V][mdns:122]:     TXT: platform = ESP32
[21:02:51][V][mdns:122]:     TXT: board = esp32-c3-devkitm-1
[21:02:51][V][mdns:122]:     TXT: network = wifi
[21:02:51][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[21:02:51][C][esphome.ota:073]: Over-The-Air updates:
[21:02:51][C][esphome.ota:074]:   Address: esp32c3.local:3232
[21:02:51][C][esphome.ota:075]:   Version: 2
[21:02:51][C][esphome.ota:078]:   Password configured
[21:02:51][C][safe_mode:018]: Safe Mode:
[21:02:51][C][safe_mode:020]:   Boot considered successful after 60 seconds
[21:02:51][C][safe_mode:021]:   Invoke after 10 boot attempts
[21:02:51][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[21:02:51][C][api:140]: API Server:
[21:02:51][C][api:141]:   Address: esp32c3.local:6053
[21:02:51][C][api:143]:   Using noise encryption: YES
[21:02:58][D][api:103]: Accepted 192.168.1.182
```

cujomalainey, 2024-11-04:
> That will fix this issue, but I have some other concerns about the component. I can finish the PR I started or we can discuss it on discord if you want.

please send a PR for me to review, thanks

cujomalainey, 2025-03-06:
Will try and test my pr this weekend and submit for review

cujomalainey, 2025-03-25:
PR is tested and ready fyi, just needs review/merge

closed 2025-04-02 by ssieb