esphome/issues #6396: Error uploading to devices in 2024.10.2

uspass, 2024-11-02:
### The problem

With HomeAssistant core 2024.10.4 and ESPHome version 2024.10.2 uploading ESP32 S3 doesn't work. It looks like it uploads but flashing actually doesn't happen. The ESPHome version stays at 2024.10.0. 
More, if serial flashing a new module using manual download (ESP32 WROOM 06), the module doesn't do anything, like getting online.

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.4

### What platform are you using?

ESP32

### Board

ESP32 WROOM 06

### Example YAML snippet

```yaml
esphome:
  name: ${device_name}
  friendly_name: ${switch_name}
  on_boot:
    priority: -100
    then:
      switch.turn_on: switch_1

esp32:
  board: esp32-c3-devkitm-1

binary_sensor:
  - platform: gpio
    id: binary_sensor_1
    pin:
      number: GPIO4
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1
  - platform: status
    name: Status

button:
  - platform: restart
    name: Restart

light:
  - platform: monochromatic
    id: light_1
    gamma_correct: 1.0
    output: output_blue_led

output:
  - platform: ledc
    id: output_blue_led
    pin: 
      number: GPIO6

sensor:
  - platform: wifi_signal
[... 577 lines cut ...]
[18:47:32][C][safe_mode:021]: Invoke after 10 boot attempts
[18:47:32][C][safe_mode:023]: Remain in safe mode for 300 seconds
[18:47:32][C][api:140]: API Server:
[18:47:32][C][api:141]: Address: 192.168.10.180:6053
[18:47:32][C][api:143]: Using noise encryption: YES
[18:47:32][C][wifi_signal.sensor:010]: WiFi Signal 'Wifi Signal Strength'
[18:47:32][C][wifi_signal.sensor:010]: Device Class: 'signal_strength'
[18:47:32][C][wifi_signal.sensor:010]: State Class: 'measurement'
[18:47:32][C][wifi_signal.sensor:010]: Unit of Measurement: 'dBm'
[18:47:32][C][wifi_signal.sensor:010]: Accuracy Decimals: 0
[18:47:32][C][debug:021]: Debug component:
[18:47:32][D][debug:035]: ESPHome version 2024.10.0
[18:47:32][D][debug:039]: Free Heap Size: 183688 bytes
[18:47:32][D][debug:162]: Flash Chip: Size=4096kB Speed=80MHz Mode=QIO
[18:47:32][D][debug:210]: Chip: Model=ESP32-C3, Features=WIFI_BGN,BLE, Cores=1, Revision=4
[18:47:32][D][debug:218]: ESP-IDF Version: v4.4.2
[18:47:32][D][debug:223]: EFuse MAC: 84:F7:03:6A:FF:00
[18:47:32][D][debug:129]: Reset Reason: Timer Group 0 Watch Dog Reset Digital Core
[18:47:32][D][debug:271]: Wakeup Reason: Unknown
[18:47:32][D][debug:129]: Reset Reason: Timer Group 0 Watch Dog Reset Digital Core
[18:47:32][D][text_sensor:064]: 'Reset Reason': Sending state 'Timer Group 0 Watch Dog Reset Digital Core'
[18:47:46][D][sensor:094]: 'Uptime': Sending state 223.96899 s with 0 decimals of accuracy
[18:47:59][D][sensor:094]: 'Wifi Signal Strength': Sending state -42.00000 dBm with 0 decimals of accuracy

Of course, the "Update" button doesn't dissapear.

jesserockz, 2024-11-10:
Please post the logs from the device itself while you are doing the ota upload.

ssieb, 2024-11-11:
What file did you use for the serial flashing?

uspass, 2024-11-13:
It looks like version 2024.10.3 fixed the problem.
[03:08:26][I][app:100]: ESPHome version 2024.10.3 compiled on Nov 13 2024, 03:05:21

closed 2024-11-13 by uspass