esphome/issues #6399: X9C on ESP32 C3 Supermini changes only in one direction

Apfelsafft, 2024-11-03:
### The problem

Hi,
I am trying to connect a X9C103 board to the supermini. 
The problem is that the poti moves only in one direction. No matter if I increase or decrease the value, the amount is only running into one direction.
I tried it on two X9C breakout boards and on different pins of the C3 (GPIO2, 3 & 4 or GPIO10, 20, 21 or GPIO 10, 8, 9)
When conneecting the boards to a ESP32-S3 it works fine. 

Also tried to increase step_delay from 1us to 10us to 100us. 

No matter what, it seems as if UP/DN information is not send on a C3 supermini.

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.10.4 OS 13.2

### What platform are you using?

ESP32

### Board

ESP32-C3 Supermini

### Component causing the issue

X9C

### Example YAML snippet

```yaml
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

output:
  - platform: x9c
    id: swc1
    cs_pin: GPIO10
    inc_pin: GPIO20
    ud_pin: GPIO21
    initial_value: 0.01
    step_delay: 1us 

switch:
  - platform: template
    name: "Function 1"
    id: fc1
    optimistic: true
    on_turn_on:
    - output.set_level:
       id: swc1
       level: 0.9  
    - delay: 1000ms
    - output.set_level:
       id: swc1
       level: 0.01    
    - switch.turn_off: fc1
```

### Additional information

I am using a X9C103 breakout board. It ist the greenisch rectangular one, where you have the control pins on one side and the resistor pins on the other.
To measure the resistance, I connected RL to GND and the wiper to RH to work as a rheostat. Measuring between wiper and GND.

Apfelsafft, 2025-03-16:
Hello,
this bug still applies. 
The U/D pin is not working, the pot value keeps decreasing. No matter what you select or pin you choose.

Any help is appreciated.

Apfelsafft, 2025-03-16:
I did some more tests.
Tried also a ESP32 C3 Zero board, which didn't work.
Tried to toggle the U/D Pin via switch.toggle (without using x9c), which worked.
Tried another S3 board which also worked.

Can't really figure out why the X9C component doesn't work on the C3.
The pin is always kept high (3v3) - is there a chance that the declared INT for ChangeAmount could be interpreted as an unsigned INT?

Apfelsafft, 2025-03-22:
It seems the issue is that there is an cast to UINT32 which removes the negative sign when compiling for ESP C3 where it is ok when compiling for an ESP S3.
The code was changed from cast to INT to UINT32 in addition to roundf() function. Switching back to INT while keeping roundf() works well for me on a C3 now.

Not sure how to patch or request a bugfix for this.

AnderVentura, 2025-05-20:
This bug still applies. Same code works well with other ESP32 boards

Apfelsafft, 2025-05-27:
If you switch back to cast to an INT instead of UINT32 it works. At least for me :)
You can override the standard code if you put the files on your local system.

AnderVentura, 2025-05-27:
> If you switch back to cast to an INT instead of UINT32 it works. At least for me :) You can override the standard code if you put the files on your local system.

That's what I've done following your previous post. Maybe in the future I'll try to push this fix to ESPHome main branch.

Virtualchronos, 2025-11-16:
I got the same problem with 2025 version ans esp32C3 super mini... Can somebody explain me how to fix it ?

Apfelsafft, 2025-11-20:
Hi.
In my HA instance I added a folder /esphome/custom_components/x9c and copied the files from the repositiory in there.
Then in the x9c.cpp you should change the static_cast<uint32> to static_cast<int>. I think it was 3 lines or so.

Esphome should use the local files before the original ones.
However, I ensured this by changing the names of the folder and files to like x9cr.h and .cpp (incl. the references in that files) and used it as a new custom platform.
But I guess just editing the casts to INT should be sufficient.