esphome/issues #6401: Max and min temperature limits not only for show

ThomasNygards, 2024-11-04:
### The problem

In the help, the following is written for [Climate Component](https://esphome.io/components/climate/#config-climate))

**visual** (_Optional_): Visual settings for the climate device - these do not affect operation and are solely for controlling how the climate device shows up in the frontend.

**min_temperature** (_Optional_, float): The minimum temperature the climate device can reach. Used to set the range of the frontend gauge.

**max_temperature** (_Optional_, float): The maximum temperature the climate device can reach. Used to set the range of the frontend gauge.

This gives the impression that max and min temperatures should not affect the limits of the controller,
but in fact they do.
With this configuration 

```
climate:
  - platform: pid
    id: stigarror_controller
    name: "Stigarrör PID Controller"
    sensor: utgaende_element_och_golv
    default_target_temperature: 30°C
    heat_output: shunt_percentage
    control_parameters:
      kp: 0.3460
      ki: 0.000187
      kd: 0
      output_averaging_samples: 5      # smooth the output over 3 samples
      derivative_averaging_samples: 5  # smooth the derivative value over 3 samples
    deadband_parameters:
      threshold_high: 0.5°C       # deadband within +/-0.5°C of target_temperature
      threshold_low: -0.5°C`
```

I get
**Error: Provided temperature 30.4 is not valid. Accepted range is 10.0 to 30.0**

If I change the config to

```
climate:
  - platform: pid
    id: stigarror_controller
    name: "Stigarrör PID Controller"
    sensor: utgaende_element_och_golv
    default_target_temperature: 30°C
    heat_output: shunt_percentage
    visual:
      min_temperature: 10
      max_temperature: 45
    control_parameters:
      kp: 0.3460
      ki: 0.000187
      kd: 0
      output_averaging_samples: 5      # smooth the output over 3 samples
      derivative_averaging_samples: 5  # smooth the derivative value over 3 samples
    deadband_parameters:
      threshold_high: 0.5°C       # deadband within +/-0.5°C of target_temperature
      threshold_low: -0.5°C
```

then all works fine. I.e. I added:
```
    visual:
      min_temperature: 10
      max_temperature: 45
```

My automation is:

```
id: '1696702239366'
alias: Update pid target temperature
description: 'updates pid thermostat based on stigarrör måltemp '
trigger:
  - platform: state
    entity_id:
      - input_number.stigarror_maltemp
condition: []
action:
  - data:
      temperature: '{{ states(''input_number.stigarror_maltemp'')| float(0) }}'
    target:
      entity_id: climate.stigarror_pid_controller
    action: climate.set_temperature
mode: single
```

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.2

### What platform are you using?

ESP32

ssieb, 2024-11-05:
That's HA.  How are you setting the temperature?  It's not the esphome climate component giving that message.

ThomasNygards, 2024-11-05:
> That's HA. How are you setting the temperature? It's not the esphome climate component giving that message.

Ok. anyhow, this is the action in the automation that sets the temperature:

```
action:
  - data:
      temperature: '{{ states(''input_number.stigarror_maltemp'')| float(0) }}'
    target:
      entity_id: climate.stigarror_pid_controller
    action: climate.set_temperature
```

ssieb, 2024-11-05:
Why are you doing that?  But regardless, it's not an esphome issue.  If you want HA to accept the temperature, you have to set the visual range to accommodate it.

closed 2024-11-05 by ssieb

ThomasNygards, 2024-11-05:
Why? Well, that is complicated to explain. I would have to explain how my control scheme is implemented. Basically, my PID controller controls the output temperature of a shunt. I do not control against the house temperature, as this is harder to make stable due to how my home heating system is built up.

Anyhow. This is what I am objecting in with this defect. I find it misleading as it does affect operation!
**Visual settings for the climate device - these do not affect operation and are solely for controlling how the climate device shows up in the frontend.**

ssieb, 2024-11-05:
It is absolutely correct.  It only affects the frontend and it's the frontend that's causing your problem.  It is not the esphome component that is rejecting the temperature change.  It does not affect the operation of the component.  If you bypass HA, you can set the temperature to whatever you want.  But how do you expect that to show up in the interface if you're setting values outside the range?  If you aren't actually using the climate interface in HA, then why does it matter what you set for the visual range.  Just set it to whatever you need.

ThomasNygards, 2024-11-05:
Thanks for your prompt reply, @ssieb !