esphome/issues #6404: web.esphome.io can't connect to pico w after firmware install

julesham57a, 2024-11-06:
### The problem

I have uploaded the esphome firmware to some of my Pico Ws. They had MicroPython on them before. I don't know if this is the issue of if this only works on new picos that have never had firmware on them before?

### Which version of ESPHome has the issue?

The latest firmware from the web.esphome.io page

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

RP2040

### Board

Pico W V1

### Component causing the issue

web.esphome.io

julesham57a, 2024-11-06:
Additionally, I have tried Windows, Ubuntu, Chrome and Edge and multiple Picos. I don't have any brand new ones. but I know all the ones I have work with MicroPython on them as I have put it back on some of them.

billmaterial, 2024-11-21:
@julesham57a 

I have many issues with the PIco W and I will not be using them any more with ESPHome unfortunately.

1. They sometimes won't reconnect to Wifi after OTA flashing them, especially if you have a TPLink device!
2. They have issues with SPI and not flashing at all over OTA (my problem is with a display) see issue #4070.

You can use the "flash_nuke.uf2" to erase the Pico W back to factory but I don't think the problem is they have been used with MicroPython before.

Moar-Beer, 2025-02-25:
Are you trying to connect to an open network with no password? I just confirmed that Pico W will not connect to an open network using esphome.

closed 2026-03-12 by bdraco