esphome/issues #6405: [LVGL] inverted colors in Waveshare e-paper display

regystro, 2024-11-06:
### The problem

The 2 possible colors supported by ESPHome Display component (black / white = pixel on / pixel off) seem to be inverted in a 2.70inv2 Waveshare e-paper display,

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

lvgl

### Example YAML snippet

```yaml
image:
  - file: mdi:alert-outline
    id: alert_icon
    resize: 80x80

display:
  - platform: waveshare_epaper
    id: my_display
    cs_pin: GPIO19
    dc_pin: GPIO17
    busy_pin: GPIO16
    reset_pin: GPIO4
    model: 2.70inv2
    auto_clear_enabled: false
    update_interval: never
    rotation: 90

lvgl:
  buffer_size: 25% # recommended
  widgets:
    - obj:
        height: 100%
        width: 100%
        widgets:
        - image:
            align: CENTER
            src: alert_icon
```

### Additional information

This is the result of the above code snippet: 
[image]

If instead of LVGL you use native ESPHome display support with this code:
```
display:
  - platform: waveshare_epaper
    id: my_display
    cs_pin: GPIO19
    dc_pin: GPIO17
    busy_pin: GPIO16
    reset_pin: GPIO4
    model: 2.70inv2
    auto_clear_enabled: false
    update_interval: never
    rotation: 270
    lambda: |-
          it.image(it.get_width()/2, it.get_height()/2, id(alert_icon), ImageAlign::CENTER);

```
this is what you correctly get:
[image]

Also notice that to achieve the same screen orientation, one must use rotation 90 in LVGL and rotation 270 in native ESPHome. I don't know if that's intentional or should I open another issue about that.

clydebarrow, 2024-11-13:
> Also notice that to achieve the same screen orientation, one must use rotation 90 in LVGL and rotation 270 in native ESPHome.

That should be no longer the case with the latest beta version. LVGL rotation was being done in the LVGL library, which treats a rotation as being of the entire display, not the image. Now it's done in the ESPHome component, and it should match the sense of rotation for the standard display.

clydebarrow, 2024-11-14:
The e-paper displays draw images with inverted colours - black (color_off) is drawn as white and anything _other_ than black is drawn as black. This means white and any colour shade other than absolute black are drawn as black.

The regular display component fills the background with color_off (black) which does appear black on an LCD but becomes white on e-paper. LVGL however defaults to a background which is a shade of white, which on e-paper becomes black. So that explains the black background when you were expecting white. To get around that you need to specify a _black_ background (just like the `display` component default.)

The image is another matter; since it defaults to a binary format (one bit per pixel) LVGL treats it as an alpha-only image and blends another colour to generate the result. The default for that is black, which on e-paper becomes white!

So to replicate the way in which `display::image()` draws, you need to specify `image_recolor: white` - you can do this for each image, or globally in a theme. And you can define some colour names that make life a little less confusing. So your lvgl config would look like this:

```
color:
  - id: paper_white # Really black, but will appear white on e-paper
    hex: 000000
  - id: paper_black # Really white, but will appear black on e-paper
    hex: FFFFFF

lvgl:
  theme:
    image:
      image_recolor: paper_black
  bg_color: paper_white
  widgets:
    - image:
        align: center
        src: alert_icon
```

closed 2024-11-14 by clydebarrow

regystro, 2024-11-15:
Wow, thanks!

I managed to get it working but had to change a couple of things in your proposed code:

1. `hex: 000000` needs to be `hex: "000000"`, otherwise compiling fails throwing "Hex color must have six digits."
2. you must add `image_recolor_opa: 100%` to the image style properties or else it won't work

I found number 2 here: https://devices.esphome.io/devices/Elecrow-5inch-ESP32-display, googling about _esphome image_recolor_.
So, let me suggest to add `image_recolor` and `image_recolor_opa` to the [image widget documentation](https://esphome.io/components/lvgl/widgets#image).

I can also confirm that just as you say in your https://github.com/esphome/issues/issues/6405#issuecomment-2474979740, rotation is fixed in 2024.11.0b2

Thanks again and greetings.