esphome/issues #6409: Sensor value precision (device_class: timestamp)

muxa, 2024-11-07:
### The problem

I have a sensor which I want to return a timestamp so that Home Assistant display a nice readable duration remaining to that timestamp (e.g. 5 minutes, 30 seconds, etc.).

However, when the sensor updates, it seems to round the sensor value to 128 seconds blocks.
 
Is there a way to fix this to have full timestamp resolution?

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.10.4

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Example YAML snippet

```yaml
- platform: template
    name: Now
    id: now1
    device_class: timestamp
    lambda: |-
      auto now = id(current_time).now();
      if (now.is_valid()) {
        ESP_LOGD("time", "Now: %d", now.timestamp);
        return now.timestamp;
      }
      return {};
    update_interval: 5s

time:
  - platform: homeassistant
    id: current_time
```

### Anything in the logs that might be useful for us?

```txt
[21:39:12][D][time:181]: Now: 1730968752
[21:39:12][D][sensor:093]: 'Now': Sending state 1730968704.00000  with 0 decimals of accuracy
[21:39:17][D][time:181]: Now: 1730968757
[21:39:17][D][sensor:093]: 'Now': Sending state 1730968704.00000  with 0 decimals of accuracy
[21:39:22][D][time:181]: Now: 1730968762
[21:39:22][D][sensor:093]: 'Now': Sending state 1730968704.00000  with 0 decimals of accuracy
[21:39:27][D][time:181]: Now: 1730968767
[21:39:27][D][sensor:093]: 'Now': Sending state 1730968704.00000  with 0 decimals of accuracy
[21:39:28][D][sensor:093]: 'Uptime': Sending state 0.00130 d with 1 decimals of accuracy
[21:39:32][D][time:181]: Now: 1730968772
[21:39:32][D][sensor:093]: 'Now': Sending state 1730968832.00000  with 0 decimals of accuracy
```

ssieb, 2024-11-07:
A sensor is a `float` value, so there is limited precision.  Not enough for a timestamp.

muxa, 2024-11-08:
Is to possible to overcome this limit with a `test_sensor` setting `device_class: timestamp? I've tried it, but the value is always `unknown` in HA.

ssieb, 2024-11-08:
If it's a `sensor:`, then the underlying storage is a `float`.