esphome/issues #6415: XIAO ESP32S3 Sense - "[Errno 104] Connection reset by peer" when using the Camera module

hwilma, 2024-11-08:
### The problem

The connection to my ESP32-S3 is constantly interrupted. This only happens when the camera module (XIAO ESP32-S3 Sense) is configured. In the frontend I was only able to view pictures of the cam only a few times.
When the camera component is not configured, the node works properly.
I checked the node with another wifi accesspoint and the error also occurs.
I also tested the following and nothing changed:
  - disabling of mDNS in the ESPHome Addon
  - configuring a fixed accesspoint to connect to over BSSID

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.11.0

### What platform are you using?

ESP32

### Board

XIAO ESP32-S3 Sense 

### Component causing the issue

XIAO camera module for "Sense Kit"

### Example YAML snippet

```yaml
esphome:
  name: cameranode01
  friendly_name: cameranode01
  
  # PlatformIO build options
  platformio_options:
    build_flags: -DBOARD_HAS_PSRAM
    board_build.arduino.memory_type: qio_opi
    board_build.f_flash: 80000000L
    board_build.flash_mode: qio

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "---"

ota:
  - platform: esphome
    password: "---"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.6.42.97
    gateway: 10.6.42.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Cameranode01 Fallback Hotspot"
    password: "---"

captive_portal:

# Configuration for the ESP32 Camera
esp32_camera:
  id: espcam
  name: Xiao Cam
  external_clock:
    pin: GPIO10
    frequency: 20MHz
  i2c_pins:
    sda: GPIO40
    scl: GPIO39
  data_pins: [GPIO15, GPIO17, GPIO18, GPIO16, GPIO14, GPIO12, GPIO11, GPIO48]
  vsync_pin: GPIO38
  href_pin: GPIO47
  pixel_clock_pin: GPIO13
  resolution: 800x600
  
# Configuration for the ESP32 Camera Web Server
esp32_camera_web_server:
  - port: 8080
    mode: stream
  - port: 8081
    mode: snapshot
```

### Anything in the logs that might be useful for us?

```txt
INFO Successfully connected to cameranode01 @ 10.6.42.97 in 0.005s
INFO Successful handshake with cameranode01 @ 10.6.42.97 in 0.061s
[22:08:11][I][app:100]: ESPHome version 2024.10.3 compiled on Nov  8 2024, 21:36:56
[22:08:11][C][wifi:600]: WiFi:
[22:08:11][C][wifi:428]:   Local MAC: 24:EC:4A:00:56:B0
[22:08:11][C][wifi:433]:   SSID: 'Router42'[redacted]
[22:08:11][C][wifi:436]:   IP Address: 10.6.42.97
[22:08:11][C][wifi:440]:   BSSID: FE:92:BF:C4:F5:F1[redacted]
[22:08:11][C][wifi:441]:   Hostname: 'cameranode01'
[22:08:11][C][wifi:443]:   Signal strength: -50 dB ▂▄▆█
[22:08:11][C][wifi:447]:   Channel: 11
[22:08:11][C][wifi:448]:   Subnet: 255.255.255.0
[22:08:11][C][wifi:449]:   Gateway: 10.6.42.1
[22:08:11][C][wifi:450]:   DNS1: 0.0.0.0
[22:08:11][C][wifi:451]:   DNS2: 0.0.0.0
[22:08:11][C][logger:185]: Logger:
[22:08:11][C][logger:186]:   Level: DEBUG
[22:08:11][C][logger:188]:   Log Baud Rate: 115200
[22:08:11][C][logger:189]:   Hardware UART: USB_CDC
[22:08:11][C][esp32_camera:048]: ESP32 Camera:
[22:08:11][C][esp32_camera:049]:   Name: Xiao Cam
[22:08:11][C][captive_portal:089]: Captive Portal:
[22:08:13][D][esp32_camera:196]: Got Image: len=25014
WARNING cameranode01 @ 10.6.42.97: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for cameranode01 @ 10.6.42.97
WARNING Disconnected from API
INFO Successfully connected to cameranode01 @ 10.6.42.97 in 0.649s
INFO Successful handshake with cameranode01 @ 10.6.42.97 in 0.155s
[22:08:56][D][esp32_camera:196]: Got Image: len=24959
WARNING cameranode01 @ 10.6.42.97: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for cameranode01 @ 10.6.42.97
WARNING Disconnected from API
INFO Successfully connected to cameranode01 @ 10.6.42.97 in 0.004s
INFO Successful handshake with cameranode01 @ 10.6.42.97 in 1.085s
[22:09:42][D][esp32_camera:196]: Got Image: len=24613
WARNING cameranode01 @ 10.6.42.97: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for cameranode01 @ 10.6.42.97
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for cameranode01 @ 10.6.42.97: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.6.42.97', port=6053))]: [Errno 111] Connect call failed ('10.6.42.97', 6053) (SocketAPIError)
INFO Trying to connect to cameranode01 @ 10.6.42.97 in the background
```

ssieb, 2024-11-09:
Can you get serial logs?  It doesn't look like it's crashing.

hwilma, 2024-11-09:
The serial Log shows a problem at connecting to my wifi. The problem doesn't exist when i comment the camera-code. Maybe there is a timing problem, at startup with the camera initialization.

INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM0 with baud rate 115200
[20:35:06][I][logger:156]: Log initialized
[20:35:06][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[20:35:06][D][esp32.preferences:114]: Saving 1 preferences to flash...
[20:35:06][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[20:35:06][I][app:029]: Running through setup()...
[20:35:07][C][wifi:048]: Setting up WiFi...
[20:35:07][C][wifi:061]: Starting WiFi...
[20:35:07][C][wifi:062]:   Local MAC: --MAC--
[20:35:07][D][wifi:482]: Starting scan...
[20:35:07][W][component:157]: Component wifi set Warning flag: scanning for networks
[20:35:12][D][wifi:497]: Found networks:
...
[20:35:12][I][wifi:313]: WiFi Connecting to 'Router_WIFI'...
[20:35:16][D][esp32_camera:196]: Got Image: len=25368
[20:35:16][W][wifi_esp32:537][arduino_events]: Event: Disconnected ssid='Router_WIFI' bssid=--BSSID--:12:32:33[redacted] reason='4-Way Handshake Timeout'
[20:35:16][W][wifi:674]: WiFi Unknown connection status 0
[20:35:17]ESP-ROM:esp32s3-20210327
[20:35:17]Build:Mar 27 2021
[20:35:17]rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
[20:35:17]Saved PC:0x420a790a
WARNING Decoded 0x420a790a: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[20:35:17]SPIWP:0xee
[20:35:17]mode:DIO, clock div:1
[20:35:17]load:0x3fce3808,len:0x43c
[20:35:17]load:0x403c9700,len:0xbec
[20:35:17]load:0x403cc700,len:0x2a3c
[20:35:17]entry 0x403c98d8
[20:35:17][I][logger:156]: Log initialized
[20:35:17][C][safe_mode:079]: There have been 1 suspected unsuccessful boot attempts
[20:35:17][D][esp32.preferences:114]: Saving 1 preferences to flash...
[20:35:17][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[20:35:17][I][app:029]: Running through setup()...
[20:35:17][C][wifi:048]: Setting up WiFi...
[20:35:17][C][wifi:061]: Starting WiFi...
[20:35:17][C][wifi:062]:   Local MAC: --MAC--
[20:35:17][D][wifi:482]: Starting scan...
[20:35:17][W][component:157]: Component wifi set Warning flag: scanning for networks
[20:35:23][D][wifi:497]: Found networks:
...

hwilma, 2024-11-10:
after reducing the wifi output power to 8.5dB the connection can be established. The boot loop is still there

[22:10:29][C][wifi:048]: Setting up WiFi...
[22:10:29][C][wifi:061]: Starting WiFi...
[22:10:29][C][wifi:062]:   Local MAC: 24:EC:4A:00:56:B0
[22:10:29][D][wifi:482]: Starting scan...
[22:10:29][W][component:157]: Component wifi set Warning flag: scanning for networks
[22:10:35][D][wifi:497]: Found networks:
[22:10:35][I][wifi:541]: - 'Router42' [redacted]▂▄▆█
[22:10:35][D][wifi:542]:     Channel: 6
[22:10:35][D][wifi:543]:     RSSI: -61 dB
[22:10:35][I][wifi:541]: - 'Router42' [redacted]▂▄▆█
[22:10:35][D][wifi:542]:     Channel: 1
[22:10:35][D][wifi:543]:     RSSI: -80 dB
[22:10:35][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[22:10:35][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[22:10:35][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[22:10:35][D][wifi:546]: - [redacted] [redacted]▂\xe2\x96
[22:10:35][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[22:10:35][D][wifi:546]: - [redacted] [redacted][D][wifi:546]: - [redacted] [redacted][D][wifi:546]: - [redacted] [redacted]\033[D][wifi:546]: - [redacted] [redacted][I][wifi:313]: WiFi Connecting to 'Router42'...
[22:10:36][I][wifi:617]: WiFi Connected!
[22:10:36][C][wifi:428]:   Local MAC: --mac--
[22:10:36][C][wifi:433]:   SSID: [redacted]
[22:10:37][C][wifi:436]:   IP Address: 10.6.42.97
[22:10:37][C][wifi:440]:   BSSID: [redacted]
[22:10:37][C][wifi:441]:   Hostname: 'cameranode01'
[22:10:37][C][wifi:443]:   Signal strength: -60 dB ▂▄▆█
[22:10:37][C][wifi:447]:   Channel: 6
[22:10:37][C][wifi:448]:   Subnet: 255.255.255.0
[22:10:37][C][wifi:449]:   Gateway: 10.6.42.1
[22:10:37][C][wifi:450]:   DNS1: 0.0.0.0
[22:10:37][C][wifi:451]:   DNS2: 0.0.0.0
[22:10:37][D][wifi:626]: Disabling AP...
[22:10:37][C][api:026]: Setting up Home Assistant API server...
[22:10:37][I][app:062]: setup() finished successfully!
[22:10:37][W][component:170]: Component wifi cleared Warning flag
[22:10:37][W][component:157]: Component api set Warning flag: unspecified
[22:10:37][I][app:100]: ESPHome version 2024.10.3 compiled on Nov 10 2024, 22:08:46
[22:10:37][C][wifi:600]: WiFi:
[22:10:37][C][wifi:428]:   Local MAC: --mac--
[22:10:37][C][wifi:433]:   SSID: [redacted]
[22:10:37][C][wifi:436]:   IP Address: 10.6.42.97
[22:10:37][C][wifi:440]:   BSSID: [redacted]
[22:10:37][C][wifi:441]:   Hostname: 'cameranode01'
[22:10:37][C][wifi:443]:   Signal strength: -60 dB ▂▄▆█
[22:10:37][C][wifi:447]:   Channel: 6
[22:10:37][C][wifi:448]:   Subnet: 255.255.255.0
[22:10:37][C][wifi:449]:   Gateway: 10.6.42.1
[22:10:37][C][wifi:450]:   DNS1: 0.0.0.0
[22:10:37][C][wifi:451]:   DNS2: 0.0.0.0
[22:10:37][C][logger:185]: Logger:
[22:10:37][C][logger:186]:   Level: DEBUG
[22:10:37][C][logger:188]:   Log Baud Rate: 115200
[22:10:37][C][logger:189]:   Hardware UART: USB_CDC
[22:10:37][C][ledc.output:180]: LEDC Output:
[22:10:37][C][ledc.output:181]:   Pin GPIO21
[22:10:37][C][ledc.output:182]:   LEDC Channel: 2
[22:10:37][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[22:10:37][C][ledc.output:184]:   Phase angle: 0.0°
[22:10:37][C][ledc.output:185]:   Bit depth: 14
[22:10:37][C][light:103]: Light 'LED onboard'
[22:10:37][C][light:105]:   Default Transition Length: 1.0s
[22:10:37][C][light:106]:   Gamma Correct: 2.80
[22:10:37][C][esp32_camera:048]: ESP32 Camera:
[22:10:37][C][esp32_camera:049]:   Name: Xiao Cam
[22:10:37][C][esp32_camera:050]:   Internal: NO
[22:10:37][C][esp32_camera:052]:   Data Pins: D0:15 D1:17 D2:18 D3:16 D4:14 D5:12 D6:11 D7:48
[22:10:37][C][esp32_camera:053]:   VSYNC Pin: 38
[22:10:37][C][esp32_camera:054]:   HREF Pin: 47
[22:10:37][C][esp32_camera:055]:   Pixel Clock Pin: 13
[... 37 lines cut ...]
[22:10:37][C][esp32_camera_web_server:088]: ESP32 Camera Web Server:
[22:10:37][C][esp32_camera_web_server:089]:   Port: 8080
[22:10:37][C][esp32_camera_web_server:091]:   Mode: stream
[22:10:37][C][esp32_camera_web_server:088]: ESP32 Camera Web Server:
[22:10:37][C][esp32_camera_web_server:089]:   Port: 8081
[22:10:37][C][esp32_camera_web_server:093]:   Mode: snapshot
[22:10:39][D][esp32_camera:196]: Got Image: len=21842
[22:10:39]ESP-ROM:esp32s3-20210327
[22:10:39]Build:Mar 27 2021
[22:10:39]rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
[22:10:39]Saved PC:0x420ae116
WARNING Decoded 0x420ae116: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[22:10:39]SPIWP:0xee
[22:10:39]mode:DIO, clock div:1
[22:10:39]load:0x3fce3808,len:0x43c
[22:10:39]load:0x403c9700,len:0xbec
[22:10:39]load:0x403cc700,len:0x2a3c
[22:10:39]entry 0x403c98d8
[22:10:40][I][logger:156]: Log initialized
[22:10:40][C][safe_mode:079]: There have been 6 suspected unsuccessful boot attempts
[22:10:40][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:10:40][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:10:40][I][app:029]: Running through setup()...
[22:10:40][C][wifi:048]: Setting up WiFi...
[22:10:40][C][wifi:061]: Starting WiFi...

ssieb, 2024-11-11:
It seems to be crashing in an idf function.  Do you have a USB uart adapter?

hwilma, 2024-11-11:
I connected my USB uart adapter and the following message was written into the serial log:

INFO ESPHome 2024.10.3
INFO Reading configuration /config/esphome/cameranode01.yaml...
INFO Starting log output from /dev/ttyUSB1 with baud rate 115200
[23:02:45]ESP-ROM:esp32s3-20210327
[23:02:45]Build:Mar 27 2021
[23:02:45]rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
[23:02:45]SPIWP:0xee
[23:02:45]mode:DIO, clock div:1
[23:02:45]load:0x3fce3808,len:0x43c
[23:02:45]load:0x403c9700,len:0xbec
[23:02:45]load:0x403cc700,len:0x2a3c
[23:02:45]entry 0x403c98d8

# here the error occurs

[23:02:56]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
[23:02:56]Debug exception reason: Stack canary watchpoint triggered (framebuffer_tas) 
[23:02:56]Core  1 register dump:
[23:02:56]PC      : 0x40380a77  PS      : 0x00060036  A0      : 0x8037f488  A1      : 0x3fce5700  
WARNING Decoded 0x40380a77: compare_and_set_native at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:288
[23:02:56]A2      : 0x3fc97610  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060023  
[23:02:56]A6      : 0x00060023  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff  
[23:02:56]A10     : 0x3fc975e8  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x3fce57d0  
[23:02:56]A14     : 0x02c97610  A15     : 0x00ffffff  SAR     : 0x00000000  EXCCAUSE: 0x00000001  
[23:02:56]EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
[23:02:56]
[23:02:56]
[23:02:56]Backtrace:0x40380a74:0x3fce57000x4037f485:0x3fce5740 0x4037d9e8:0x3fce5770 0x4037d9de:0xa5a5a5a5  |<-CORRUPTED
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40380a74: compare_and_set_native at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:288
WARNING Decoded 0x4037f485: vPortEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/include/freertos/portmacro.h:578
 (inlined by) vTaskSwitchContext at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3458
WARNING Decoded 0x4037d9e8: _frxt_dispatch at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:436
WARNING Decoded 0x4037d9de: _frxt_int_exit at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:231
[23:02:56]
[23:02:56]
[23:02:56]
[23:02:56]
[23:02:56]ELF file SHA256: 0000000000000000
[23:02:56]
[23:02:56]Rebooting...
[23:02:56]ESP-ROM:esp32s3-20210327
[23:02:56]Build:Mar 27 2021
[23:02:56]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[23:02:56]Saved PC:0x420ae116
WARNING Decoded 0x420ae116: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[23:02:56]SPIWP:0xee
[23:02:56]mode:DIO, clock div:1
[23:02:56]load:0x3fce3808,len:0x43c
[23:02:56]load:0x403c9700,len:0xbec
[23:02:56]load:0x403cc700,len:0x2a3c
[23:02:56]entry 0x403c98d8

# second error after boot

[23:03:06]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
[23:03:06]Debug exception reason: Stack canary watchpoint triggered (framebuffer_tas) 
[23:03:06]Core  1 register dump:
[23:03:06]PC      : 0x40380a77  PS      : 0x00060036  A0      : 0x8037f488  A1      : 0x3fce5700  
WARNING Decoded 0x40380a77: compare_and_set_native at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:288
[23:03:06]A2      : 0x3fc97610  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060023  
[23:03:06]A6      : 0x00060023  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff  
[... 7 lines cut ...]
WARNING Decoded 0x40380a74: compare_and_set_native at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:288
WARNING Decoded 0x4037f485: vPortEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/include/freertos/portmacro.h:578
 (inlined by) vTaskSwitchContext at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3458
WARNING Decoded 0x4037d9e8: _frxt_dispatch at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:436
[... cut ...]
[23:03:06]Rebooting...
[23:03:06]ESP-ROM:esp32s3-20210327
[23:03:06]Build:Mar 27 2021
[23:03:06]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[23:03:06]Saved PC:0x420ae116
WARNING Decoded 0x420ae116: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[23:03:06]SPIWP:0xee
[23:03:06]mode:DIO, clock div:1
[23:03:06]load:0x3fce3808,len:0x43c
[23:03:06]load:0x403c9700,len:0xbec
[23:03:06]load:0x403cc700,len:0x2a3c
[23:03:06]entry 0x403c98d8

ssieb, 2024-11-11:
That's a stack overflow, but it appears to be happening in an esp-idf task, so I don't know what we can do about it.

hwilma, 2024-11-16:
After a lot of trial and error i ended up buying a new board. 
Tadaa problem solved. The camera module seems to be faulty.
Thank you for the support. I appreciate it.

closed 2024-11-16 by hwilma