esphome/issues #6419: 2024.10 broke the touch on WT32-SC01 module running LVGL

Arno80, 2024-11-10:
### The problem

Up until the update everything worked perfectly fine, after the update the touch states that Y is below 0 if I touch in the upper corner and it is all shifted, not able to touch the buttons on bottom right, it triggers the bottom center button.
Tried to play around with the calibration but LVGL output keeps stating that the touch is below 0, not able to get it right.

### Which version of ESPHome has the issue?

2024.10.x

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10 / 2024.11

### What platform are you using?

ESP32-IDF

### Board

WT32-SC01 3.5"

### Component causing the issue

touchscreen / lvgl

### Example YAML snippet

```yaml
touchscreen:
  - platform: ft63x6
    display: lcd_display
    transform:
      mirror_x: false
      mirror_y: false
      swap_xy: false
#    calibration:
#      x_min: 320
#      y_min: 480
#      x_max: 0
#      y_max: 0
    id: ${id_prefix}_touch
    i2c_id: i2c_bus_intern
    interrupt_pin: GPIO39
    on_touch:
      - lambda: |-
            ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
                touch.x,
                touch.y,
                touch.x_raw,
                touch.y_raw
                );

display:
  - platform: ili9xxx
    id: lcd_display
    model: st7796
    cs_pin: GPIO15
    dc_pin: GPIO21
    reset_pin: GPIO22
    rotation: 270
    update_interval: never
    auto_clear_enabled: false
    invert_colors: false
```

### Anything in the logs that might be useful for us?

```txt
11:41:40	[I]	[cal:390]	
x=1, y=309, x_raw=1, y_raw=464
11:41:40	[I]	[lvgl:000]	
[Info]	(725.785, +337267)	 indev_proc_press: pressed at x:309 y:318 	(in lv_indev.c line #818)
11:41:41	[W]	[component:237]	
Component lvgl took a long time for an operation (65 ms).
11:41:41	[W]	[component:238]	
Components should block for at most 30 ms.
11:41:41	[I]	[lvgl:000]	
[Info]	(725.859, +74)	 indev_proc_press: pressed at x:309 y:318 	(in lv_indev.c line #818)
11:41:41	[I]	[lvgl:000]	
[Info]	(725.898, +39)	 indev_proc_release: released 	(in lv_indev.c line #966)
11:41:41	[I]	[cal:390]	
x=233, y=59, x_raw=156, y_raw=90
11:41:41	[I]	[lvgl:000]	
[Info]	(726.084, +186)	 indev_proc_press: pressed at x:59 y:86 	(in lv_indev.c line #818)
11:41:41	[W]	[component:237]	
Component lvgl took a long time for an operation (63 ms).
11:41:41	[W]	[component:238]	
Components should block for at most 30 ms.
11:41:41	[I]	[lvgl:000]	
[Info]	(726.158, +74)	 indev_proc_press: pressed at x:59 y:86 	(in lv_indev.c line #818)
11:41:41	[I]	[lvgl:000]	
[Info]	(726.193, +35)	 indev_proc_release: released 	(in lv_indev.c line #966)
11:41:41	[D]	[homeassistant.text_sensor:017]	
'light.wled_2': Got state 'on'
11:41:41	[D]	[text_sensor:064]	
'ts_cab_light': Sending state 'on'
11:42:53	[I]	[cal:390]	
x=466, y=14, x_raw=311, y_raw=22
11:42:53	[I]	[lvgl:000]	
[Warn]	(798.195, +72002)	 indev_pointer_proc: Y is -147 which is smaller than zero 	(in lv_indev.c line #364)
11:42:53	[I]	[lvgl:000]	
[Info]	(798.198, +3)	 indev_proc_press: pressed at x:14 y:-147 	(in lv_indev.c line #818)
11:42:53	[I]	[lvgl:000]	
[Warn]	(798.228, +30)	 indev_pointer_proc: Y is -147 which is smaller than zero 	(in lv_indev.c line #364)
11:42:53	[I]	[lvgl:000]	
[Info]	(798.231, +3)	 indev_proc_press: pressed at x:14 y:-147 	(in lv_indev.c line #818)
11:42:53	[I]	[lvgl:000]	
[Warn]	(798.260, +29)	 indev_pointer_proc: Y is -147 which is smaller than zero 	(in lv_indev.c line #364)
11:42:53	[I]	[lvgl:000]	
[Info]	(798.262, +2)	 indev_proc_press: pressed at x:14 y:-147 	(in lv_indev.c line #818)
11:42:53	[I]	[lvgl:000]	
[Warn]	(798.298, +36)	 indev_pointer_proc: Y is -147 which is smaller than zero 	(in lv_indev.c line #364)
11:42:53	[I]	[lvgl:000]	
[Warn]	(798.329, +31)	 indev_pointer_proc: Y is -147 which is smaller than zero 	(in lv_indev.c line #364)
11:42:53	[I]	[lvgl:000]	
[Warn]	(798.361, +32)	 indev_pointer_proc: Y is -147 which is smaller than zero 	(in lv_indev.c line #364)
11:42:53	[I]	[lvgl:000]	
[Warn]	(798.404, +43)	 indev_pointer_proc: Y is -147 which is smaller than zero 	(in lv_indev.c line #364)
11:42:53	[I]	[cal:390]	
x=226, y=97, x_raw=151, y_raw=147
11:42:53	[I]	[lvgl:000]	
[Info]	(798.452, +48)	 indev_proc_press: pressed at x:97 y:93 	(in lv_indev.c line #818)
11:42:53	[I]	[lvgl:000]	
[Info]	(798.485, +33)	 indev_proc_press: pressed at x:97 y:93 	(in lv_indev.c line #818)
11:42:53	[I]	[lvgl:000]	
[Info]	(798.515, +30)	 indev_proc_release: released 	(in lv_indev.c line #966)
11:42:53	[D]	[homeassistant.text_sensor:017]	
'light.wled_2': Got state 'off'
11:42:53	[D]	[text_sensor:064]	
'ts_cab_light': Sending state 'off'
```

clydebarrow, 2024-11-13:
You need to use `transform:` in the touchscreen config to match the display rotation.

closed 2024-11-13 by clydebarrow

Arno80, 2024-11-16:
So, I transformed again.
The touch is showing the upper left as 0,0, the bottom right as 480,320, as you explained me (and as I did before using the docs).

lvgl is freaking out:
`
08:22:28 | [I] | [cal:390] | x=443, y=319, x_raw=0, y_raw=444
-- | -- | -- | --
08:22:28 | [I] | [lvgl:000] | [Warn]	(127.921, +12697)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Info]	(127.924, +3)	 indev_proc_press: pressed at x:319 y:-124 	(in lv_indev.c line #818)
08:22:28 | [I] | [lvgl:000] | [Warn]	(127.954, +30)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Info]	(127.956, +2)	 indev_proc_press: pressed at x:319 y:-124 	(in lv_indev.c line #818)
08:22:28 | [I] | [lvgl:000] | [Warn]	(127.992, +36)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Warn]	(128.024, +32)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Warn]	(128.056, +32)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Warn]	(128.088, +32)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Warn]	(128.120, +32)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Warn]	(128.152, +32)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Warn]	(128.184, +32)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Warn]	(128.216, +32)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Warn]	(128.248, +32)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:28 | [I] | [lvgl:000] | [Warn]	(128.280, +32)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
08:22:29 | [I] | [lvgl:000] | [Warn]	(128.323, +43)	 indev_pointer_proc: Y is -124 which is smaller than zero 	(in lv_indev.c line #364)
`

This is after I updated to 2024.10

So I would suggest to re-open and have a look.

Before the update it worked perfectly fine.

Arno80, 2024-11-16:
I needed to use transform and dimensions in the display config... found out because of issue #6352