esphome/issues #6420: ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)

deanfourie1, 2024-11-11:
### The problem

When attempting to update ESP8266 devices, I get the following error in esphome.

`ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)`

This appears to be only effecting the board

```
esp8266:
  board: esp01_1m
```

### Which version of ESPHome has the issue?

2024.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.11.1

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

not sure?

### Example YAML snippet

```yaml
esphome:
  name: camper-wall-lights
  friendly_name: camper-wall-lights

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""

web_server:
  port: 80
  log: true
  local: True
  version: 3
  include_internal: true
  auth:
    username: !secret web_server_username
    password: !secret web_server_password

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
    bssid: 
  fast_connect: true
  power_save_mode: none

  # Optional manual IP
  manual_ip:
    static_ip: 
    gateway: 
    subnet: 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Camper-Wall-Lights"
    password: ""

captive_portal:

output:
  - platform: gpio
    pin: GPIO0
    id: wall_lights

light:
  - platform: binary
    name: "Camper Wall Lights"
    output: wall_lights

sensor:
  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 60s
    entity_category: "diagnostic"

  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"

  - platform: uptime
    name: Uptime

time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: Device Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: IP Address
    bssid:
      name: Connected BSSID

  # human readable update text sensor from sensor:uptime
  - platform: template
    name: Uptime Human Readable
    id: uptime_human
    icon: mdi:clock-start
```

palves, 2024-12-16:
Seems like more recent ESPHome versions generate larger firmware binaries for this platform.  No idea if this is a genuine regression, or just a fact of life of dependencies just slowly growing bigger and a threshold was just recently crossed.  I suspect the former, but I'm not set up to bisect what could have caused it.

Note you can still OTA, but you have to do what the error suggests:
  "Please try flashing a minimal firmware (remove everything except ota)"

Thankfully you can upload firmware from the command line, so this is scriptable.

MattWestb, 2024-12-17:
From ESPH 2024-11 i cant updating my ESP-01S or LED drivers with 1M flash.
It was the same problem before start compressing the OTA and was working after re flashing then with cable for getting the bootloader updated.
Its not on very large issue but its annoying and the system is installing too much crap that is not needed for one simple project on one 1M device.

ssieb, 2024-12-17:
If you want it smaller, put less in the config.

MattWestb, 2024-12-17:
@ssieb Thanks for _good advice_ but its like baying one 6 seat car and going to the garage for service and then they is taking out 2 seats and you is only having 4 of them = not the same "product" as you was installing from the beginning.
Better recommending keeping one container of ESPH 2024.10 for small flash devices and hope its not serious security flaws that is not being patched.
And also what component is you recommending taking away then the RGB-Light with effect is the cor functionality. OTA, API, time service, WiFi or only the encryption of the API ???

randybb, 2024-12-17:
web server with local v3 files needs ~140kb.
Unfortunately, always when switching to newer libs they need more space. So if you still want to keep esp8266 (still supported by esphome but not recommended) you will need to stick with older version, or find a different solution - own code or a different project that is more optimized.

palves, 2024-12-18:
> There's a combination of things working against you here -- as was pointed out, external libraries/dependencies are becoming more complex and growing in size, configuration complexity plays into it and also the (by todays standards) limited storage capacity of these devices.

This is affecting pretty small configs with basic things, though.  If external dependencies are growing that much larger, then they are probably pulling in too much unnecessary code / bloat.  These small configs aren't really gaining new features with the upgrades, other than bugfixes.  The right thing to do here would be for someone to bisect and find exactly what is growing too big, and then that dependency can probably be broken down.

wesleyscaldwell, 2025-01-28:
I'm getting this on an 8266 that was purchased from athom for home assistant. So while it's getting phased out, why would they be selling devices that now can't be updated. 

https://github.com/athom-tech/athom-configs/blob/main/athom-relay-board-x4.yaml