esphome/issues #6423: Modbus error function code: 0x3 exception: 2 (ESP32 + RS485 Module)

michelmalle, 2024-11-11:
### The problem

Hello,

I try to read out several values of my Siemens 7KT1666 (MID) Smartmeter and facing the following issue:

- Using the tool "ModbusRtuMaster" on my Macbook + USB RS486 RTU adapter gets me a valid Modbus connection and the ability to read out the slave device.
- Trying the same using an ESP32 + RS485 module gets me Modbus Exception Codes
- Error appears with and without terminating by 120 Ohm resistor

[19:58:24][D][modbus:125]: Modbus error function code: 0x83 exception: 2
[19:58:24][E][modbus_controller:092]: Modbus error function code: 0x3 exception: 2 
[19:58:24][E][modbus_controller:101]: Modbus error - last command: function code=0x3  register address = 0x1  registers count=1 payload size=0

I attached the logfile.
[logs_sm-modbus_run.txt](https://github.com/user-attachments/files/17707652/logs_sm-modbus_run.txt)

Any ideas, what I could try to fix that?

Thanks!
Michel

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.9.3

### What platform are you using?

ESP32

### Board

JZK TTL to RS485

### Example YAML snippet

```yaml
esphome:
  name: sm-modbus
  friendly_name: SM-Modbus

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  - platform: esphome
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  output_power: 8.5dB

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sm-Modbus Fallback Hotspot"
    password: "xxx"

captive_portal:

web_server:

############### MODBUS CONFIGURATION ##############
uart:
  id: uart_1
  tx_pin: GPIO21
  rx_pin: GPIO20
  baud_rate: 38400 # Laut Setting am Smartmeter: default: 9600
  parity: NONE
  stop_bits: 1

modbus:
  id: modbus_1
  uart_id: uart_1 

modbus_controller:
  - id: energy_meter
    address: 0x01  # Modbus-Adresse des Smartmeters
    modbus_id: modbus_1
    update_interval: 10s #später hochsetzen......

sensor:
  - platform: modbus_controller
    modbus_controller_id: energy_meter
    name: "Spannung_L1"
    register_type: holding # "holding": function code 3 / "read": function code 4
    address: 0001  # Adresse für Spannung L1 mit Offset -1
    value_type: U_WORD
    unit_of_measurement: "V"
    filters:
      - multiply: 0.01 # Eventuell anpassen je nach Skala des Zählers
```

### Anything in the logs that might be useful for us?

```txt
[19:57:54][D][modbus:125]: Modbus error function code: 0x83 exception: 2
[19:57:54][E][modbus_controller:092]: Modbus error function code: 0x3 exception: 2 
[19:57:54][E][modbus_controller:101]: Modbus error - last command: function code=0x3  register address = 0x1  registers count=1 payload size=0
```

michelmalle, 2024-11-11:
Results using an usb to RS485 adapter on my Macbook:
[image]

The TTL to RS485 device: 
[image]

ssieb, 2024-11-11:
error code 2 is an invalid address.  Add `debug:` to the uart config to see what is sent and received.
The problem is that you're using the uart0 pins and haven't disabled serial logging.  Use different pins.

closed 2024-11-11 by ssieb

michelmalle, 2024-11-11:
Thanks for your fast reply. I´ve enable the debugging:

[21:14:14][D][uart_debug:114]: >>> 01:03:00:01:00:01:D5:CA
[21:14:14][D][modbus:125]: Modbus error function code: 0x83 exception: 2
[21:14:14][E][modbus_controller:092]: Modbus error function code: 0x3 exception: 2 
[21:14:14][E][modbus_controller:101]: Modbus error - last command: function code=0x3  register address = 0x1  registers count=1 payload size=0
[21:14:14][D][uart_debug:114]: <<< 01:83:02:C0:F1

Regarding the serial logging: From what I read in the documentation, thats only needed when using an ESP8266?

ssieb, 2024-11-11:
You don't need to disable the serial logging, but you should use different pins.  On a C3, esphome will log over the USB by default, but the bootloader will still write things to uart0 at boot.
esphome is sending the same command as on your Mac except that it's only 1 register.  Why is ModbusRtuMaster reading two registers?

michelmalle, 2024-11-11:
Isn´t esphome "automatically" readying out 2 registers? => **U_DWORD_R: unsigned 32 bit integer from 2 registers low word first.**
As per documentation of the Siemens device you have to read 2 registers:
[image]

Here the setting in ModbusRtuMaster:
[image]

Regarding the next steps: Can I assign any other PINs for RX/TX or should I use a special pair?
on https://esphome.io/components/logger#default-uart-gpio-pins no UART1 nor UART2 PINs are defined.

ssieb, 2024-11-11:
An ESP32 can use almost any pins for a uart.  There's a page here with some description: https://www.studiopieters.nl/esp32-c3-pinout/
Your config has `U_WORD`, not `U_DWORD_R`, so try changing that.  Also, the address is 2, not 1.

michelmalle, 2024-11-11:
Simply embarrassing ..... I´ve "double checked" that several times and was pretty sure U_DWORD_R was defined in my config. Pretty stupid simple! Did cost me the last 3 days. 

Thanks a lot, I should start pair programming;-)

Best,
Michel