esphome/issues #6424: First time user

berasomnath, 2024-11-12:
### The problem

I have installed "Empty ESPHome Device" on an ESP32 board. It got installed. The first page is opening at 192.168.43.16. The board is connected to my serial port. But I am not able to connect any "new device" as there is no "new device" button in the page. A
[image]
[image]
@lso there is an "OTA update" window for which I do not know. What to do from here now

### Which version of ESPHome has the issue?

ESPHome Web ea07b0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP-Wroom-32

berasomnath, 2024-11-12:
I have installed "Empty ESPHome Device" on an ESP32 board. It got installed. The first page is opening at 192.168.43.16. The board is connected to my serial port. But I am not able to connect any "new device" as there is no "new device" button in the page. How to proceed from here.

ssieb, 2024-11-12:
That is not the esphome dashboard, that's the device web page.
This is not the place for support questions.  Come ask for help on the esphome discord server.

closed 2024-11-12 by ssieb