esphome/issues #6427: DHT gets unresponsive after 10-15 minutes

rufinus, 2024-11-13:
### The problem

After a few minutes after a hard reboot the DHT21 sensor gets inresponsive. But is working again after hard reset.

### Which version of ESPHome has the issue?

2024.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.4

### What platform are you using?

ESP32

### Board

esp wroom 32

### Component causing the issue

dht

### Example YAML snippet

```yaml
esphome:
  name: ams-box
  friendly_name: AMS Box

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ams-Box Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxx"

captive_portal:

sensor:
  - platform: dht
    pin: 26
    temperature:
      name: "AMS Box Left Temperature"
      id: "temp_left"
    humidity:
      name: "AMS Box Left Humidity"
      id: "hum_left"
    update_interval: 15s  
  - platform: dht    
    pin: 27
    temperature:
      name: "AMS Box Right Temperature"
      id: "temp_right"
    humidity:
      name: "AMS Box Right Humidity"
      id: "hum_right"
    update_interval: 15s    
  - platform: mqtt_subscribe
    id: "heater_power"
    topic: "esp32/ams/power"   

mqtt:
  broker: 192.168.8.102

text_sensor:    
  - platform: mqtt_subscribe
    id: my_heater_status
    topic: "esp32/ams/status"

font:
  - file: "gfonts://Roboto"
    id: roboto_30
    size: 30    
  - file: "gfonts://Roboto"
    id: roboto_24
    size: 24
  - file: "gfonts://Roboto"
    id: roboto_16
    size: 16            
  - file: "gfonts://Roboto@700"
    id: header
    size: 30  

spi:
  clk_pin: 18
  mosi_pin: 23

display:
  - platform: waveshare_epaper
    cs_pin: 5
    dc_pin: 17
    busy_pin: 4
    reset_pin: 16
    model: 2.90inv2-r2
    full_update_every: 600
    update_interval: 15s
    rotation: 90°
    lambda: |-
      it.print(it.get_width()/2, 0, id(header), TextAlign::TOP_CENTER, "AMS Box Climate");
      it.printf(5, 50, id(roboto_30), "%.1f°C", id(temp_left).state);
      it.printf(5, 80, id(roboto_24), "  %.1f%%", id(hum_left).state);

      it.printf(it.get_width()-5, 50, id(roboto_30), TextAlign::TOP_RIGHT, "%.1f°C", id(temp_right).state);
      it.printf(it.get_width()-5, 80, id(roboto_24), TextAlign::TOP_RIGHT, "%.1f%%  ", id(hum_right).state);
      it.printf(it.get_width()/2, 110, id(roboto_16), TextAlign::TOP_CENTER, "Heizung ist %s (%.1f Watt)", id(my_heater_status).state.c_str(), id(heater_power).state );
```

### Anything in the logs that might be useful for us?

```txt
[bootup]
[20:18:28][W][component:237]: Component display took a long time for an operation (149 ms).
[20:18:28][W][component:238]: Components should block for at most 30 ms.
[20:18:28][D][dht:048]: Got Temperature=37.1°C Humidity=11.5%
[20:18:28][D][sensor:094]: 'AMS Box Left Temperature': Sending state 37.10000 °C with 1 decimals of accuracy
[20:18:28][D][sensor:094]: 'AMS Box Left Humidity': Sending state 11.50000 % with 0 decimals of accuracy
[20:18:29][D][dht:048]: Got Temperature=45.1°C Humidity=5.0%
[20:18:29][D][sensor:094]: 'AMS Box Right Temperature': Sending state 45.10000 °C with 1 decimals of accuracy
[20:18:29][D][sensor:094]: 'AMS Box Right Humidity': Sending state 5.00000 % with 0 decimals of accuracy
[20:18:31][D][sensor:094]: 'heater_power': Sending state 10.09000  with 1 decimals of accuracy
[20:18:43][W][component:237]: Component display took a long time for an operation (149 ms).
[20:18:43][W][component:238]: Components should block for at most 30 ms.
[20:18:43][D][dht:048]: Got Temperature=37.2°C Humidity=11.6%
[20:18:43][D][sensor:094]: 'AMS Box Left Temperature': Sending state 37.20000 °C with 1 decimals of accuracy
[20:18:43][D][sensor:094]: 'AMS Box Left Humidity': Sending state 11.60000 % with 0 decimals of accuracy
[20:18:44][D][dht:048]: Got Temperature=45.1°C Humidity=5.0%
[20:18:44][D][sensor:094]: 'AMS Box Right Temperature': Sending state 45.10000 °C with 1 decimals of accuracy
[20:18:44][D][sensor:094]: 'AMS Box Right Humidity': Sending state 5.00000 % with 0 decimals of accuracy
[20:18:44][W][component:237]: Component dht.sensor took a long time for an operation (53 ms).
[20:18:44][W][component:238]: Components should block for at most 30 ms.
[20:18:45][D][sensor:094]: 'heater_power': Sending state 10.22000  with 1 decimals of accuracy
[20:18:58][W][component:237]: Component display took a long time for an operation (149 ms).
[20:18:58][W][component:238]: Components should block for at most 30 ms.
[20:18:58][D][dht:048]: Got Temperature=37.2°C Humidity=11.6%
[20:18:58][D][sensor:094]: 'AMS Box Left Temperature': Sending state 37.20000 °C with 1 decimals of accuracy
[20:18:58][D][sensor:094]: 'AMS Box Left Humidity': Sending state 11.60000 % with 0 decimals of accuracy
[20:18:59][D][dht:048]: Got Temperature=45.2°C Humidity=4.9%
[20:18:59][D][sensor:094]: 'AMS Box Right Temperature': Sending state 45.20000 °C with 1 decimals of accuracy
[20:18:59][D][sensor:094]: 'AMS Box Right Humidity': Sending state 4.90000 % with 0 decimals of accuracy
[20:19:00][D][sensor:094]: 'heater_power': Sending state 10.21000  with 1 decimals of accuracy
[20:19:13][W][component:237]: Component display took a long time for an operation (150 ms).
[20:19:13][W][component:238]: Components should block for at most 30 ms.
[20:19:13][D][dht:048]: Got Temperature=37.2°C Humidity=11.6%
[20:19:13][D][sensor:094]: 'AMS Box Left Temperature': Sending state 37.20000 °C with 1 decimals of accuracy
[20:19:13][D][sensor:094]: 'AMS Box Left Humidity': Sending state 11.60000 % with 0 decimals of accuracy
[20:19:14][D][dht:048]: Got Temperature=45.2°C Humidity=4.9%
[20:19:14][D][sensor:094]: 'AMS Box Right Temperature': Sending state 45.20000 °C with 1 decimals of accuracy
[20:19:14][D][sensor:094]: 'AMS Box Right Humidity': Sending state 4.90000 % with 0 decimals of accuracy
[20:19:14][W][component:237]: Component dht.sensor took a long time for an operation (53 ms).
[20:19:14][W][component:238]: Components should block for at most 30 ms.
[20:19:15][D][sensor:094]: 'heater_power': Sending state 10.27000  with 1 decimals of accuracy
[20:19:28][W][component:237]: Component display took a long time for an operation (149 ms).
[20:19:28][W][component:238]: Components should block for at most 30 ms.
[20:19:28][D][dht:048]: Got Temperature=37.2°C Humidity=11.5%
[20:19:28][D][sensor:094]: 'AMS Box Left Temperature': Sending state 37.20000 °C with 1 decimals of accuracy
[20:19:28][D][sensor:094]: 'AMS Box Left Humidity': Sending state 11.50000 % with 0 decimals of accuracy
[20:19:29][D][dht:048]: Got Temperature=45.3°C Humidity=4.9%
[20:19:29][D][sensor:094]: 'AMS Box Right Temperature': Sending state 45.30000 °C with 1 decimals of accuracy
[20:19:29][D][sensor:094]: 'AMS Box Right Humidity': Sending state 4.90000 % with 0 decimals of accuracy
[20:19:30][D][sensor:094]: 'heater_power': Sending state 10.19000  with 1 decimals of accuracy
[20:19:43][W][component:237]: Component display took a long time for an operation (150 ms).
[20:19:43][W][component:238]: Components should block for at most 30 ms.
[20:19:43][D][dht:048]: Got Temperature=37.2°C Humidity=11.5%
[20:19:43][D][sensor:094]: 'AMS Box Left Temperature': Sending state 37.20000 °C with 1 decimals of accuracy
[20:19:43][D][sensor:094]: 'AMS Box Left Humidity': Sending state 11.50000 % with 0 decimals of accuracy
[20:19:44][D][dht:048]: Got Temperature=45.2°C Humidity=4.9%
[20:19:44][D][sensor:094]: 'AMS Box Right Temperature': Sending state 45.20000 °C with 1 decimals of accuracy
[20:19:44][D][sensor:094]: 'AMS Box Right Humidity': Sending state 4.90000 % with 0 decimals of accuracy
[20:19:44][W][component:237]: Component dht.sensor took a long time for an operation (53 ms).
[20:19:44][W][component:238]: Components should block for at most 30 ms.
[20:19:45][D][sensor:094]: 'heater_power': Sending state 10.37000  with 1 decimals of accuracy
[20:19:49][D][sensor:094]: 'heater_power': Sending state 12.88000  with 1 decimals of accuracy
[20:19:54][D][sensor:094]: 'heater_power': Sending state 336.09000  with 1 decimals of accuracy
[20:19:58][W][component:237]: Component display took a long time for an operation (149 ms).
[20:19:58][W][component:238]: Components should block for at most 30 ms.
[20:19:58][W][dht:174]: Requesting data from DHT failed!
[20:19:58][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number) and consider manually specifying the DHT model using the model option.
```

ssieb, 2024-11-13:
Do both sensors fail?
If you add a reboot button, does that fix it?  Or does it need a power cycle?

routeroscn, 2024-11-21:
I have also encountered this issue. I am using two DHT22 sensors, and one of them stops working in some cases, displaying NAN. I have an ESP8266 with a relay connected to control an AC motor or fan switch. Occasionally, when turning the relay on or off, the DHT sensor stops working. Therefore, I believe this may be a hardware issue.