esphome/issues #6429: Platform xpt2046 and binary sensor platform touchscreen not working anymore

TegameTD, 2024-11-13:
### The problem

Hi, I have several Aztouch devices that use that touchscreen platform, for years they have always worked even with esphome updates, I have just updated two of these devices from version 2024.6.6 to the latest available and in both devices the binary sensors configured for the touchscreen stopped working, downgrading solves the problem.

Here is the relevant part of my configuration:

'''
#This enable the touch screen
touchscreen:
  platform: xpt2046
  id: xpt2046_touch
  cs_pin: 14 
  interrupt_pin: 27 
  update_interval: 50ms
  transform:
    mirror_x: false
    mirror_y: false
    swap_xy: true
  calibration: 
    x_min: 3575
    x_max: 368
    y_min: 3800
    y_max: 441
  #When the screen is touched turn on the backlight
  on_touch:
    - script.stop: backlight_timer
    - script.execute: backlight_timer
#  Uncomment for calibrations
    - lambda: |-
        ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
          touch.x,
          touch.y,
          touch.x_raw,
          touch.y_raw
           );
binary_sensor:

#Touch key minus
  - platform: touchscreen
    id: touch_key_minus
    x_min: 10
    x_max: 60
    y_min: 240
    y_max: 320
    on_press:
      - climate.control:
          id: termostato1
          target_temperature: !lambda return (id(termostato1).target_temperature - 0.5);

'''

When I click the virtual button in the logs I see that x and y at the point where I touched are exactly inside the virtual square declared in the binary sensor but in fact it doesn't work, the binary sensor is not triggered. Do you know why?
Is there anything I can do to fix it?

Thank You.

### Which version of ESPHome has the issue?

2024.10.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

Azdelivery Aztouch

clydebarrow, 2024-11-13:
I tested this and can't find an issue. You didn't show your display config - it matters. Also attach some logs showing the touch report.

TegameTD, 2024-11-14:
Here the complete code:
```
esphome:
  name: termostato-meteo
  platform: ESP32
  board: esp32dev
  on_boot:

#If you want to start thermostat in heat mode when it loose power uncomment the bottom lines
#    - climate.control:
#        id: termostato1
#        mode: HEAT

#This command starts the backlight subsystem
    - script.execute: backlight_timer

#Substitutions
substitutions:
#Set here your entities
# Heater entity from Home Assistant. In my case, I used a Shelly 1v3 to control the gas heater's on/off status
#You can also use a relay connected to a gpio but I do not recommend it especially if the load is alternating current the absorption peaks could glue the contacts of the cheap relays.
#I also noticed that the electromagnetic peak sometimes crashes the esp32. Since we are talking about a delicated system like a gas heater, it is better to opt for a certified device.
  heater: switch.shelly_caldaia
# Weather entity from home assistant
  weather_entity: weather.casa
# Total consumption sensor imported from Home Assistant. I use a Shelly EM with two CT sensors. This first sensor measures the main phase of my home's electrical grid
  tc: sensor.consumo_totale_power
# Partial consumption entity, such as a washing machine, imported from Home Assistant. This second CT sensor measures the garage's electrical phase
  pc: sensor.consumo_cantina_power
#Weather now icon size
  icon_xy: '65x65'

# Enable logging
logger:
  logs:
    component: ERROR

# Enable Home Assistant API
api:
  services:
  # Receive sound from homeassistant as service (only buzzer rtttl)
    - service: play_rtttl
      variables:
        song_str: string
      then:
        - rtttl.play:
            rtttl: !lambda 'return song_str;'

# Buzzer
rtttl:
  output: rtttl_out

# This enable firmware upload via ota
ota:
  platform: esphome

#This is the wifi block connection, change your wifi credential in secrets.yaml before tryng a firmware upload
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Meteo-thermo"
    password: !secret fallpass 

# Enables the captive portal. If the ESP32 cannot connect to the existing WiFi network, it creates an access point with the 'Meteo-thermo' SSID.
# Users can connect to this AP and configure the WiFi credentials through the portal
captive_portal:

#This import arduinojson library for extract weather forecasts
json:
[... 956 lines cut ...]
[18:24:17][D][binary_sensor:036]: 'Movimento': Sending state OFF
[18:24:17][D][homeassistant.sensor:024]: 'sensor.consumo_totale_power': Got state 190.48
[18:24:17][D][sensor:094]: 'consumo_t': Sending state 190.48000  with 1 decimals of accuracy
[18:24:20][D][sensor:094]: 'Temperatura': Sending state 18.40676 °C with 1 decimals of accuracy
[18:24:20][D][climate:396]: 'Termostato' - Sending state:
[18:24:20][D][climate:399]:   Mode: HEAT
[18:24:20][D][climate:401]:   Action: HEATING
[18:24:21][D][climate:419]:   Current Temperature: 18.41°C
[18:24:21][D][climate:425]:   Target Temperature: 19.00°C
[18:24:21][D][sensor:094]: 'Pressione': Sending state 950.63782 hPa with 1 decimals of accuracy
[18:24:21][D][sensor:094]: 'Umidita': Sending state 40.40625 % with 1 decimals of accuracy
[18:24:21][D][binary_sensor:036]: 'Movimento': Sending state ON
[18:24:21][D][light:036]: 'Luce termostato' Setting:
[18:24:21][D][light:051]:   Brightness: 100%
[18:24:21][D][light:085]:   Transition length: 1.0s
[18:24:23][D][homeassistant.sensor:024]: 'sensor.consumo_totale_power': Got state 201.51
[18:24:23][D][sensor:094]: 'consumo_t': Sending state 201.50999  with 1 decimals of accuracy
[18:24:24][D][xpt2046:055]: Touchscreen Update [677, 3650], z = 440
[18:24:24][D][light:036]: 'Luce termostato' Setting:
[18:24:24][D][light:051]:   Brightness: 100%
[18:24:24][D][light:085]:   Transition length: 1.0s
[18:24:24][I][cal:131]: x=14, y=216, x_raw=677, y_raw=3650 <-When i click "minus key"
```

Nothing happen the log not reporting anything, no binary sensor message.

clydebarrow, 2024-11-16:
Remove the `rotation: 90` from your display config and replace it with an appropriate `transform` config. You will need `swap_xy` true for rotation but might have to try different combinations of `mirror_x` and `mirror_y`. You will need to specify the width and height of the display *after* transformation - width 320, height 240.

Once the display is oriented correctly fix the touchscreen config - put the calibration in correct order (min less than max) then adjust the `transform` until it works correctly. The required transform will quite possibly be different to what is in the display config! The touchscreen and display hardware alignment do not necessarily match, which is why some experimentation is required.

clydebarrow, 2024-11-18:
Well, that's a different problem. I get that error on your original code anyway.

TegameTD, 2024-11-18:
Really strange, if I put rotation 90 back and comment on the transform section the monitor starts working again without problems.

TegameTD, 2024-11-25:
I've made some progress in identifying the problem.

In the latest ESPHome update (2024.11.1), I'm no longer experiencing the "out of memory" error. I applied the new transform: section to both the display and the touchscreen, and this has resolved the issue of incorrect touch coordinates.

Here is the updated configuration:
```
touchscreen:
  platform: xpt2046
  id: xpt2046_touch
  cs_pin: 14 
  interrupt_pin: 27 
  update_interval: 50ms
  threshold: 400
  transform:
    mirror_x: true
    mirror_y: true
    swap_xy: true
  calibration: 
    x_min: 400
    x_max: 3700
    y_min: 590
    y_max: 3760

display:
  - platform: ili9xxx
    model: ili9341
    cs_pin: 5 
    dc_pin: 4  
    reset_pin: 22 
    id: tft_ha
    invert_colors: false
    transform:
      swap_xy: true
    dimensions:
      height: 240
      width: 320
    pages:
      - id: termostato
        lambda: |- ...
```

With this setup, the touch coordinates are now accurate.

However, there’s still a major issue with the binary sensors. When touching the binary sensors, only about 1 in 10 presses is registered correctly. Frequently, touches are interpreted as Touchscreen Update events rather than binary sensor presses, which makes the functionality unreliable.

Additionally:

Sometimes, pressing a binary sensor sets the state to ON, but it never switches back to OFF until another press is made.
I've been experimenting with different values for delay and threshold, but none of these seem to resolve the issue completely.
Here’s an example of the logs:

```
[13:54:48][D][binary_sensor:036]: 'touch_key_next': Sending state ON
[13:54:48][D][binary_sensor:036]: 'touch_key_next': Sending state OFF
[13:54:50][D][xpt2046:055]: Touchscreen Update [705, 3642], z = 646
[13:54:51][D][xpt2046:055]: Touchscreen Update [691, 3689], z = 507
[13:54:52][D][xpt2046:055]: Touchscreen Update [682, 3620], z = 750
[13:54:53][D][xpt2046:055]: Touchscreen Update [714, 3621], z = 788
[13:54:54][D][xpt2046:055]: Touchscreen Update [692, 3682], z = 433
[13:54:55][D][xpt2046:055]: Touchscreen Update [723, 3631], z = 770
[13:54:56][D][xpt2046:055]: Touchscreen Update [2196, 2665], z = 49
```

Or:
```
[14:11:54][D][binary_sensor:036]: 'touch_key_next': Sending state ON

(Off state missing)

[14:11:55][D][xpt2046:055]: Touchscreen Update [748, 3589], z = 734
[14:11:56][D][xpt2046:055]: Touchscreen Update [704, 3645], z = 672
[14:11:57][D][xpt2046:055]: Touchscreen Update [694, 3664], z = 668

```

It seems there’s a conflict between the Touchscreen Update events and the binary sensors (on_press and on_release). I suspect this is causing unreliable behavior.

Have you experienced similar issues, or do you have any suggestions on how to fix this? I'd appreciate any advice to improve the reliability of the binary sensors.

Thanks in advance!

clydebarrow, 2024-12-01:
It's likely that the touchscreen controller is not generating an interrupt when a touch is released. There is a config option to deal with that - see https://arc.net/l/quote/ozlcrhmk - or just do what you did - remove the interrupt pin definition, which makes it fall back to polling.

closed 2024-12-01 by clydebarrow