esphome/issues #6434: ble_client sensor with two characteristics values are crossed

shaneinak, 2024-11-15:
### The problem

I have a BLE sensor with two characteristics.  One is for the amount of oil in my tank in gallons, the other is the percent full .   It was all working as expected and displaying in my home assistant dashboard as expected. This setup is still in a lab environment and I wanted to ensure it could survive a reboot.  After the reboot the readings appear but the percentage reading shows up as the gallons and the gallons show up as the percentage.  It is a 300 gallon tank that has 262.8 gallons that comes out to 87.6% full.  How did a reboot of the ble_sensor and the BLE proxy device flip these readings?

[image]

### Which version of ESPHome has the issue?

ESPHome 2024.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.11.1

### What platform are you using?

ESP32

### Board

ESP-WROOM-32 ESP32 ESP-32S 

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-1fff70
  friendly_name: BLE Proxy
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
esp32_ble_tracker:
  
ble_client:
  - mac_address: 40:4C:CA:48:45:66
    id: Tank_Level_Sensor
sensor:
  - platform: ble_client
    type: characteristic
    ble_client_id: Tank_Level_Sensor
    name: "Amount of Oil"
    service_uuid: '5624c4a0-a813-48d1-84f8-39aadcf45b76'
    characteristic_uuid: 'bc7f5493-977c-40a9-8546-9b987ff71586'
    unit_of_measurement: "gal"
    icon: "mdi:gauge"
    accuracy_decimals: 1
    lambda: |-
      return *((float*)(&x[0]));
  - platform: ble_client
    type: characteristic
    ble_client_id: Tank_Level_Sensor
    name: "Tank Percent Full"
    service_uuid: '5624c4a0-a813-48d1-84f8-39aadcf45b76'
    characteristic_uuid: 'efb623c7-1297-4ca5-bf35-cfcaa1949ce7'
    unit_of_measurement: "%"
    icon: "mdi:gauge"
    accuracy_decimals: 1
    lambda: |-
      return *((float*)(&x[0]));
wifi:
[... 197 lines cut ...]
  //pAdvertising->addManufacturerSpecificData(MANUFACTURER_ID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
}

void loop()
{
  if (!deviceConnected && oldDeviceConnected) {
    delay(500);                   // give the bluetooth stack the chance to get things ready
    BLEDevice::startAdvertising(); // restart advertising
    oldDeviceConnected = deviceConnected;
  }
  // connecting
  if (deviceConnected && !oldDeviceConnected) {
    // do stuff here on connecting
    oldDeviceConnected = deviceConnected;
  }

    pCharacteristicGallon->setValue(gallons_of_oil);
    pCharacteristicPercent->setValue(percent);

    delay(1000);
}

shaneinak, 2024-11-15:
Well... I rebooted again and the readings are correct again.  I guess this is just one of those quirky issues that gets fixed with the magic ON/OFF button.  if anyone knows what may be causing this odd behavior I'd appreciate any tips.  At one point I had three characteristics, the third being the battery level of the sensor.  I was getting many combinations of crisscrossed values.  I ended up removing the battery since the battery life was to short, I decided to just hardwire the sensor power for reliable power.  The values getting crossed up is a strange one and I am  a total armature and most likely don't have the most efficient coding.
[image]

clydebarrow, 2024-11-16:
Next time it happens, use a tool like nRFConnect to check what the device itself is actually sending.

closed 2024-11-16 by clydebarrow

shaneinak, 2024-11-16:
That was the first thing I tried. I should have mentioned that in my post.
nRF connect  always shows the correct values for the respective
characteristic UUIDs.