esphome/issues #6435: SPI control not working since 2024.9.0

Audrey-P-Cooper, 2024-11-16:
### The problem

I have an issue when initialising an SPI digital potentiometer (MCP4131) where the processor fails to boot with a wild pointer error. I have simplified my program to only use the SPI relevant  parts.

Apologies, this is my second attempt at reporting this problem- I accidentally closed the original issue whilst trying to simplify the code

### Which version of ESPHome has the issue?

> 2024.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

> 2024.9.0

### What platform are you using?

ESP8266

### Board

WEMOS D1 mini clone

### Component causing the issue

SPI

### Example YAML snippet

```yaml
esphome:
  name: spicontrol
  friendly_name: eStimControl

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "FL8MVizjiPx1vhWx3MMSKSKK2N/lEq0u5Dg6O0st5zE="

ota:
  - platform: esphome
    password: "c58e7ac736ee8db8b42ca01b31878441"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "spiTest Fallback Hotspot"
    password: "gTpwdlkDZ1Ew"

   
spi:
  clk_pin: D5
  mosi_pin: D7
  miso_pin: D6
  interface: hardware

spi_device:
    id: spi_power
    cs_pin: D8
    data_rate: 2MHz
    mode: 3
    bit_order: lsb_first

number:
  - platform: template
    id: Power_Slider
    name: "SPI Power"
    mode: slider
    min_value: 0
    max_value: 127
    step: 1
    initial_value: '0'
    optimistic: true
    on_value:
      then:
        lambda: !lambda |-
          id(spi_power).enable();
          id(spi_power).write_byte(0x00);
          id(spi_power).write_byte((byte)(127-(int)id(Power_Slider)));
          id(spi_power).disable();
```

### Anything in the logs that might be useful for us?

```txt
[10:46:37][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[10:46:37][I][app:029]: Running through setup()...
[10:46:37][D][spi:035]: Setting up SPI bus...
[10:46:37][D][number:012]: 'SPI Power': Sending state 0.000000
[10:46:37]
[10:46:37]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[10:46:37]
[10:46:37]Exception (28):
WARNING Exception type: Access to invalid address: LOAD (wild pointer?)
[10:46:37]epc1=0x4021c06e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
WARNING Decoded 0x4021c06e: std::_Function_handler<void (float), setup::{lambda(float)#1}>::_M_invoke(std::_Any_data const&, float&&) at main.cpp
[10:46:37]
[10:46:37]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[10:46:37]
[10:46:37]ctx: cont
[10:46:37]sp: 3ffffd80 end: 3fffffd0 offset: 0150
[10:46:37]3ffffed0:  3fffff00 3ffffef0 00000010 3ffef320  
[10:46:37]3ffffee0:  3fffff00 3ffffef0 3fff0e8c 4021c0d2  
WARNING Decoded 0x4021c0d2: esphome::LambdaAction<float>::play(float)
[10:46:37]3ffffef0:  00000000 3fffff38 00000004 402248cb  
WARNING Decoded 0x402248cb: operator new(unsigned int)
[10:46:37]3fffff00:  3ffe8f75 3ffef320 00000001 402278b4  
WARNING Decoded 0x402278b4: esphome::Action<float>::play_complex(float)
[10:46:37]3fffff10:  00000000 3fff0e74 3fff0e64 402140a8  
WARNING Decoded 0x402140a8: esphome::number::Number::publish_state(float)
[10:46:37]3fffff20:  00000000 00000000 3ffef500 3fff0f08  
[10:46:37]3fffff30:  00000000 3fff0f04 3ffef31c 4021bc77  
WARNING Decoded 0x4021bc77: esphome::Scheduler::process_to_add()
[10:46:37]3fffff40:  3fff0e24 00000003 3fff0ddc 40215ff1  
WARNING Decoded 0x40215ff1: esphome::template_::TemplateNumber::setup()
[10:46:37]3fffff50:  00000000 00000002 3ffef31c 4021c1a4  
WARNING Decoded 0x4021c1a4: esphome::preferences::IntervalSyncer::setup()
[10:46:37]3fffff60:  3fff0eac 00000003 3fff0e24 4021b04e  
WARNING Decoded 0x4021b04e: esphome::PollingComponent::call_setup()
[10:46:37]3fffff70:  3fff0d4c 00000002 3ffef31c 4021a995  
WARNING Decoded 0x4021a995: esphome::Application::setup()
[10:46:37]3fffff80:  0000000a 0000000a 3fff0eac feefeffe  
[10:46:37]3fffff90:  00000010 3fff0b64 3fffff6c 3fff0bbc  
[10:46:37]3fffffa0:  feefeffe feefeffe feefeffe 3ffef760  
[10:46:37]3fffffb0:  3fffdad0 00000000 3ffef734 40224cd8  
WARNING Decoded 0x40224cd8: loop_wrapper() at core_esp8266_main.cpp
[10:46:37]3fffffc0:  feefeffe feefeffe 3fffdab0 401003cd  
WARNING Decoded 0x401003cd: cont_wrapper
[10:46:37]<<<stack<<<
[10:46:37]
[10:46:37]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[10:46:37]
[10:46:37] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[10:46:37]
[10:46:37]load 0x4010f000, len 3424, room 16 
[10:46:37]tail 0
[10:46:37]chksum 0x2e
[10:46:37]load 0x3fff20b8, len 40, room 8 
[10:46:37]tail 0
[10:46:37]chksum 0x2b
[10:46:37]csum 0x2b
[10:46:37]v000703c0
[10:46:37]~ld
[10:46:37]\x84\xe3\xe4\033\x83o\xe4\xdb{\xc3\xfbg|\xec$\xc4$`c\x9b\x9f|;\x93d\x9c'\xe0\x83o\xe3
```

### Additional information

I note that there was a small change to spi.h and spi.cpp in August which I assume would have been released in with 2024.9.0

jesserockz, 2024-11-16:
Add `setup_priority: 500` to the template number.

Audrey-P-Cooper, 2024-11-30:
Thank you. That has resolved the boot up issue. I still have a problem with the SPI potentiometer not working, but I will do further investigation and raise a new ticket if necessary.

closed 2024-11-30 by Audrey-P-Cooper