esphome/issues #6438: TCA9555 does not report input changes

NdK73, 2024-11-17:
### The problem

binary_sensors using GPIO pins from tca9555 do not change regardless of the signals applied.

The relevant yaml code, with outputs and lights left to check that the chip is working, is in the dedicated section below.

### Which version of ESPHome has the issue?

2024.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.11.2

### What platform are you using?

ESP8266

### Board

esp12e

### Component causing the issue

tca9555

### Example YAML snippet

```yaml
substitutions:
  device_name: test
  upper_devicename: Test

esp8266:
board: esp12e

i2c: # required for tca9555

tca9555:
  - id: exp0 # Consider only base IOs
    address: 0x20 # 20-27, 20 is the first expander, default would be 0x21
output:
  - id: pin_ledusr
    platform: gpio
    pin:
      tca9555: exp0
      number: 9
      mode: output
light:
  - id: ledconn
    platform: status_led
    pin:
      tca9555: exp0
      number: 8
      mode: output
      #inverted: true
  - id: ledusr
    platform: binary
    name: "${upper_devicename} Led Usr"
    output: pin_ledusr
binary_sensor:
  - id: in1
    name: "${upper_devicename} In 1"
    platform: gpio
    pin:
      tca9555: exp0
      number: 2
      mode: INPUT # already tried 'input'

esphome:
  name: $device_name

# Enable Home Assistant API
api:
  encryption:
    key: "##################"
  reboot_timeout: 0s
    

ota:
  - platform: esphome
    password: "###############"

web_server:
  port: 80
  version: 2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${upper_devicename}"
    password: !secret wifi_fallback_pass

captive_portal:

# Enable logging
logger:
  baud_rate: 0 # Do not log over serial port
```

### Additional information

The circuit is working  correctly with an older custom component (quite messy code that I wrote many years ago), so it's not a hardware issue. The board where the esp12e is mounted is my DomoNode 2.1 (https://github.com/NdK73/HomeAutomation/tree/master/domonode2).

NdK73, 2024-11-21:
Just tested new ESPHome: no change :(

`ESPHome version 2024.11.0 compiled on Nov 21 2024, 19:56:06`

holk, 2024-11-25:
Same problem here.
Tested on ESP32 with ESPHome 2024.10.0

I've 3 tca9555 on the bus (16 inputs, 16 outputs, 8 I/O), all the outputs are working properly but none of the inputs detects a change.

holk, 2024-11-26:
Sniffed the i2c bus, can see configuration commands and output port commands (0x02, 0x03) but none read port command (0x00, 0x01) between the esp32 and tca9555s

ViktorPankraz, 2024-11-27:
Still not working with ESPHome 2024.11.1

NdK73, 2024-11-29:
IIUC it seems TCA9555Component::digital_read_hw never gets called.
It should either be called either by TCA9555Component::loop (instead of clearing the cache, but that would slow down the system) or by TCA9555::digital_read_cache if the cache have been invalidated.

NdK73, 2024-12-03:
Digging deeper.

I found that TCA9555Component::digital_read_hw should be called by CachedGpioExpander::digital_read() , but reads do not get triggered probably because of a logic flaw (if my C++ is not too rusty):
  - the loop() initializes read_cache_invalidated_[*] to false
  - the first call to digital_read(x) sets read_cache_invalidated_[x] to true and reads **cached** value
  - to actually trigger a read_hw(x) it requires *two* calls to digital_read(x)
  - **every** other call to digital_read(x) before loop() actually triggers i2c traffic for read_hw!

It seems digital_read_cache(pin) and digital_read_hw(pin) are swapped in cached_gpio.h

Also, this comment in tca9555 seems wrong:
  /// Mask for the pin mode - 1 means output, 0 means input
Actually, TCA9555Component::pin_mode sets the opposite (1 for Input, 0 for Output) -- the mnemonic way :)

Hope someone more expert in ESPHome internals can have a look at it.

NdK73, 2024-12-05:
Another logical flaw is that caching considers single bits, but should act on the two banks (0 and 1), clearing or setting the invalidated flag on all the pins of the same bank at the same time. It simply misses the notion of banks :(
Currentl logic leads (once the above bug is fixed) to bursts of i2c traffic: a read for bit 0, another for bit 1 & so on. Moreover, reads are not "coherent": if I read pin1, then pin2, then pin1 again I could get a different value from the second read of pin1: quite unexpected from a cached read!

Actually, the TCA9555Component treats all the IOs as a single bank and digital_read_hw() always reads both banks:
  success = this->read_bytes(TCA9555_INPUT_PORT_REGISTER_0, data, 2);
That should be the most efficient way to use the bus.

haus-bus, 2024-12-05:
The TCA9555 has an interrupt line to indicate if any input has changed its state. Wouldn't it make sense to use that line to invalidate all pins. For the case this pin is not used on the board maybe a polling time can be configured to invalidate the inputs. So that the pin can be read always from cache as long as the interrupt line is not set or the polling time is not elapsed.

So the logic could be something like this in digital_read_hw():

if ( intPinIsUsed() && intPin.isLow() ) read_bytes()
else if ( !intPinIsUsed() && pollingTime.isElapsed() ) read_bytes()
else return cachedPinState

This should avoid the bursts on i2c

NdK73, 2024-12-05:
That would be better iif the interrupt pin is used at all (on my DomoNode it's not connected). And that would require some major changes (you have to tell the component that the int pin is used and where it's connected -> changes to the yaml code).
I'm trying to focus on more trivial fixes that can be implemented quickly. Having just a single "cache_valid" flag that can either be reset from loop() or from interrupt signal would be a bonus.

NdK73, 2024-12-13:
@mobrembski could you please have a look?

I tried changing gpio_expander/cached_gpio.h to:
```
#pragma once

#include <array>
#include <cstdint>
#include "esphome/core/hal.h"

namespace esphome {
namespace gpio_expander {

/// @brief A class to cache the read state of a GPIO expander.
template<typename T, T N> class CachedGpioExpander {
 public:
  bool digital_read(T pin) {
    if (this->read_cache_invalidated_[0]) { //@@@ 0 instead of pin: single global cache
      this->read_cache_invalidated_[0] = false; //@@@ idem
      return this->digital_read_hw(pin);
    }
    return this->digital_read_cache(pin);
  }

  void digital_write(T pin, bool value) { this->digital_write_hw(pin, value); }

 protected:
  virtual bool digital_read_hw(T pin) = 0;
  virtual bool digital_read_cache(T pin) = 0;
  virtual void digital_write_hw(T pin, bool value) = 0;

  void reset_pin_cache_() {
    for (T i = 0; i < N; i++) {
      this->read_cache_invalidated_[i] = true;
    }
  }

  std::array<bool, N> read_cache_invalidated_{};
};

}  // namespace gpio_expander
}  // namespace esphome
```
and now all [*] inputs are read correctly.
I know this is just a quick&dirty workaround that will possibly break some other expander, but I'm no expert in EspHome internals...

[*] Well, actually all except the first, but this could be a HW issue, couldn't test with my old code yet...

NdK73, 2024-12-14:
Maybe also @jesserockz could be interested: IMHO the highlighted issues should be fixed before extending the caching mechanism to other components.

NdK73, 2024-12-14:
I just ruled out HW problems: with my old (really ugly) code, it works. Every input and every output.

Digging in the code, I noticed another issue: TCA9555Component::digital_read_hw() does **not** return the pin state, just the success of the read operation (true if OK). So, instead of `return this->digital_read_hw(pin);` it should be:
`if(!this->digital_read_hw(pin)) return false; // Error reading from expander`
leaving to the fall-through `return this->digital_read_cache(pin);` the actual reading of the value.

With this change, inputs are **sometimes** working as expected (can change behaviour after a simple reboot, with no code changes!). Might(?) be related to TCA9555's mode mask that IIUC defaults to all **inputs**, so in tca9555.h it should be `uint16_t mode_mask_{0xff};` ?

That **does not** fix the issue with the banks .

NdK73, 2024-12-19:
Still not working (w/o patch) in 2024.12.1 :(

mobrembski, 2024-12-19:
I'll take a look at this issue after new year, preferably in second half. I
dont have board with TCA9555 currently where i'm living.

freddiemare, 2024-12-21:
Thanks for having a look.

I also experienced the same the outputs are working but the inputs do not change states inverter true or false also tested swoped pins with output pins but inputs refuse to change state only state that change is in HA when I change the invert but input stays the same when trigerd with 3.3v or 5v.

NdK73, 2024-12-23:
Here's my patch. It's currently working in single-bank mode to minimize changes to the rest of the code.
```
git diff
diff --git a/esphome/components/gpio_expander/cached_gpio.h b/esphome/components/gpio_expander/cached_gpio.h
index 784c5f0f4..a23eaa47f 100644
--- a/esphome/components/gpio_expander/cached_gpio.h
+++ b/esphome/components/gpio_expander/cached_gpio.h
@@ -11,11 +11,11 @@ namespace gpio_expander {
 template<typename T, T N> class CachedGpioExpander {
  public:
   bool digital_read(T pin) {
-    if (!this->read_cache_invalidated_[pin]) {
-      this->read_cache_invalidated_[pin] = true;
-      return this->digital_read_cache(pin);
+    if (this->read_cache_invalidated_[0]) { //@@@ 0 instead of pin: single global cache
+      this->read_cache_invalidated_[0] = false; //@@@ idem
+      if(!this->digital_read_hw(pin)) return false; // Failed reading
     }
-    return this->digital_read_hw(pin);
+    return this->digital_read_cache(pin);
   }

   void digital_write(T pin, bool value) { this->digital_write_hw(pin, value); }
@@ -27,7 +27,7 @@ template<typename T, T N> class CachedGpioExpander {

   void reset_pin_cache_() {
     for (T i = 0; i < N; i++) {
-      this->read_cache_invalidated_[i] = false;
+      this->read_cache_invalidated_[i] = true;
     }
   }
```

It accounts for:
  - digital_read_hw() does not return the pin state but only the result of the read operation of both banks
  - there's currently no way to tell in which bank a given pin is, and digital_read_hw() would read both banks anyway
  - read_cache_invalidated[x] should be more intuitive if 'true means invalidated'

It does not:
  - add a function to base class to map pin to bank (in tca9555 it could be an inline like `uint8_t pin2bank(uint8t pin) {return (pin<8)?0:1}`
  - does not change reset_pin_cache() to force only a single bank
  - does not change tca9555.{h,cpp} to initialize just 2 banks instead of 16 pins for caching
  - does not change tca9555::digital_read_hw() to read a single bank

mobrembski, 2024-12-23:
> Here's my patch. It's currently working in single-bank mode to minimize changes to the rest of the code.
> 
> ```
> git diff
> diff --git a/esphome/components/gpio_expander/cached_gpio.h b/esphome/components/gpio_expander/cached_gpio.h
> index 784c5f0f4..a23eaa47f 100644
> --- a/esphome/components/gpio_expander/cached_gpio.h
> +++ b/esphome/components/gpio_expander/cached_gpio.h
> @@ -11,11 +11,11 @@ namespace gpio_expander {
>  template<typename T, T N> class CachedGpioExpander {
>   public:
>    bool digital_read(T pin) {
> -    if (!this->read_cache_invalidated_[pin]) {
> -      this->read_cache_invalidated_[pin] = true;
> -      return this->digital_read_cache(pin);
> +    if (this->read_cache_invalidated_[0]) { //@@@ 0 instead of pin: single global cache
> +      this->read_cache_invalidated_[0] = false; //@@@ idem
> +      if(!this->digital_read_hw(pin)) return false; // Failed reading
>      }
> -    return this->digital_read_hw(pin);
> +    return this->digital_read_cache(pin);
>    }
> 
>    void digital_write(T pin, bool value) { this->digital_write_hw(pin, value); }
> @@ -27,7 +27,7 @@ template<typename T, T N> class CachedGpioExpander {
> 
>    void reset_pin_cache_() {
>      for (T i = 0; i < N; i++) {
> -      this->read_cache_invalidated_[i] = false;
> +      this->read_cache_invalidated_[i] = true;
>      }
>    }
> ```
> 
> It accounts for:
> 
> * digital_read_hw() does not return the pin state but only the result of the read operation of both banks
> * there's currently no way to tell in which bank a given pin is, and digital_read_hw() would read both banks anyway
> * read_cache_invalidated[x] should be more intuitive if 'true means invalidated'
> 
> It does not:
> 
> * add a function to base class to map pin to bank (in tca9555 it could be an inline like `uint8_t pin2bank(uint8t pin) {return (pin<8)?0:1}`
> * does not change reset_pin_cache() to force only a single bank
> * does not change tca9555.{h,cpp} to initialize just 2 banks instead of 16 pins for caching
> * does not change tca9555::digital_read_hw() to read a single bank
> * does not implement interrupt usage

Thanks for the patch! :)
About single bank support...would it be beneficial in any way?
I was thinking about it when i wrote first version but i cant really find any usage for reading a single bank.
Obviously it will get us faster read speed thus we're reading a single byte instead of 2 bytes but besides this - i'm not sure. Most of GPIO drivers in esphome doesnt support banking.

BTW what board did you guys use? TCA9555 AFAIK isnt very popular, at least not as much as PCF8574, just curious  :)

NdK73, 2024-12-23:
I designed my DomoNode2.1 to use it because of the 16 I/Os with predictable state at powerup (MCP23017, that I tried in DomoNode2.0, was unpredictable and harder to init correctly, like after an ESP reboot w/o external reset).

IIUC cached_gpio was writtten to be shared with other expanders, and supporting multiple banks could theoretically be useful in a more general scenario (maybe some expanders are way slower... think bit-banging or slow serial links). Since it was already in the code I assumed it's something desiderable: I tried to minimize the changes. What I don't understand is the reasoning behind the original code, keeping track of the "cached" status of every GPIO, but that can be seen as a special case of banking, where every GPIO uses its own "bank".
The only reason I could think to handle the two banks separately is if interrupts gets implemented: if the changes are in a single bank, it's possible to slightly reduce I2C traffic. Worth it? I can't judge. But for completeness it could be useful to have cached outputs too (atomically update multiple GPIOs with a single I2C transaction).

freddiemare, 2024-12-23:
@NdK73  grate work Ndk73 just tested and it is working perfectly with my modified version of tca9555

I hacked at it with GPT yesterday and here is my Frankenstein version of the .h and .cpp it is of any use this is now running on my ESP32 S3.

tsa9555.cpp

```
#include "tca9555.h"
#include "esphome/core/log.h"

static const uint8_t TCA9555_INPUT_PORT_REGISTER_0 = 0x00;
static const uint8_t TCA9555_INPUT_PORT_REGISTER_1 = 0x01;
static const uint8_t TCA9555_OUTPUT_PORT_REGISTER_0 = 0x02;
static const uint8_t TCA9555_OUTPUT_PORT_REGISTER_1 = 0x03;
static const uint8_t TCA9555_POLARITY_PORT_0 = 0x04;
static const uint8_t TCA9555_POLARITY_PORT_1 = 0x05;
static const uint8_t TCA9555_CONFIGURATION_PORT_0 = 0x06;
static const uint8_t TCA9555_CONFIGURATION_PORT_1 = 0x07;

namespace esphome {
namespace tca9555 {

static const char *const TAG = "tca9555";

void TCA9555Component::setup() {
  ESP_LOGCONFIG(TAG, "Setting up TCA9555...");
  if (!this->read_gpio_modes_()) {
    this->mark_failed();
    return;
  }
  if (!this->read_gpio_outputs_()) {
    this->mark_failed();
    return;
  }
  if (!this->read_gpio_polarities_()) {
    this->mark_failed();
    return;
  }
}

void TCA9555Component::dump_config() {
  ESP_LOGCONFIG(TAG, "TCA9555:");
  LOG_I2C_DEVICE(this)
  if (this->is_failed()) {
    ESP_LOGE(TAG, "Communication with TCA9555 failed!");
  }
}

void TCA9555Component::pin_mode(uint8_t pin, gpio::Flags flags) {
  if (flags == gpio::FLAG_INPUT) {
    // Set mode mask bit
    this->mode_mask_ |= 1 << pin;
  } else if (flags == gpio::FLAG_OUTPUT) {
    // Clear mode mask bit
    this->mode_mask_ &= ~(1 << pin);
  }
  // Write GPIO to enable input mode
  this->write_gpio_modes_();
}

void TCA9555Component::loop() {
  this->reset_pin_cache_();
}

bool TCA9555Component::read_gpio_outputs_() {
  if (this->is_failed())
    return false;
  uint8_t data[2];
  if (!this->read_bytes(TCA9555_OUTPUT_PORT_REGISTER_0, data, 2)) {
    this->status_set_warning("Failed to read output register");
[... 370 lines cut ...]
      this->read_cache_invalidated_[0] = false; //@@@ idem
      if(!this->digital_read_hw(pin)) return false; // Failed reading
    }
    return this->digital_read_cache(pin);
  }

  void digital_write(T pin, bool value) { this->digital_write_hw(pin, value); }

 protected:
  virtual bool digital_read_hw(T pin) = 0;
  virtual bool digital_read_cache(T pin) = 0;
  virtual void digital_write_hw(T pin, bool value) = 0;

  void reset_pin_cache_() {
    for (T i = 0; i < N; i++) {
      this->read_cache_invalidated_[i] = true;
    }
  }

  std::array<bool, N> read_cache_invalidated_{};
};

}  // namespace gpio_expander
}  // namespace esphome
```

freddiemare, 2024-12-23:
Thank you for the feedback You are most welcome to post a PR @mobrembski
 I was just in a bit of a panic since I developed controllers that use this chip and I'm quite impressed with the performance one can use it for many other functions also but I believe the component needs a tad bit more work to really make use of the full capability of the TCA9555.
 
 Thanks much again.

mobrembski, 2024-12-23:
Okay i think i have a fix with banks support. I will create PR later afer christmas.

NdK73, 2024-12-24:
Wonderful. I'm still inexperienced with all this PR stuff and would
probably take me a lot to do it.
Thanks!

veveritza, 2025-01-31:
I can also confirm the bug. I'm using a TCA9555PWR and can't read inputs.
I've tried @mobrembski 's fix and can confirm it works. Posting with the hope it gets pulled into the main version.

pfefferCode, 2025-02-10:
I have the same problem. 
What do you think, when will you merge the PR? And when will it get to the next release?

mobrembski, 2025-02-10:
Dont ask me, i have no power here :)

@jesserockz - Alerady asked you in PR, but can we merge https://github.com/esphome/esphome/pull/8003 please?

pfefferCode, 2025-02-11:
How can I test the fix? overwriting the according files with the fixed ones does not work because esphome compile replaces them with the original ones.

NdK73, 2025-02-11:
I am using a local override: using the file editor I created esphome/components/gpio_expander/ with __init__.py and cached_gpio.h .
Then added this snippet to my yaml code:
```
external_components:
  - source:
      type: local
      path: components
```
This way the modified gpio_expander changes override the default files.
Still waiting for inclusion in upstream EspHome.

miesch1, 2025-03-02:
@NdK73 and @pfefferCode, thank you! I was pulling my hair out thinking my PCA9555 inputs were somehow getting stuck high. Luckily saw this post which worked immediately.

Just to add to a comment above about why this port expander is preferred, in addition to the predictable state of the I/O at startup, these chips use a Totem-poll output (can both sink and source a large amount of current), as opposed to the PCF8574 which uses a Quasi-output type (can sink but not source a large amount of current). Thank you https://opensprinkler.com/i-o-expander-options/. Plus, this version with built-in I/O pullups is super nice to interface with cheap 817 optocoupler modules (https://www.aliexpress.us/item/3256805259555358.html).

NdK73, 2025-03-20:
Already 3 ESPHome updates and it's not been merged yet :(
Is there someone else to notify?

lk-ek, 2025-04-17:
(crosspost from the PR) 
fyi, I've created a repository so that this branch and fix can be easily used by others until it's finally merged:

https://github.com/lk-ek/esphome-tca9555

The only change in that repository is the addition of the update()-function that simply invalidates the cache and hence allows for timing-critical gpio readings.

To use in your devices, simply add this external component:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/lk-ek/esphome-tca9555
      ref: main
    components: ["tca9555", "gpio_expander"]
```

closed 2025-05-02 by jesserockz