esphome/issues #6439: Flesh writes Wear Leveling

vgdh, 2024-11-17:
### The problem

It is not clear is ESPhome uses Wear Leveling for  **Global Variables** with options: **restore_value: true** and **esp8266_restore_from_flash** or no.
If no it should be doing so. https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/wear-levelling.html
How often can I write into it until the flash will die?

### Which version of ESPHome has the issue?

 2024.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

latest

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
globals:
  - id: saved_position_1
    type: int
    initial_value: '0'
    restore_value: true
```

ssieb, 2024-11-18:
Your link is for ESP32, not ESP8266.  I don't think there is any wear levelling for 8266.  You can still do thousands of writes before the flash will be damaged.  How often do you expect to be writing?

vgdh, 2024-11-18:
> Your link is for ESP32, not ESP8266. I don't think there is any wear leveling for 8266. You can still do thousands of writes before the flash will be damaged. How often do you expect to be writing?

I think about one write every 10 minutes. With a simple math it will be only 69 days.

closed 2025-04-22 by github-actions[bot]