esphome/issues #6441: Different indicator bytes in plaintext and encrypted API. Differing naming schemes. Lacking API documentation.

MariusVB, 2024-11-18:
### The problem

Hello,

I've been implementing a Rust library which can interact with ESPHome devices through the protocol buffers API, as I want to use ESPHome without Home Assistant. It was a really painstaking process, since the API is really not documented at all. When reading the API (docs)[https://esphome.io/components/api.html], I got the impression that it would be relatively easy, but I had to look through this source code and the source code of the Python (library)[https://github.com/esphome/aioesphomeapi] to get it working.

One thing that really frustrated me was the `BAD_INDICATOR` error I kept seeing in the logs. I looked through the ESPHome source code and found that this happened if the first byte was not 0x01. But even setting that I still got the error. Turns out the indicator byte is different with and without encryption, which is not at all documented anywhere. Is this on purpose?
I got the impression that whatever interacted with the ESPHome device would be told in the `ConnectResponse` if the device wanted an encryption key, but instead you are faced with this obscure error. Also, in the `api.proto` file the encryption key is called a password, whereas in the ESPHome API docs it's called an encryption key, further adding to the confusion.

All in all, I just think that the API documentation should be expanded (I'm willing to help), and that some of the naming should be streamlined. It was extremely confusing and frustrating to work with, which does not exactly inspire new developers to work with and extend the ESPHome ecosystem.

https://github.com/esphome/esphome/blob/6e41c22e9d7f777a5c556725b1a54cf33a669e39/esphome/components/api/api_frame_helper.cpp#L212

https://github.com/esphome/esphome/blob/6e41c22e9d7f777a5c556725b1a54cf33a669e39/esphome/components/api/api_frame_helper.cpp#L819

### Which version of ESPHome has the issue?

2024.10

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

MariusVB, 2025-03-19:
Still an issue

ssieb, 2025-03-19:
There isn't really any documentation for the API.  You are certainly welcome to make that if you want.  Developer docs are being worked on at https://github.com/esphome/developers.esphome.io.  You can come chat about it in the `developer-documentation` channel on the discord server.

bdraco, 2025-05-17:
I documented it in https://github.com/esphome/esphome/blob/reuse_buffer/esphome/components/api/protocol_details.md

closed 2025-05-18 by jesserockz

bdraco, 2025-05-18:
Its now live at https://developers.esphome.io/architecture/api/protocol_details/

MariusVB, 2025-05-19:
Beautiful!! Thanks a lot for this