esphome/issues #6442: ILI9486 washed out colors

yazz007, 2024-11-19:
### The problem

I have Waveshare 4inch Touch LCD Shield with ILI9486 driver. It seems to work with ESPhome but the colors are not totally right. Looks like washed out or missed:
[image]

The test should draw the image like this one:
[image]

I tried, I guess, almost all configuration settings and this is the best what I can do.
Any clue what is wrong?

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP32 S2 mini with 2MB PSRAM

### Component causing the issue

display/ili9xxx

### Example YAML snippet

```yaml
display:
  - platform: ili9xxx
    model: ili9486
    dc_pin: GPIO4
    cs_pin: GPIO12
    reset_pin: GPIO2
    data_rate: 20Mhz
    dimensions: 480x320
    auto_clear_enabled: False
    update_interval: never
    id: tft
    invert_colors: false
    show_test_card: true
    transform:
      swap_xy: true
      mirror_x: true
      mirror_y: true
```

### Anything in the logs that might be useful for us?

```txt
[08:44:24][C][spi:064]: SPI bus:
[08:44:24][C][spi:065]:   CLK Pin: GPIO7
[08:44:24][C][spi:066]:   SDI Pin: GPIO9
[08:44:24][C][spi:067]:   SDO Pin: GPIO11
[08:44:24][C][spi:072]:   Using HW SPI: SPI
[08:44:24][C][ledc.output:180]: LEDC Output:
[08:44:24][C][ledc.output:181]:   Pin GPIO14
[08:44:24][C][ledc.output:182]:   LEDC Channel: 0
[08:44:24][C][ledc.output:183]:   PWM Frequency: 19531.0 Hz
[08:44:24][C][ledc.output:184]:   Phase angle: 0.0°
[08:44:24][C][ledc.output:185]:   Bit depth: 11
[08:44:24][C][ili9xxx:094]: ili9xxx
[08:44:24][C][ili9xxx:094]:   Rotations: 0 °
[08:44:24][C][ili9xxx:094]:   Dimensions: 480px x 320px
[08:44:24][C][ili9xxx:095]:   Width Offset: 0
[08:44:24][C][ili9xxx:096]:   Height Offset: 0
[08:44:24][C][ili9xxx:102]:   Color mode: 16bit
[08:44:24][C][ili9xxx:111]:   Data rate: 20MHz
[08:44:24][C][ili9xxx:113]:   Reset Pin: GPIO2
[08:44:24][C][ili9xxx:114]:   CS Pin: GPIO12
[08:44:24][C][ili9xxx:115]:   DC Pin: GPIO4
[08:44:24][C][ili9xxx:117]:   Color order: BGR
[08:44:24][C][ili9xxx:118]:   Swap_xy: YES
[08:44:24][C][ili9xxx:119]:   Mirror_x: YES
[08:44:24][C][ili9xxx:120]:   Mirror_y: YES
[08:44:24][C][ili9xxx:121]:   Invert colors: NO
[08:44:24][C][ili9xxx:126]:   Update Interval: never
```

clydebarrow, 2024-11-21:
try adding `pixel_mode: 18bit` to the display config.

yazz007, 2024-11-21:
Thanks, I tried that as well, a bit smoother but the same :(

spike11302000, 2025-03-06:
@yazz007 Im running into the same issue with the same display did you ever end up finding a solution?