esphome/issues #6443: ble_client.connect not working

chedda61, 2024-11-19:
### The problem

I'm trying to manually connect to a Seeed XIAO nrf52840 via an ESP32. I tried with a switch and a button as described here:
https://esphome.io/components/ble_client.html#ble-client-connect-action
The ESP32 attempts the connection but nothing happens afterwards.

**However**, setting the ble_client to autoconnect successfully connects to the device, but this is not what I want.

### Which version of ESPHome has the issue?

2024.7.3; 2024.10.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.11.1

### What platform are you using?

ESP32

### Board

ESP32F; Firebeetle

### Component causing the issue

ble_client

### Example YAML snippet

```yaml
esp32_ble_tracker:

ble_client:
  - mac_address: XX:XX:XX:XX:XX:XX
    id: nrf
    auto_connect: false

switch:
  - platform: template
    name: "NRF"
    turn_on_action:
      - esp32_ble_tracker.stop_scan:
      - ble_client.connect: nrf
    turn_off_action:
      - ble_client.disconnect: nrf
      - esp32_ble_tracker.start_scan:
```

### Anything in the logs that might be useful for us?

```txt
When the button is pressed, all I get is this log (logger is set to DEBUG):
With Arduino Framework:
  0x00 Attempting BLE connection
With esp-idf:
  0x00 Attempting BLE connection
  [BTU_TASK]: W BT_APPL: BTA got unregistered event id 31
```

### Additional information

When autoconnect is true AND a ble_client.connect is defined on a button/switch, autoconnect won't work

Archomeda, 2025-01-31:
I seem to encounter the same issue. Mine is just stuck at "Attempting BLE connection".
I've tried both frameworks: same result. I have tried both stopping the scan beforehand, and not stopping it: made no difference. Also tried setting auto_connect to true and false: also no difference.

I don't know how to work around this issue. I don't want the BLE component to be connected 24/7, as that means that the device is no longer available for other connections. I basically want to periodically poll its characteristic values, and also able to manipulate those values. For more context: the devices in question are Valve Index base stations (lighthouse v2).

alexsavuto, 2025-05-01:
I'm in the same situation — the moment I use esp32_ble_tracker.stop_scan, the ble_client crashes. But while esp32_ble_tracker occupies the BLE stack, it causes connection failures. It's a vicious cycle.
[https://github.com/esphome/issues/issues/6983](url)

Scob166, 2025-12-03:
Is this issue ongoing or solved?

poppopdockery, 2026-09-23:
I hit the same "status 133 / connect never happens" behaviour driving a BLE tilt motor, and found what looks like the root cause of why it only works **once**.

**Symptom:** `ble_client.connect` (fired from a script/button) works the first time or two, but after any failed/timed-out attempt it stops connecting entirely — the action fires but no "Connecting" line is ever logged again. A reboot temporarily fixes it.

**Root cause (reading `esphome/components/ble_client/automation.h`):** `BLEClientConnectAction::play_complex()` begins with:

```cpp
if (this->num_running_ != 0) {
  this->stop_complex();
  return;
}
this->num_running_++;
```

`num_running_` is only decremented back to 0 via `play_next_` on `ESP_GATTC_SEARCH_CMPL_EVT` (success) or by `stop_complex()` on `ESP_GATTC_DISCONNECT_EVT`. But a **failed open** (`ESP_GATTC_OPEN_EVT` with `status != ESP_GATT_OK`) goes through `set_idle_()` in `ble_client_base.cpp` and does **not** emit `ESP_GATTC_DISCONNECT_EVT`. So after a failed connect, `num_running_` is stuck at 1, and every subsequent invocation hits the `if (num_running_ != 0)` guard, calls `stop_complex()`, and returns without ever calling `connect()`. That matches the "works until the first failure, then dead until reboot" behaviour exactly.

**Workaround that's been 100% reliable for me** — bypass the action, call the client directly from a lambda, and reset the client's state first if it isn't IDLE:

```yaml
script:
  - id: blind_send
    mode: restart
    then:
      - repeat:
          count: 3
          then:
            - if:
                condition:
                  lambda: 'return !id(cmd_done);'   # your own "did it land" flag
                then:
                  - lambda: |-
                      auto st = id(my_ble_client).state();
                      if (st != esp32_ble_tracker::ClientState::IDLE) {
                        id(my_ble_client).unconditional_disconnect();
                        id(my_ble_client).set_state(esp32_ble_tracker::ClientState::IDLE);
                      }
                  - delay: 300ms
                  - lambda: 'id(my_ble_client).connect();'
                  - wait_until:
                      condition:
                        lambda: 'return id(cmd_done) || id(my_ble_client).state() == esp32_ble_tracker::ClientState::IDLE;'
                      timeout: 12s
                  - if:
                      condition:
                        lambda: 'return !id(cmd_done);'
                      then:
                        - delay: 1s
```

`unconditional_disconnect()` + `set_state(IDLE)` clears the stuck client, and calling `connect()` on the client object avoids the action's `num_running_` guard entirely. Retrying 3× also rides out genuinely-busy targets.

If a maintainer agrees the fix belongs in the action itself, decrementing/clearing `num_running_` on a failed `OPEN_EVT` (or treating a return-to-IDLE as a terminal state for the action) would fix it upstream. Happy to test a patch.

Environment: ESP32 (esp-idf), ESPHome 2026.9.0, BLE device with a random static address.

GovernmentHack, 2026-10-03:
> Is this issue ongoing or solved?

I have the same issue with version `2026.9.1` on an esp32-s3

```
[11:04:18.546][I][esp32_ble_client:148]: [0] [FC:D2:24:F7:1E:E0] 0x00 Connecting
[11:04:18.551][D][esp32_ble_tracker:145]: connecting: 1, discovered: 0, disconnecting: 0, active: 0
[11:04:38.559][D][esp32_ble_client:424]: [0] [FC:D2:24:F7:1E:E0] ESP_GATTC_DISCONNECT_EVT, reason 0x100
[11:04:38.608][D][esp32_ble_client:256]: [0] [FC:D2:24:F7:1E:E0] ESP_GATTC_OPEN_EVT
[11:04:38.612][E][esp32_ble_client:369]: [0] [FC:D2:24:F7:1E:E0] ESP_GATTC_OPEN_EVT in DISCONNECTING state (status=133)
[11:04:38.612][W][esp32_ble_client:268]: [0] [FC:D2:24:F7:1E:E0] Connection open error, status=133
[11:04:38.612][D][esp32_ble_tracker:145]: connecting: 0, discovered: 0, disconnecting: 0, active: 0
```