esphome/issues #6445: ESP32 S2 I2c broken using ESPHome

robnrie, 2024-11-19:
### The problem

starting a few days ago all my ESP32 S2 mini boards that ran correctly now have issues with any i2c device attached they can be seen but will not communicate but on my ESP32 S3 I do not see the same problem

### Which version of ESPHome has the issue?

2024.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2024.11.2

### What platform are you using?

ESP32

### Board

ESP32 S2 mini

### Component causing the issue

i2c any

### Example YAML snippet

```yaml
esphome:
  name: s2-fountain
  friendly_name: S2 Fountain

esp32:
  board: lolin_s2_mini
  variant: esp32s2
  framework:
    type: esp-idf
#    version: 5.3.0  #   5.3.0 is currently the latest 
#    platform_version: 6.8.1  # https://github.com/platformio/platform-espressif32/releases/

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "xwjWUOxVjIK8qDUmRwxyWK/LI7OyC9bfmNM6TQhflHw="

ota:
  - platform: esphome
    password: "cf412b6a06722b86fcc01e4e64aa211a"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "S2-Fountain Fallback Hotspot"
    password: "uwO7pHjmXrhw"

captive_portal:

i2c:
  sda: GPIO33
  scl: GPIO35
[... 161 lines cut ...]
[18:08:36][E][bh1750.sensor:121]: Communication with BH1750 failed!
[18:08:36][C][bh1750.sensor:124]:   Update Interval: 60.0s
[18:08:36][E][component:082]:   Component bh1750.sensor is marked FAILED
[18:08:36][C][captive_portal:089]: Captive Portal:
[18:08:36][C][mdns:116]: mDNS:
[18:08:36][C][mdns:117]:   Hostname: s2-fountain
[18:08:36][C][esphome.ota:073]: Over-The-Air updates:
[18:08:36][C][esphome.ota:074]:   Address: s2-fountain.local:3232
[18:08:36][C][esphome.ota:075]:   Version: 2
[18:08:36][C][esphome.ota:078]:   Password configured
[18:08:36][C][safe_mode:018]: Safe Mode:
[18:08:36][C][safe_mode:020]:   Boot considered successful after 60 seconds
[18:08:36][C][safe_mode:021]:   Invoke after 10 boot attempts
[18:08:36][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[18:08:36][C][api:140]: API Server:
[18:08:36][C][api:141]:   Address: s2-fountain.local:6053
[18:08:36][C][api:143]:   Using noise encryption: YES
[18:08:36][C][wifi_signal.sensor:010]: WiFi Signal 'WiFi Signal Sensor'
[18:08:36][C][wifi_signal.sensor:010]:   Device Class: 'signal_strength'
[18:08:36][C][wifi_signal.sensor:010]:   State Class: 'measurement'
[18:08:36][C][wifi_signal.sensor:010]:   Unit of Measurement: 'dBm'
[18:08:36][C][wifi_signal.sensor:010]:   Accuracy Decimals: 0
[18:08:58][D][sensor:094]: 'WiFi Signal Sensor': Sending state -61.00000 dBm with 0 decimals of accuracy
[18:09:04][D][text_sensor:064]: 'BME680 IAQ Classification': Sending state 'unknown'
```

CarlosGS, 2024-11-24:
Same issue here, in my case for `platform: scd4x` when updating from 2024.9.1 to 2024.11.1.
The I2C device is detected but communication fails with this ESPHome version.

```yaml
esphome:
  name: co2test

esp32:
  board: esp32-s2-saola-1
  framework:
    type: esp-idf

logger:

api:
  encryption:
    key: !secret encryption_key
  services:
    - service: calibrate_co2_value
      variables:
        co2_ppm: int
      then:
      - scd4x.perform_forced_calibration:
          value: !lambda 'return co2_ppm;'
          id: my_scd41

ota:
  platform: esphome
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

i2c:
  sda: 33
  scl: 35

sensor:
  - platform: scd4x
    id: my_scd41
    automatic_self_calibration: false
    measurement_mode: low_power_periodic
    ambient_pressure_compensation: 913 # mBar
    temperature_offset: 0
    update_interval: 30s
    co2:
      name: "Sensirion CO2"
    temperature:
      name: "Sensirion Temperature"
    humidity:
      name: "Sensirion Humidity"
```

```
[13:04:27][I][app:100]: ESPHome version 2024.11.1 compiled on Nov 24 2024, 11:59:54
[13:04:27][C][wifi:600]: WiFi:
[13:04:27][C][wifi:428]:   Local MAC: [redacted]
[13:04:27][C][wifi:433]:   SSID: [redacted]
[13:04:27][C][wifi:436]:   IP Address: 192.168.1.32
[13:04:27][C][wifi:440]:   BSSID: [redacted]
[13:04:27][C][wifi:441]:   Hostname: 'co2test'
[13:04:27][C][wifi:443]:   Signal strength: -10 dB ▂▄▆█
[13:04:27][C][wifi:447]:   Channel: 11
[13:04:27][C][wifi:448]:   Subnet: 255.255.255.0
[13:04:27][C][wifi:449]:   Gateway: 192.168.1.1
[13:04:27][C][wifi:450]:   DNS1: 192.168.1.1
[13:04:27][C][wifi:451]:   DNS2: 0.0.0.0
[13:04:27][C][logger:185]: Logger:
[13:04:27][C][logger:186]:   Level: DEBUG
[13:04:27][C][logger:188]:   Log Baud Rate: 115200
[13:04:27][C][logger:189]:   Hardware UART: USB_CDC
[13:04:27][C][i2c.idf:075]: I2C Bus:
[13:04:27][C][i2c.idf:076]:   SDA Pin: GPIO33
[13:04:27][C][i2c.idf:077]:   SCL Pin: GPIO35
[13:04:27][C][i2c.idf:078]:   Frequency: 50000 Hz
[13:04:27][C][i2c.idf:084]:   Recovery: bus successfully recovered
[13:04:27][I][i2c.idf:094]: Results from i2c bus scan:
[13:04:27][I][i2c.idf:100]: Found i2c device at address 0x62
[13:04:27][C][scd4x:094]: scd4x:
[13:04:27][C][scd4x:095]:   Address: 0x62
[13:04:27][W][scd4x:099]: Communication failed! Is the sensor connected?
[13:04:27][C][scd4x:112]:   Automatic self calibration: OFF
[13:04:27][C][scd4x:118]:   Altitude compensation disabled
[13:04:27][C][scd4x:119]:   Ambient pressure compensation: 913mBar
[13:04:27][C][scd4x:130]:   Measurement mode: low power periodic (30s)
[13:04:27][C][scd4x:139]:   Temperature offset: 0.00 °C
[13:04:27][C][scd4x:140]:   Update Interval: 30.0s
[13:04:27][C][scd4x:141]:   CO2 'Sensirion CO2'
[13:04:27][C][scd4x:141]:     Device Class: 'carbon_dioxide'
[13:04:27][C][scd4x:141]:     State Class: 'measurement'
[13:04:27][C][scd4x:141]:     Unit of Measurement: 'ppm'
[13:04:27][C][scd4x:141]:     Accuracy Decimals: 0
[13:04:27][C][scd4x:141]:     Icon: 'mdi:molecule-co2'
[13:04:27][C][scd4x:142]:   Temperature 'Sensirion Temperature'
[13:04:27][C][scd4x:142]:     Device Class: 'temperature'
[13:04:27][C][scd4x:142]:     State Class: 'measurement'
[13:04:27][C][scd4x:142]:     Unit of Measurement: '°C'
[13:04:27][C][scd4x:142]:     Accuracy Decimals: 2
[13:04:27][C][scd4x:142]:     Icon: 'mdi:thermometer'
[13:04:27][C][scd4x:143]:   Humidity 'Sensirion Humidity'
[13:04:27][C][scd4x:143]:     Device Class: 'humidity'
[13:04:27][C][scd4x:143]:     State Class: 'measurement'
[13:04:27][C][scd4x:143]:     Unit of Measurement: '%'
[13:04:27][C][scd4x:143]:     Accuracy Decimals: 2
[13:04:27][C][scd4x:143]:     Icon: 'mdi:water-percent'
[13:04:27][E][component:082]:   Component scd4x.sensor is marked FAILED
[13:04:27][C][mdns:116]: mDNS:
[13:04:27][C][mdns:117]:   Hostname: co2test
[13:04:27][C][esphome.ota:073]: Over-The-Air updates:
[13:04:27][C][esphome.ota:074]:   Address: co2test.local:3232
[13:04:27][C][esphome.ota:075]:   Version: 2
[13:04:27][C][esphome.ota:078]:   Password configured
[13:04:27][C][safe_mode:018]: Safe Mode:
[13:04:27][C][safe_mode:020]:   Boot considered successful after 60 seconds
[13:04:27][C][safe_mode:021]:   Invoke after 10 boot attempts
[13:04:27][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[13:04:27][C][api:140]: API Server:
[13:04:27][C][api:141]:   Address: co2test.local:6053
[13:04:27][C][api:143]:   Using noise encryption: YES
```

I notice we are both using ESP-IDF framework, let's try switching to Arduino and see if it still happens.

CarlosGS, 2024-11-24:
Indeed it works OK with `framework: type: arduino`, so this must be related to some I2C change in the ESP-IDF implementation :thinking:

dab0g, 2024-11-28:
That's what my logic analyzer showed me.The framework is esp-idf, the frequency is set to 100kHz. The screenshot was taken during a tire scan. Initially, the speed is set correctly, because as a rule, bus scanning works, and then at some point in time the bus frequency decreases by 80 times!
[image]

whatthebottleneck, 2024-12-13:
ESP-IDF version 5.3.2 appears to fix a number of I2C related issues, for example:

[https://github.com/espressif/esp-idf/commit/deb91c7abe8b1fafc5e9410fd2fdfd2aeebcea57](url)
[https://github.com/espressif/esp-idf/commit/189db78bec83ada18ddced7f24a63ac3b87b48ec](url)
[https://github.com/espressif/esp-idf/commit/7be31411958ce83437ee83ec4d4f72508325e165](url)
[https://github.com/espressif/esp-idf/commit/403bd86a213fb40158d6df65ccc69ddbbcd838f3](url)

However, 5.3.1 is the latest version that currently builds successfully in esphome (related: #6535). Does anybody know how to get 5.3.2 to build?

swifty99, 2024-12-22:
ESP32 framework for S2 derivates is addicted to I2S issues for a while.

I temporarily fixed my sensor with this:

```
i2c:
  - id: busi2c
    sda: GPIO13
    scl: GPIO14
    scan: true
    frequency: 800kHz
```

By overspending the bus a lot, the division by 80 will be a reasonable speed in the end.

sbrk-r, 2025-01-10:
I'm having the same issue, still broken with 5.3.1/6.9.0, the newest option in platformio, it seems.