esphome/issues #6446: When specifying AUTO_LOAD = ["animation"] the error `Failed config animation: None`

andrewjswan, 2024-11-20:
### The problem

When specifying in component:
```
AUTO_LOAD = ["animation"]
```
The error:
```
Failed config

animation: None
  
  'id' is a required option for [animation].
  []
required key not provided
```
If you add a file to the config that uses this string component:
```
animation:
  - file: images/1pixel.gif
    id: empty_animation
```
the build goes without problems. 
Moreover, if you specify `AUTO_LOAD = ["image"]` then there is no error and the `image` component gets into the config.

### Which version of ESPHome has the issue?

2024.10.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

animation

### Example YAML snippet

```yaml
AUTO_LOAD = ["image", "animation"]
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.10.3
INFO Reading configuration /config/esphome/test.yaml...
INFO Detected timezone 'Europe/Kiev'
Failed config

animation: None
  
  'id' is a required option for [animation].
  []
required key not provided
```

andrewjswan, 2024-11-20:
There is a similar problem with `http_request` in the Arduino framework, but there the error says that SSL needs to be disabled.

ssieb, 2024-11-20:
This is not a yaml error.  You must be making a component.  What are you trying to do?

andrewjswan, 2024-11-21:
@ssieb 
Here is the component: https://github.com/andrewjswan/EspHoMaTriXv2/tree/dependencies

After the release of 2023.7.0, this component stopped building because the image and animation were not loaded because they were not specified in Yaml.
Issue: 
- https://github.com/lubeda/EspHoMaTriXv2/issues/62

As a temporary solution, the following code was added to Yaml:
https://github.com/andrewjswan/EspHoMaTriXv2/blob/2ab564c7c3964c8d8ce0f2a0caed5f6cf8db4366/tests/ulanzi-test.yaml#L248-L254
```Yaml
animation:
  - id: animation_black
    file: 1pixel.gif

image:
  - id: image_black
    file: 1pixel.gif
```
But we have the ability to specify which components need to be loaded even if they are not specified (AUTO_LOAD).
I add these components:
https://github.com/andrewjswan/EspHoMaTriXv2/blob/2ab564c7c3964c8d8ce0f2a0caed5f6cf8db4366/components/ehmtxv2/__init__.py#L22

I'm removing the temporary solution:
https://github.com/andrewjswan/EspHoMaTriXv2/blob/860664da99dec15faeddfdc585a2dea984e0dc1f/tests/ulanzi-test.yaml#L248-L254

And get Error:
https://github.com/andrewjswan/EspHoMaTriXv2/actions/runs/11949830185/job/33310085308#step:3:843

If I re add animation workaround:
https://github.com/andrewjswan/EspHoMaTriXv2/blob/92596c4a682bf7bc73fad92b73025d7064de97f8/tests/ulanzi-test.yaml#L248-L254

Build without Error:
https://github.com/andrewjswan/EspHoMaTriXv2/actions/runs/11949866274/job/33310196574

andker87, 2024-12-06:
I also have the same problem, I am testing with the ulanzi display and the icons/animations are not working.

closed 2025-01-13 by jesserockz