esphome/issues #6451: http_request compilation error if called from a script with parameters

ronengr, 2024-11-21:
### The problem

Compilation fails if calling `http_request.get` from a script that has parameters.

This used to work in 2024.10

Might be related to esphome/esphome#7696

### Which version of ESPHome has the issue?

2024.11.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

http_request

### Example YAML snippet

```yaml
script:
  - id: test_script
    parameters:
      arg1: string
    then:
      - http_request.get:
          url: https://esphome.io

esphome:
  name: test
  on_boot:
    then:
      - script.execute:
            id: test_script
            arg1: "test"

wifi:
  ssid: MySSID

esp32:
  board: esp32dev

http_request:
  id: http_request_data
  verify_ssl: false
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome.h:5,
                 from src/main.cpp:3:
src/esphome/components/http_request/http_request.h: In instantiation of 'void esphome::http_request::HttpRequestSendAction<Ts>::play(Ts ...) [with Ts = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >}]':
src/esphome/components/http_request/http_request.h:166:8:   required from here
src/esphome/components/http_request/http_request.h:192:9: error: no matching function for call to 'esphome::Trigger<>::trigger(std::__cxx11::basic_string<char>&)'
         trigger->trigger(x...);
         ^~~~~~~
In file included from src/esphome/components/http_request/http_request.h:11,
                 from src/esphome.h:5,
                 from src/main.cpp:3:
src/esphome/core/automation.h:95:8: note: candidate: 'void esphome::Trigger<Ts>::trigger(Ts ...) [with Ts = {}]'
   void trigger(Ts... x) {
        ^~~~~~~
src/esphome/core/automation.h:95:8: note:   candidate expects 0 arguments, 1 provided
*** [.pioenvs/test/src/main.cpp.o] Error 1
```

kevdel, 2024-11-21:
Similar output (float instead of string http post) with the following yaml

```
sensor:
  - platform: dallas_temp
    update_interval: 5s
    address: 0x0700000004612c28
    id: sens_0
    name: "Temperature 1"
    filters:
       - lambda: return x * (9.0/5.0) + 32.0;
    on_value:
       then:
        - http_request.post:
            url: http://192.168.2.2:8086/write?db=sensors
            body: !lambda |-
              char buf[64];
              sprintf(buf, "%0.1f", x);
              return ((std::string) "temperature.1,location=leitrim value="+buf).c_str();
```

```
In file included from src/esphome.h:20,
                 from src/main.cpp:3:
src/esphome/components/http_request/http_request.h: In instantiation of 'void esphome::http_request::HttpRequestSendAction<Ts>::play(Ts ...) [with Ts = {float}]':
src/esphome/components/http_request/http_request.h:166:8:   required from here
src/esphome/components/http_request/http_request.h:192:25: error: no matching function for call to 'esphome::Trigger<>::trigger(float&)'
  192 |         trigger->trigger(x...);
      |         ~~~~~~~~~~~~~~~~^~~~~~
In file included from src/esphome/components/api/api_server.h:9,
                 from src/esphome/components/api/api_connection.h:8,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/automation.h:95:8: note: candidate: 'void esphome::Trigger<Ts>::trigger(Ts ...) [with Ts = {}]'
   95 |   void trigger(Ts... x) {
      |        ^~~~~~~
src/esphome/core/automation.h:95:8: note:   candidate expects 0 arguments, 1 provided
*** [.pioenvs/clare/src/main.cpp.o] Error 1
========================= [FAILED] Took 30.16 seconds =========================
```

clydebarrow, 2024-11-21:
Fixed with https://github.com/esphome/esphome/pull/7790

```
external_components:
  - source: github://pr#7790
    components: [http_request]
```