esphome/issues #6454: Potential Risk in ESPHome OTA Upgrade via Web Interface

routeroscn, 2024-11-21:
### The problem

I’ve noticed a potential risk in the OTA upgrade process via the ESPHome Web interface. Specifically, if the network signal is unstable, there is a risk of multiple submissions occurring. This happens when the user submits an OTA request, but due to network issues, the system doesn't receive a response within a certain period, leading to duplicate submissions.

While the interface might show a success message after the submission, in practice, this can result in a damaged system. I have verified this issue in real-world scenarios.

To address this, I made some adjustments to the submission code, which I believe would help mitigate the risk. These changes are mostly related to the www.js file, which handles the web interface’s OTA submission behavior. I would recommend looking into this area for improvement in future versions of ESPHome.

Additionally, the OTA functionality via the hotspot web interface also faces the same issue but with greater consequences, as the risk of damaging the system is even higher in this scenario, especially when the device is in a critical state during an upgrade.

Here is the modified code I used to address the issue:

<h2>OTA Update</h2>
<form method="POST" action="${r}/update" enctype="multipart/form-data" id="otaForm" onsubmit="disableButton();">
    <input class="btn" type="file" name="update">
    <input class="btn" type="submit" value="Update" id="submitBtn">
</form>

<script>
  function disableButton() {
    // Disable the submit button
    document.getElementById('submitBtn').disabled = true;
    
    // Change the button text to indicate update is in progress
    document.getElementById('submitBtn').value = "Updating...";
  }
</script>

I hope this issue can be addressed in upcoming releases of ESPHome to improve the reliability and safety of the OTA process.

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

www.js

### Example YAML snippet

```yaml
..
```

### Anything in the logs that might be useful for us?

```txt
..
```

### Additional information

..

routeroscn, 2024-11-21:
`<h2>OTA Update</h2>
<form method="POST" action="${r}/update" enctype="multipart/form-data" id="otaForm" onsubmit="disableButton();">
    <input class="btn" type="file" name="update">
    <input class="btn" type="submit" value="Update" id="submitBtn">
</form>

<script>
  function disableButton() {
    document.getElementById('submitBtn').disabled = true;
    document.getElementById('submitBtn').value = "Updating...";
  }
</script>`

routeroscn, 2024-11-21:
**h2>OTA Update/h2>
form method="POST" action="${r}/update" enctype="multipart/form-data" id="otaForm" onsubmit="disableButton();">
    input class="btn" type="file" name="update">
    input class="btn" type="submit" value="Update" id="submitBtn">
/form>

script>
  function disableButton() {
    document.getElementById('submitBtn').disabled = true;
    document.getElementById('submitBtn').value = "Updating...";
  }
/script>**

[test.txt](https://github.com/user-attachments/files/17841606/test.txt)

swoboda1337, 2024-12-08:
Can create a PR here: https://github.com/esphome/dashboard

routeroscn, 2024-12-09:
Don't know how to use GitHub.