esphome/issues #6456: ESP32-S3-DevKitC-1 N16R8 is really sluggish, wifi problems

rmeissn, 2024-11-21:
### The problem

I was able to bring up the ESP32-S3-DevKitC-1 N16R8 using esphome 2024.11.
Unfortunately, the controller appears to be unusually sluggish, and I can’t determine the cause. My board takes anywhere from 10 seconds to 5 minutes to connect to Wi-Fi after starting up. It is supposed to report the Wi-Fi signal strength every 3 seconds, but this reporting is very inconsistent. It occasionally pauses completely, then recovers by outputting 10–20 updates at once before stopping again. This behaviour seems tied to Wi-Fi connectivity - when I’m close to my access point, it works better, but it deteriorates when I’m about 5 meters away, which isn't a distance. All other devices connect just fine to my AP, thus I guess it's not related to my AP.

### Which version of ESPHome has the issue?

2024.11.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3-DevKitC-1 N16R8

### Example YAML snippet

```yaml
esphome:
  name: test
  friendly_name: test
  name_add_mac_suffix: true
  min_version: 2024.9.0
  platformio_options:
    board_build.flash_mode: dio 
  on_boot: 
    then:
      - light.turn_on:
          id: rgb_led # must be physically connected on the dev board
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB  
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_ESP32_S3_BOX_BOARD: "y"
  
psram:
  mode: octal
  speed: 80MHz

api:

logger:
  level: debug

ota:
  platform: esphome

wifi:
  ssid: ...
  password: ...
  #ap:
  fast_connect: true
  #manual_ip: # setting an ip manually helped in connecting to the controller
    #static_ip: 192.168.0.203
    #gateway: 192.168.0.1
[... 228 lines cut ...]
[09:35:43][D][esp-idf:000][wifi]: Set ps type: 0
[09:35:43]
[09:35:43][D][esp-idf:000][wifi]: 
[09:35:43]
[09:35:43][W][component:157]: Component wifi set Warning flag: associating to network
[09:35:43][D][esp-idf:000][wifi]: I (1277) wifi:
[09:35:43][D][esp-idf:000][wifi]: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
[09:35:43][D][esp-idf:000][wifi]: 
[09:35:43]
[09:35:43][D][esp-idf:000][wifi]: I (1671) wifi:
[09:35:43][D][esp-idf:000][wifi]: state: init -> auth (b0)
[09:35:43][D][esp-idf:000][wifi]: 
[09:35:43]
[09:35:44][D][esp-idf:000][wifi]: I (2673) wifi:
[09:35:44][D][esp-idf:000][wifi]: state: auth -> init (200)
[09:35:44][D][esp-idf:000][wifi]: 
[09:35:44]
[09:35:44][D][esp-idf:000][wifi]: I (2675) wifi:
[09:35:44][D][esp-idf:000][wifi]: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
[09:35:44][D][esp-idf:000][wifi]: 
[09:35:44]
[09:35:44][W][wifi_esp32:704]: Event: Disconnected ssid='PYUR CB8FB' bssid=[redacted] reason='Auth Expired'
[09:35:44][W][wifi:653]: Error while connecting to network.
```
```

rmeissn, 2024-11-21:
I've tested two (of three) identical boards so far. The second board seems to be even worse.
I've noticed that the above issue mainly arises when the board is connected to a breadboard, with nothing else than the devkit on the breadboard (which is correctly positioned and connected). When the devkit is not on the breadboard, it connects to my wifi, but still seems sluggish and sometimes the above error arises.

Is there some test I can execute to dig to the root cause of this issue (which might be the wifi antenna)?

rmeissn, 2024-11-21:
I've tested three of my three identical boards so far. The second board seems to be even worse, the third one about the same as the first one.
I've noticed that the above issue mainly arises when the board is connected to a breadboard, with nothing else than the devkit on the breadboard (which is correctly positioned and connected). When the devkit is not on the breadboard, it connects to my wifi, but still seems sluggish and sometimes the above error arises.

Furthermore, I noticed that when wifi `output_power: 8.5db` is set, the whole board works much more reliably and is more responsive. I took this from https://github.com/esphome/issues/issues/4893 and got not Ubiquity network devices.

Is there some test I can execute to dig to the root cause of this issue (which might be related to the wifi antenna or interference thereof)?

rvalitov, 2024-11-22:
Seems to be an issue with the board. I have the same board S3 and I use it to control Bluetooth devices. It works fine, no issues with WiFi or whatsoever. But I use the `next` branch of ESPHome. Perhaps, you should try that.

rmeissn, 2024-11-24:
The wifi `output_power: 8.5db` addition seems to have fixed all my of boards. At least these reliably report logging statements and reliably connect to Home Assistant through the API (which wasn't possible before).

Still, my boards seem to show problems. Code (for a MicroWakeWord Voice Assistant), that seems runs fine for other users, causes my boards to reboot. Maybe I've simply received bad boards....

@rvalitov Are you talking about next of esphome or next/latest for esp-idf? How do you specify next for esphome? I'm using the docker container and there are only "latest" and "dev" builds available.

rvalitov, 2024-11-24:
@rmeissn sorry, I meant the `dev` build of ESPHome. Below is a part of the YAML config I use for S3:
```yaml
esp32:
    board: esp32-s3-devkitc-1
    framework:
        type: esp-idf
        version: 5.3.1
        platform_version: 6.9.0
        sdkconfig_options:
          CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
          CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
          CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"    
```

ssieb, 2024-11-25:
Are you using the same config as those other users?  Watch the serial logs to find out why it's crashing.  You might need a USB TTL uart adapter for that.  Come ask in the voice assist channel on the esphome discord server.

closed 2024-11-25 by ssieb

rvalitov, 2024-12-19:
1. From what I can see in reviews and comments to S3 and other boards in online shops that many boards are produced in China and have defects. As a result for some users everything works perfectly but others face strange issues.

2. Another problem is that people just use the default esphome configuration and not the one posted here and there which is confirmed to be functional.

3. Finally S3 as any other esp32 board can be slow if you make it work too much. It's powerful but it's not x86.

rmeissn, 2024-12-20:
I've contacted people on the voice assistant discord of esphome and uploaded a serial log there. It seems like no one got a clue of what's wrong. To me, it seems like the boards are simply buggy.

[serial_voice-assisstant_logs.txt](https://github.com/user-attachments/files/18210410/serial_voice-assisstant_logs.txt)