esphome/issues #6464: Audio.h missing from ESPHome

noahhamilton2018, 2024-11-21:
### The problem

When installing new ESPHome device the Audio.h file seems to be missing.

### Which version of ESPHome has the issue?

2024.11.0 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.11.2

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: esp32s3
  friendly_name: ReSpeakerv3
  platformio_options:
    board_build.flash_mode: dio
    board_build.mcu: esp32s3
  # on_boot:
  #   then:
  #     - if:
  #            condition:
  #              switch.is_on: 
  #            then:
  #              - voice_assistant.start_continuous:

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf
    version: recommended

logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  - platform: esphome
    password: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Respeaker-S3 Fallback Hotspot"
    password:

captive_portal:
  
external_components:
  - source: github://QingWind6/ESPHome_XIAO-ESP32S3

i2s_audio_xiao:
  i2s_lrclk_pin: GPIO7
  i2s_bclk_pin: GPIO8
  i2s_mclk_pin: GPIO9

microphone:
  - platform: i2s_audio_xiao
    id: xiao_mic
    adc_type: external
    i2s_din_pin: GPIO44
    pdm: false
    bits_per_sample: 32bit
    channel: left

speaker:
  - platform: i2s_audio_xiao
    id: xiao_speaker
    dac_type: external
    i2s_dout_pin: GPIO43
    mode: stereo

voice_assistant:
  microphone: xiao_mic
  use_wake_word: true
  noise_suppression_level: 0
  auto_gain: 0dBFS
  volume_multiplier: 1
  speaker: xiao_speaker
  id: assist
  on_listening:
    - light.turn_on:
        id: led
        blue: 100%
        red: 0%
        green: 0%
        effect: "Slow Pulse"
  on_stt_vad_end:
    - light.turn_on:
        id: led
        blue: 100%
        red: 0%
        green: 0%
        effect: "Fast Pulse"
  on_tts_start:
    - light.turn_on:
        id: led
        blue: 100%
        red: 0%
        green: 0%
        brightness: 100%
        effect: none
  on_end:
    - delay: 100ms
    - wait_until:
        not:
          speaker.is_playing:
    - script.execute: reset_led
  on_error:
    - light.turn_on:
        id: led
        red: 100%
        green: 0%
        blue: 0%
        brightness: 100%
        effect: none
    - delay: 1s
    - script.execute: reset_led
  on_client_connected:
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - voice_assistant.start_continuous:
          - script.execute: reset_led
  on_client_disconnected:
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - voice_assistant.stop:
          - light.turn_off: led

light:
  - platform: esp32_rmt_led_strip
    id: led
    name: None
    disabled_by_default: true
    entity_category: config
    pin: GPIO1
    default_transition_length: 0s
    chipset: ws2812
    num_leds: 1
    rgb_order: grb
    rmt_channel: 0
    effects:
      - pulse:
          name: "Slow Pulse"
          transition_length: 250ms
          update_interval: 250ms
          min_brightness: 50%
          max_brightness: 100%
      - pulse:
          name: "Fast Pulse"
          transition_length: 100ms
          update_interval: 100ms
          min_brightness: 50%
          max_brightness: 100%

output:
  - platform: ledc
    id: light_output
    pin: GPIO21
    inverted: true

script:
  - id: reset_led
    then:
      - if:
          condition:
            - switch.is_on: use_wake_word
            - switch.is_on: use_listen_light
          then:
            - light.turn_on:
                id: led
                effect: none

          else:
            - light.turn_off: 
               id: led

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - lambda: id(assist).set_use_wake_word(true);
      - if:
          condition:
            not:
              - voice_assistant.is_running
          then:
            - voice_assistant.start_continuous
    on_turn_off:
      - voice_assistant.stop
      - lambda: id(assist).set_use_wake_word(false);

  - platform: template
    name: Use Listen Light
    id: use_listen_light
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - script.execute: reset_led
    on_turn_off:
      - script.execute: reset_led
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/voice_assistant/voice_assistant.h:16,
                 from src/esphome/components/api/api_connection.cpp:22:
src/esphome/components/speaker/speaker.h:13:10: fatal error: esphome/components/audio/audio.h: No such file or directory
 #include "esphome/components/audio/audio.h"
```

CamronBorealis, 2024-11-21:
I am receiving this as well when using the 2024.11.0 esphome Python library on Windows with the ESP32-S3-BOX-3B

CamronBorealis, 2024-11-21:
I solved the boot loop issue by monitoring the device and finding that the idf_v4.4_freertos.patch wasn't applied. I applied it and now everything is working as expected.

FelixGaebler, 2024-11-21:
Thanks for the quick fix @CamronBorealis!

MikeDembek, 2024-11-21:
Adding the "audio:" line also worked for me

bert269, 2024-11-22:
I honestly do not know how to fix this. Can anyone please help? People say it's working, but not for me. I have now tried to re-compile this code for the last 12 hours - no luck.
Can some please give more details on how to fix this compile error?
Thank you

Anto79-ops, 2024-11-22:
11.1 seems to have fixed it for me without any modification to the yaml

closed 2025-04-18 by github-actions[bot]