esphome/issues #6465: Error Resolving Ip Adress of xxx.xxx.xxx.xxx

sokarovski, 2024-11-21:
### The problem

3 out of 4 ESPs in my home failed to install wirelessly today, all reporting **Error Resolving Ip Address of xxx.xxx.xxx.xxx**  
ESPs communicate with home assistant without any problems, And also inside ESPHome when you press to see the logs wirelessly everything works.  

```
INFO ESPHome 2024.11.0
INFO Reading configuration /config/esphome/esp-garage.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp-garage (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
RAM:   [====      ]  42.1% (used 34464 bytes from 81920 bytes)
Flash: [=====     ]  48.3% (used 494213 bytes from 1023984 bytes)
========================= [SUCCESS] Took 1.26 seconds =========================
INFO Successfully compiled program.
ERROR Error resolving IP address of 192.168.10.135. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Int or String expected
```

### Which version of ESPHome has the issue?

2024.11.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.11.2

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
esphome:
  name: "esp-garage"
  friendly_name: "Garage"

esp8266:
  board: esp01_1m

logger:

api:
  encryption:
    key: "..."

ota:
  platform: esphome
  port: 8266
  password: "..."

wifi:
  ssid: !secret ...
  password: !secret ...

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esp-garage"
    password: !secret ...

captive_portal:
```

rest77, 2024-11-21:
One more. Same issue here.

lhassell, 2024-11-21:
I am having the same issue. On the devices I can directly connect to USB, the update works fine. Any attempt to update via WiFi produces this error.

Doridian, 2024-11-22:
I hit the same problem on all my Sonoff S31 outlets. Everything else updated fine.

Downgraded to 2024.10 and confirmed I can still OTA fine from there (which works perfectly still)

asperger514, 2024-11-22:
It's been over a year that this has solved all my wifi problems, it may solve yours.  

on_boot:
    - priority: 300 # AFTER_BLUETOOTH (Before WiFi)
      then:
        lambda: |-
          WiFi.setPhyMode(WIFI_PHY_MODE_11G);

fkuiper, 2024-11-22:
Same here. 

Tried several options (like switching to a static ip and switching, adding the 'use_address' and 'domain' attributes to the wifi configuration) but can't upgrade my esp8266 boards over wifi. Since they are in hard to reach places I haven't tried to upgrade them via USB yet.

Tried the option @asperger514 mentioned on one of the boards but it didn't work for me but I think this more or less is the chicken and the egg situation: I can't upgrade my boards so I want to try this setting but to try this setting I first have to upgrade my boards... which of course I still can't :) 

A downgrade @Doridian mentioned is something to consider, but I rather wait a little bit more to see if there will be a fix available soon. Besides upgrading, everything else seems to work fine (boards still communicate fine with home assistent and even after a reboot they reconnect perfectly.

sokarovski, 2024-11-22:
I have Managed to pinpoint the problem. The problem seems to be **port** on the ota. When there is a set port for the OTA it seems that the resolver has problem resolving the hostname to ip. I am assuming it's because different kind of object is passed to the resolver rather than Int or String that the resolver is expecting. 

```
ota:
  - platform: esphome
  **- port: 8266**
  - password: "..."
```

The only device that It was working with the latest 2024.11.x actually didnt had any port set. So i tried to remove the port property on the other esp's and they started resolving and updating. 

ESP8266 devices didnt had any kind of problem they just immediately started working.

ESP32 devices were not that easy. All my devices were set to ota over port 8266, even the ESP32's. The default port for ESP32 is 3232. These devices ware not working obviously because esphome tried to connect to port 3232 while the devices were listening to 8266. I have manually flashed this devices with cable, and after that they started working with OTA updates. 

But this looks like a bug in esphome and probably should be fixed.

fkuiper, 2024-11-23:
Removing the port attribute worked for me aswell!

Thanks @sokarovski !

rest77, 2024-11-23:
Thanks @sokarovski!

You can change the port via OTA by using:

  on_boot:
    - lambda: |-
        id(my_ota).set_port(8266);

similar to this:

https://esphome.io/components/ota/esphome.html

Don't forget the ID

id: my_ota in OTA:

/Reik

TripitakaBC, 2024-11-23:
@rest77 solution seems to be hampered by this bug. I've been meaning to look for the solution to changing ports wirelessly on ESP32 devices for a while and never got to looking for it properly. Thanks for the pointer.

The issue here for those of us with ESP32 on ports other than defaults is that the current ESPhome refuses to connect to a device when the port is specified in the YAML. If the port is removed, ESPhome updates the device just fine. Thanks @sokarovski 

The on_boot code to change the port requires the code on the device to be updated which we cannot do as ESPhome won't connect to anything other than 3232 on an ESP32 currently. If your port is set as 8266 then the bug needs fixing before you can use this to change it to the default 3232.

Useful in general and a good lesson but doesn't allow progress here for working around the bug.

Jogiarea, 2024-11-24:
Same Problem here with esp32cam and port 8266
Can i change the port OTA or only with direct Plug?
I Dont have an OTA ID

sokarovski, 2024-11-24:
> Same Problem here with esp32cam and port 8266 Can i change the port OTA or only with direct Plug? I Dont have an OTA ID

The only way i know now is to use cable, and remote the ota port. But hopefully they will fix the bug soon and you can do it the wireless way if you are not in a rush.

TripitakaBC, 2024-11-27:
ESPhome just updated to 2024.11.2 but no fix included for this bug.

closed 2024-12-01 by frenck