esphome/issues #6466: 2024.11.0 `visual_temperature` Changes Break Previous Working Configurations

barndawgie, 2024-11-21:
### The problem

This configuration was working prior to 2024.11.0:
```yaml
climate:
    visual:
      min_temperature: "43.3333"
      max_temperature: "60"
      temperature_step: 1.0f
```

Now, it gives the following error:
```
climate.econet: [source build-yaml/../econet_tankless_water_heater.yaml:16]
climate:
  visual: 
    min_temperature: 43.3333
    max_temperature: 60
    
    Expected visual_temperature with unit, got 1e-15.
    temperature_step: 1.0f
```

### Which version of ESPHome has the issue?

2024.11.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

climate:

### Example YAML snippet

```yaml
climate:
    visual:
      min_temperature: "43.3333"
      max_temperature: "60"
      temperature_step: 1.0f
```

### Anything in the logs that might be useful for us?

```txt
climate.econet: [source build-yaml/../econet_tankless_water_heater.yaml:16]
climate:
  visual: 
    min_temperature: 43.3333
    max_temperature: 60
    
    Expected visual_temperature with unit, got 1e-15.
    temperature_step: 1.0f
```
```

### Additional information

Please see this issue opened against our package: https://github.com/esphome-econet/esphome-econet/issues/429

Our nightly build started failing with the 2024.11.0 release: https://github.com/esphome-econet/esphome-econet/actions/runs/11944436631

barndawgie, 2024-11-21:
https://github.com/esphome-econet/esphome-econet/issues/429 is the related issue on our side.

jesserockz, 2024-11-21:
Capital F for Fahrenheit!

closed 2024-11-21 by jesserockz