esphome/issues #6469: "ESPHome Device Compiler" HA add-on title

jbtwo, 2024-11-22:
### The problem

It seems the ESPHome add-on for Home Assistant has been renamed to `ESPHome Device Compiler`. This label is used in the sidebar in the Home Assistant UI, but the long name makes it looks very out of place compared to other items on the sidebar. AFAIK Home Assistant doesn't provide a method for users to manually rename sidebar items.
[image]

### Which version of ESPHome has the issue?

2024.11.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

maxexcloo, 2024-11-22:
Same issue for me, would prefer the older name if possible

jbtwo, 2024-11-22:
Hey @balloob! I just noticed you originally pushed this change in https://github.com/esphome/issues/issues/6469. Curious if you're able to provide additional context, or if you can comment on whether we can consider reverting it?

bigmak40, 2024-11-23:
Agreed on this being an annoyance. It looks even worse if using the Beta.

[image]

I saw their reasoning for the rename here; I see where they're coming from, but I wish it was just the title on install but not on use.

https://esphome.io/changelog/2024.11.0.html

> Further to this, we have slightly changed the identity of ESPHome as an add-on in Home Assistant. It is now called ESPHome Device Compiler, because that is what it does, and most people do not need it if they are not going to actually compile anything.

DerEmder, 2024-11-23:
The same here. It just doesn't look right and the renaming was unneccessary. Beside - it does not only compile things - it's an editor, a flashtool and a log function - so the new title is just wrong. It is beyond me why someone even considers to make this loooong title instead of just "espHome" like it was all the time. So, please undo this or give us the possibility to rename it ourselves. Thank you.

wildekek, 2024-11-24:
1. There is no such thing as a "Device Compiler", you can't compile a device, you can compile firmware. So even if the add-on name needed to change (which it doesn't) it is the wrong name.
2. It is much more than a compiler:
- It is also the "ESPHome Log Viewer"
- It is also the "ESPHome Update Manager"
- It is also the "ESPHome Secrets Manager"
- It is also the "ESPHome Device Adoption Manager"

DerEmder, 2024-11-28:
Todays update did not change a thing. 

I agree, it is not the most important thing in the world, but it shows me the arrogant mindset of the devs. I stopped supporting it financially already and I think about going back to my own 'manual' ESP programming and MQTT. That's sad because ESPhome was what brought me to Home Assistant in the first place. What a stupid decision of the devs ...

spitfire, 2024-11-28:
While I disagree with the change I'm not as upset about it as the others. It doesn't help with anything, and it takes up the most space in my sidebar (when I have it expanded to show names).

DerEmder, 2024-11-28:
> Since 2024.11.2 it is "ESPHome Compiler".

Nope, still 'ESPhome device compiler' on my system. Doen't matter anyway - it still sucks (and is still wrong).

DerEmder, 2024-11-28:
> While I disagree with the change I'm not as upset about it as the others. It doesn't help with anything, and it takes up the most space in my sidebar (when I have it expanded to show names).

It is even worse on mobile devices with lower resolutions.

alexandrezia, 2024-12-14:
I would like to rename it back to `ESPHome`, this new long name makes the vertical menu bar largeer, and takes more screen space, and on small devices such tablets (which I run on my wall) the menu is unnecessarily large. I prefer to reclaim right side space rather than have a larger left side menu ...

jbtwo, 2024-12-18:
Looks like the sidebar title has been updated to `ESPHome Compiler`. Thank you! 

Although I do prefer the original title `ESPHome`, this is definitely better than the previous long title. 🙏

[image]

randybb, 2024-12-18:
Compiler? Today it is called ESPHome Builder :D

jbtwo, 2024-12-18:
@randybb Ah good point, on the Add-ons page it's called `ESPHome Device Builder`.

Any idea why they're different, or if/how the sidebar title can be updated?

[image]

randybb, 2024-12-18:
Refresh? Maybe it is cashed in the browser. In the dashboard pane should be visible the shortened name as is in the [addon config](https://github.com/esphome/home-assistant-addon/blob/main/esphome/config.yaml#L34-L35)

spitfire, 2024-12-18:
> Compiler? Today it is called ESPHome Builder :D

that is on 2024.11.3, wonder what's it going to be called tomorrow ;)

spitfire, 2024-12-18:
> Refresh? Maybe it is cashed in the browser. In the dashboard pane should be visible the shortened name as is in the [addon config](https://github.com/esphome/home-assistant-addon/blob/main/esphome/config.yaml#L34-L35)

Arbitrary8481, 2024-12-19:
Apparently, esphome builder, which doesn't change the sidebar width issue. Nice try, though.

closed 2025-05-04 by balloob

Arbitrary8481, 2025-05-04:
Still isn't really fixed but they obviously don't care.

bigmak40, 2025-05-04:
> Still isn't really fixed but they obviously don't care.

I ended up just hiding it from the left bar, that's my stupid way of addressing it...

balloob, 2025-05-05:
It has been fixed. It's called ESPHome Builder and not ESPHome Device Compiler as per the original issue.