esphome/issues #6470: Raspberry pi pico w fail to compile

freddiemare, 2024-11-22:
### The problem

After Both the Current version: 2024.11.1 and Previous version: 2024.11.0 of ESPHome fail to update my Raspberry Pi Pico W RP2040's.

### Which version of ESPHome has the issue?

2024.11.0 and 2024.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.11.2 Supervisor 2024.11.4 Operating System 13.2 Frontend 20241106.2

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Component causing the issue

Compiling

### Example YAML snippet

```yaml
# Full functional code. 
# This is one of my devices but I have three more Pi Pico W RP2040 devices and they all result in the same way when I try to update them. there are only two that have Display's
substitutions:
  ipaddress: 192.168.2.61
  gate_way: 192.168.2.1
  sub_net: 255.255.255.0
  dns_1: 192.168.2.1
  dns_2: 1.1.1.1
  friendly_name: Soldering Pad
  projectname: "plot35.solderingpad"
  device_name: "solderingpad"
  time_timezone: "Africa/Johannesburg"
  device_description: "Plot35 Soldering Pad V ${version}"
  created_by: "Freddie Mare 2024"
  devicename: plot35boreholepump
  wifichannel: "7"
  version: "1.0.3.8"

  # Display
  # Graphing reducing x-scale to prevent too many gridlines
  timeduration: "10min"
  xgrid: "1min"
  ygrid: "2.0"
  # Big Pad
  #pid_kp: "0.08230"
  #pid_ki: "0.00082"
  #pid_kd: "2.06166"

  #Small Pad
  pid_kp: "0.03675"
  pid_ki: "0.00062"
  pid_kd: "0.54579"

  processnoise: "0.0028"
  
esphome:
  name: '${device_name}'
  friendly_name: '${friendly_name}'
  comment: $device_description
[... 788 lines cut ...]
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
Dependency Graph
|-- WiFi @ 1.0.0
|-- LEAmDNS @ 1.2
|-- Updater @ 1.0
|-- SPI @ 1.0
|-- MD5Builder @ 1.0.0
|-- lwIP-Ethernet @ 1
|-- lwIP_CYW43 @ 1
Compiling .pioenvs/solderingpad/src/main.cpp.o
Linking .pioenvs/solderingpad/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `std':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0xda): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x124): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x17c): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `__sfp':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:201:(.text+0x1ea): undefined reference to `__retarget_lock_init_recursive'
collect2: error: ld returned 1 exit status
*** [.pioenvs/solderingpad/firmware.elf] Error 1
========================= [FAILED] Took 541.08 seconds =========================
```

ssieb, 2024-11-22:
Try removing the `framework:` section.

freddiemare, 2024-11-22:
Thank you ssieb That resolved the issue the issue on all devices.

closed 2024-11-22 by freddiemare