esphome/issues #6471: Devices appear Offline after some time

sokarovski, 2024-11-22:
### The problem

Devices updated to 2024.11.x Start to appear off-line in ESPHome after some time (several hours). Doing a restart on the devices fixes them and they appear online again. During this off-line period they appear to be working fine with Home Assistant. 

My guess is that they stop sending multicast messages. 

Having this problem on 2 updated devices which have wildly different yaml setup so i dont believe it's related to any yaml configuration.

Older 2024.10.x devices doesnt appear to be having this problem.

### Which version of ESPHome has the issue?

2024.11.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.11.1
INFO Reading configuration /config/esphome/alarm.yaml...
INFO Starting log output from esp-utility.local using esphome API
WARNING Can't connect to ESPHome API for esp-utility.local: Error resolving IP address: [Errno -5] No address associated with hostname (APIConnectionError)
INFO Trying to connect to esp-utility.local in the background
```

SGXander, 2024-11-22:
I too have this problem but it's not after some time, rather immediately. some devices are "online" but most are offline in the ESPHome page. all devices are actually quite happy in HA though.

sokarovski, 2024-11-25:
Update on the problem, only 1 of the devices continued with this behaviour, The device stops sending mDNS advertisements. I use a mdns query tool only that device wasn't present. When added the platform: restart button and after pressing the button, the device appeared in ESPHome and in the mDNS debugging tool. Note that the device works nicely with home assistant, that is how i pressed the button.

SGXander, 2024-11-25:
restarting them does seem to "heal" them in the esphome page and they appear online however they still report the error above for me when attempting to compile and update them.

SGXander, 2024-11-25:
worth noting as well that ethernet connected ESPs are unaffected. It seems as though a timeout or threshold has been changed in this ESPHome version that doesn't allow long enough for a response/connection

sokarovski, 2024-11-27:
> restarting them does seem to "heal" them in the esphome page and they appear online however they still report the error above for me when attempting to compile and update them.

@SGXander  If the Update problem says: "Int or String expected", that is another issue described here which has workaround: https://github.com/esphome/issues/issues/6465

SGXander, 2024-11-27:
> @SGXander  If the Update problem says: "Int or String expected", that is another issue described here which has workaround: https://github.com/esphome/issues/issues/6465

No it's the same mDNS query failure. Even after a fresh restart of the ESP.