esphome/issues #6472: Failure to compile Google Fonts

fawick, 2024-11-22:
### The problem

After updating to 2024.11.1 (from 2024.10.2), the font block prevents compilation:

```
INFO ESPHome 2024.11.1
INFO Reading configuration /config/esphome/og1floorheater.yaml...
INFO Detected timezone 'Europe/Berlin'
Failed config

font: [source /config/esphome/og1floorheater.yaml:233]
  
  Font Roboto@400@False@v1.ttf is missing 1 glyph:
      → (b'\\u2192').
  - file: gfonts://Roboto
    id: roboto_small
    size: 24
    glyphs: 
      - !"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@
      - →
  - file: gfonts://Roboto
    id: roboto_large
    size: 56
    glyphs: 
      - !"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqr
  ```
  
  
  The YAML block in the configuration looks like this:
  
 ```yaml
 font:
  - file: 'gfonts://Roboto'
    id: roboto_small
    size: 24
    glyphs: [
      "!\"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@",
      "\u2192", # →
      ]
  - file: "gfonts://Roboto"
    id: roboto_large
    size: 56
    glyphs: [
      "!\"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@",
      "\u2192", # →
      ]
```

### Which version of ESPHome has the issue?

2024.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.11.2

### What platform are you using?

ESP32

### Board

m5stack-coreink

### Component causing the issue

font

### Example YAML snippet

```yaml
- file: 'gfonts://Roboto'
    id: roboto_small
    size: 24
    glyphs: [
      "!\"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@",
      "\u2192", # →
      ]
  - file: "gfonts://Roboto"
    id: roboto_large
    size: 56
    glyphs: [
      "!\"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@",
      "\u2192", # →
      ]
```

### Anything in the logs that might be useful for us?

```txt
Compilation log:

INFO ESPHome 2024.11.1
INFO Reading configuration /config/esphome/og1floorheater.yaml...
INFO Detected timezone 'Europe/Berlin'
Failed config

font: [source /config/esphome/og1floorheater.yaml:233]
  
  Font Roboto@400@False@v1.ttf is missing 1 glyph:
      → (b'\\u2192').
  - file: gfonts://Roboto
    id: roboto_small
    size: 24
    glyphs: 
      - !"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@
      - →
  - file: gfonts://Roboto
    id: roboto_large
    size: 56
    glyphs: 
      - !"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqr
```

fawick, 2024-11-22:
To limit the range of potential commits I tried more ESPHome tags and found

- 2024.10.3 compiles normally
- 2024.11.0b1 fails with the same error message listed above

fawick, 2024-11-22:
`git bisect` identified the cause to be https://github.com/esphome/esphome/pull/7429

```
a043022444609d11b2a8b040ae9515e54ce940f2 is the first bad commit
commit a043022444609d11b2a8b040ae9515e54ce940f2 (HEAD)
Date:   Thu Oct 31 05:36:23 2024 +0200

    [font] Add support for "glyphsets" (#7429)
```

/cc @paravoid

fawick, 2024-11-22:
After reading the discussion on https://github.com/esphome/esphome/pull/7429 I understand that the font is indeed missing the glyph for the rightwards arrow. And that I get an error message instead of a warning because I explicitely request that glyph.  

Before, older versions of ESPHome did not provide me with any written hint so I was unaware of the problem which existed longer than that.

Technically, my issue is resolved and I could close it. However, I'd like to redefine the issue to be about the nature of the error message now. 

Instead of only getting the message `Font Roboto@400@False@v1.ttf is missing 1 glyph:` I would like ESPHome to be more clarifying and verbose to make it clear that this is not bug in the code of ESPHome itself. For example:

```
Font Roboto@400@False@v1.ttf is missing 1 glyph:
[...]
ESPHome versions since 2024.11.0 introduced testing for missing glyphs. The problem is most likely in the font itself. Check the documentation."
```

clydebarrow, 2024-11-22:
I understand the frustration when something that was apparently working just fine suddenly won't build, and I'm glad you were able to resolve it so quickly.

Detecting errors at build time rather than having to debug running code is a key goal of ESPHome, and this change, while causing some short term inconvenience to some people, makes the product better for the long run.

closed 2024-11-22 by clydebarrow

samperk1, 2025-01-07:
i am confused what is the fix for this? i have the esp32-s2-box and had yaml that was working. now i am not able to compile the updates the thing works with the old software but cant update.

''font: [source /config/esphome/esp32-s3-box-3-5ab4f8.yaml:573]
  
  Font Roboto@500@False@v1.ttf is missing 632 glyphs:
      א (b'\\u05d0')
      ב (b'\\u05d1')
      ג (b'\\u05d2')
      ד (b'\\u05d3')
      ה (b'\\u05d4')
      ו (b'\\u05d5')
      ז (b'\\u05d6')
      ח (b'\\u05d7')
      ט (b'\\u05d8')
      י (b'\\u05d9')
      and 622 more.
  - file: gfonts://Roboto@500
    id: font_small
    size: 14''
this is the message i am getting. not sure what to do.

clydebarrow, 2025-01-07:
> * not sure what to do

Ask for help on the Discourse forum, or in the Discord #general-support channel. Read [this](https://community.home-assistant.io/t/how-to-help-us-help-you-or-how-to-ask-a-good-question/114371/3) first.