esphome/issues #6473: T-Display S3 AMOLED Display not working

shmuelzon, 2024-11-22:
### The problem

Hello,

I'm trying to get my T-Display S3 AMOLED v2 (without touch), which should be the same as the touch version, just without the additional touch panel. The original firmware works fine so the HW is OK but when loading ESPHome, I get the following log messages in a loop:
```
[10:11:33][W][component:237]: Component display took a long time for an operation (688 ms).
[10:11:33][W][component:238]: Components should block for at most 30 ms.
[10:11:34][W][component:237]: Component display took a long time for an operation (688 ms).
[10:11:34][W][component:238]: Components should block for at most 30 ms.
[10:11:35][W][component:237]: Component display took a long time for an operation (688 ms).
[10:11:35][W][component:238]: Components should block for at most 30 ms.
```
With nothing displayed on the screen.

I used the sample configuration from the [QSPI_DPI}(https://esphome.io/components/display/qspi_dbi.html) instructions but am not sure why it's not working.

I would appreciate any help to get this working.

Thanks in advance!

### Which version of ESPHome has the issue?

2024.11.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

esp32-s3-devkitc-1 / T-Display S3 AMOLED

### Component causing the issue

qspi_dbi

### Example YAML snippet

```yaml
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf

psram:
  mode: octal

spi:
  id: quad_spi
  type: quad
  clk_pin: 47
  data_pins:
    - 18
    - 7
    - 48
    - 5

display:
  - platform: qspi_dbi
    model: RM67162
    id: my_display
    dimensions:
      height: 240
      width: 536
    transform:
      mirror_x: true
      swap_xy: true
    color_order: rgb
    brightness: 255
    cs_pin: 6
    reset_pin: 17
    enable_pin: 38
    lambda: |-
      auto rand_col = Color::random_color();
      it.filled_rectangle(0, 0, it.get_width(), it.get_height(), rand_col);
```

### Anything in the logs that might be useful for us?

```txt
[10:46:39][W][component:237]: Component display took a long time for an operation (697 ms).
[10:46:39][W][component:238]: Components should block for at most 30 ms.
[10:46:40][W][component:237]: Component display took a long time for an operation (696 ms).
[10:46:40][W][component:238]: Components should block for at most 30 ms.
[10:46:41][W][component:237]: Component display took a long time for an operation (696 ms).
[10:46:41][W][component:238]: Components should block for at most 30 ms.
[10:46:42][W][component:237]: Component display took a long time for an operation (695 ms).
[10:46:42][W][component:238]: Components should block for at most 30 ms.
```

### Additional information

This is the specific model: https://lilygo.cc/products/t-display-s3-amoled?variant=43506902368437
V2.0 Non-Soldered Pin [H713]. AFAIK, it's the same board as the touch version, same pinout and all.

clydebarrow, 2024-11-22:
I have reproduced the problem, will arrange a fix, for now you can work around it by reverting to the previous driver:

```
external_components:
  - source: github://esphome/esphome@6c11f0bd5152b32579b3f86802365f2e308e9ca4
    components: [qspi_amoled]

display:
  - platform: qspi_amoled
    model: RM67162
    etc....
```

shmuelzon, 2024-11-22:
@clydebarrow Thanks for that! It started working (after I noticed I also need to change the platform back to `qspi_amoled` 🤦‍♂️).

I still see the
```
[16:21:51][W][component:237]: Component display took a long time for an operation (681 ms).
[16:21:51][W][component:238]: Components should block for at most 30 ms.
```
log messages, though. Is that expected with this display?

clydebarrow, 2024-11-22:
> Is that expected with this displa

Usin the display lambda, yes. Use LVGL instead, it is much faster.

closed 2024-11-24 by jesserockz