esphome/issues #6475: lvgl widget Line (is it possible to set up points using lambda?)

LGhost1973, 2024-11-22:
### The problem

Is it possible to set up points using lambda?

### Which version of ESPHome has the issue?

last

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

last

### What platform are you using?

ESP32

clydebarrow, 2024-11-23:
yes, it should be. you will just need to figure out the correct syntax. I'd suggest you look at the generated code for lines to do that.

closed 2024-11-23 by clydebarrow

LGhost1973, 2024-11-23:
is it possible to describe this in the documentation for widget Line?

LGhost1973, 2024-11-23:
and is there a transfer plan for the widget Charts?

clydebarrow, 2024-11-23:
> is it possible to describe this in the documentation for widget Line?

No, it's not practical to include in the documentation all the implementation details of widgets. The chart widget will be implemented at some point, can't offer a timeline ATM.