esphome/issues #6479: ESP32-S3-BOX Get stuck on answering page until hard reset.

maxi1134, 2024-11-22:
### The problem

The ESPBox will answer the query and then get stuck on this page:

[image]

### Which version of ESPHome has the issue?

Irrelevant, this has been happening on the last 10 versions.

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Irrelevant, this has been happening on the last 10 versions.

### What platform are you using?

ESP8266

### Board

ESP32-s3-box 

### Component causing the issue

assist

### Additional information

My 7 ESP32-s3-boxes encounter this issue on various versions of ESPhome

maxi1134, 2024-11-22:
It seems like if it get stuck but a previously set timer goes off, when pressing the button it unstuck and goes back to listening.

maxi1134, 2024-11-25:
I managed to get logs of when this occurs:

```

[10:45:26][W][component:237]: Component voice_assistant took a long time for an operation (247 ms).
[10:45:26][W][component:238]: Components should block for at most 30 ms.
[10:45:26][D][esp_adf.microphone:285]: Microphone stopped
[10:45:26][D][voice_assistant:512]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[10:45:26][D][voice_assistant:635]: Event Type: 4
[10:45:26][D][voice_assistant:663]: Speech recognised as: " What is maxi doing"
[10:45:26][D][text_sensor:064]: 'text_request': Sending state ' What is maxi doing'
[10:45:26][W][component:237]: Component voice_assistant took a long time for an operation (241 ms).
[10:45:26][W][component:238]: Components should block for at most 30 ms.
[10:45:26][D][voice_assistant:635]: Event Type: 5
[10:45:26][D][voice_assistant:668]: Intent started
[10:45:37][D][voice_assistant:635]: Event Type: 6
[10:45:37][D][voice_assistant:635]: Event Type: 7
[10:45:37][D][voice_assistant:691]: Response: "It seems that I couldn't find any information about Maxi's current activity.

However, according to the previous state, Maxi was "Listening to Music"."
[10:45:37][D][text_sensor:064]: 'text_response': Sending state 'It seems that I couldn't find any information about Maxi's current activity.

However, according to the previous state, Maxi was "Listening to Music".'
[10:45:37][D][voice_assistant:635]: Event Type: 98
[10:45:37][D][voice_assistant:779]: TTS stream start
[10:45:37][D][esp-idf:000][speaker_task]: I (48276718) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=8

[10:45:37][D][esp-idf:000][speaker_task]: I (48276725) I2S: I2S0, MCLK output by GPIO2

[10:45:37][D][esp-idf:000][speaker_task]: I (48276731) AUDIO_PIPELINE: link el->rb, el:0x3d040518, tag:raw, rb:0x3d040688

[10:45:37][D][esp-idf:000][speaker_task]: I (48276737) AUDIO_ELEMENT: [raw-0x3d040518] Element task created

[10:45:37][D][esp-idf:000][speaker_task]: I (48276742) AUDIO_ELEMENT: [i2s-0x3d03f668] Element task created

[10:45:37][D][esp-idf:000][speaker_task]: I (48276746) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:16587867 Bytes, Inter:84752 Bytes, Dram:84752 Bytes

[10:45:37][D][esp-idf:000][i2s]: I (48276753) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1

[10:45:37][D][esp-idf:000][i2s]: I (48276755) I2S_STREAM: AUDIO_STREAM_WRITER

[10:45:37][D][esp-idf:000][speaker_task]: I (48276758) AUDIO_PIPELINE: Pipeline started

[10:45:37][W][component:237]: Component voice_assistant took a long time for an operation (275 ms).
[10:45:37][W][component:238]: Components should block for at most 30 ms.
[10:45:37][D][voice_assistant:635]: Event Type: 8
[10:45:37][D][voice_assistant:713]: Response URL: "http://192.168.0.15:8123/api/tts_proxy/ee9392d5c3abb0ae51392330db776fa554afe936_en-us_f632bda3b3_tts.piper_machinelearning_gpu.wav"
[10:45:37][D][voice_assistant:512]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[10:45:37][D][voice_assistant:518]: Desired state set to STREAMING_RESPONSE
[10:45:37][D][voice_assistant:635]: Event Type: 2
[10:45:37][D][voice_assistant:727]: Assist Pipeline ended
[10:45:39][D][esp-idf:000][speaker_task]: W (48278327) AUDIO_PIPELINE: There are no listener registered

[10:45:39][D][esp-idf:000][speaker_task]: I (48278332) AUDIO_PIPELINE: audio_pipeline_unlinked

[10:45:39][D][esp-idf:000][speaker_task]: W (48278339) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE

[10:45:39][D][esp-idf:000][speaker_task]: I (48278348) I2S: DMA queue destroyed

[10:45:39][D][esp-idf:000][speaker_task]: W (48278355) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERMINATE

[10:45:39][D][esp-idf:000][speaker_task]: W (48278366) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINATE

[10:45:39][D][esp-idf:000][speaker_task]: I (48278387) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=8

[10:45:39][D][esp-idf:000][speaker_task]: I (48278393) I2S: I2S0, MCLK output by GPIO2

[10:45:39][D][esp-idf:000][speaker_task]: I (48278402) AUDIO_PIPELINE: link el->rb, el:0x3d040518, tag:raw, rb:0x3d040688

[10:45:39][D][esp-idf:000][speaker_task]: I (48278408) AUDIO_ELEMENT: [raw-0x3d040518] Element task created

[10:45:39][D][esp-idf:000][speaker_task]: I (48278419) AUDIO_ELEMENT: [i2s-0x3d03f668] Element task created

[10:45:39][D][esp-idf:000][speaker_task]: I (48278424) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:16609475 Bytes, Inter:106360 Bytes, Dram:106360 Bytes

[10:45:39][D][esp-idf:000][i2s]: I (48278431) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1

[10:45:39][D][esp-idf:000][i2s]: I (48278436) I2S_STREAM: AUDIO_STREAM_WRITER

[10:45:39][D][esp-idf:000][speaker_task]: I (48278441) AUDIO_PIPELINE: Pipeline started

[10:45:47][D][esp32.preferences:114]: Saving 1 preferences to flash...
[10:45:47][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[10:45:47][D][voice_assistant:635]: Event Type: 99
[10:45:47][D][voice_assistant:789]: TTS stream end
[10:45:47][D][voice_assistant:377]: End of audio stream received
[10:45:47][D][voice_assistant:512]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[10:45:47][D][voice_assistant:518]: Desired state set to RESPONSE_FINISHED
[10:45:48][D][esp-idf:000][speaker_task]: W (48287546) AUDIO_PIPELINE: There are no listener registered

[10:45:48][D][esp-idf:000][speaker_task]: I (48287551) AUDIO_PIPELINE: audio_pipeline_unlinked

[10:45:48][D][esp-idf:000][speaker_task]: W (48287556) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE

[10:45:48][D][esp-idf:000][speaker_task]: I (48287564) I2S: DMA queue destroyed

[10:45:48][D][esp-idf:000][speaker_task]: W (48287571) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERMINATE

[10:45:48][D][esp-idf:000][speaker_task]: W (48287578) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINATE
```

schneekluth, 2024-11-27:
I no longer have this Issue with the latest ESPHome Release `2024.11.2`.

maxi1134, 2024-11-27:
> I no longer have this Issue with the latest ESPHome Release `2024.11.2`.

Just updated and the issue happened after the first try.

Still present.

maxi1134, 2024-12-07:
The last update 2024.11.3 did not fix the issue.

Sometimes even default intents such as "Turn the lights to red" end up on a frozen "Done" answer.

alucryd, 2024-12-21:
In the same boat, happens to Atom Echos too.

maxi1134, 2025-04-22:
Not stale afaik