esphome/issues #6481: fatal error: pins_arduino.h: No such file or directory

jorgemontalvao, 2024-11-23:
### The problem

If you follow the steps to create a docker image from [getting_started_command_line - installation](https://esphome.io/guides/getting_started_command_line#installation), and then 

1. Click on the "New Device" button
2. Enter a name and click on "Next" ("x" in this case)
3. Skip the "Installation" step
4. With "Use recommended settings" selected, click on "ESP32"
5. Click on the "Install" button
6. Click on the "Manual download"
After "Compiling .pioenvs/x/src/esphome/components/md5/md5.cpp.o", the system returns yellow and this red message "/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory"

[image]
Here is the log
[logs_x_compile.txt](https://github.com/user-attachments/files/17880002/logs_x_compile.txt)

Here is the x.yaml as "txt" file
[x.yaml.txt](https://github.com/user-attachments/files/17880015/x.yaml.txt)

It happens many time
[image]

As a work around, swoboda1337 recommended to change "esp32 framework type" from "arduino" to "esp-idf", and it works, but I still believe it is a bug. The default esp32 framework type is arduino, and should work any way.

Thanks to swoboda1337 <3
[Link to Discord Topic: [SOLVED] fatal error: pins_arduino.h: No such file or directory](https://discord.com/channels/429907082951524364/1309454699123966002/1309524797885124699)

### Which version of ESPHome has the issue?

ESPHome 2024.11.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

pins_arduino.h

### Example YAML snippet

```yaml
esphome:
  name: x
  friendly_name: x

esp32:
  board: esp32dev
  framework:
    type: arduino
```

### Anything in the logs that might be useful for us?

```txt
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory
```

### Additional information

services:
  esphome:
    container_name: esphome
    image: ghcr.io/esphome/esphome:stable
    volumes:
      - C:\esphome\config:/config
      - /etc/localtime:/etc/localtime:ro
    restart: always
    privileged: true
#    network_mode: host
    ports:
      - 6052:6052
      - 6123:6123

tcb211, 2024-11-28:
Same issue here with 2024.11.3. Relevant YAML below.

`esphome:
  name: garage3
  friendly_name: garage3
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32.git
    board: arduino_nano_esp32
    framework: arduino

esp32:
  board: arduino_nano_esp32
  variant: esp32s3
  framework:
    type: arduino`

tcb211, 2024-12-19:
> Same issue here with 2024.11.3. Relevant YAML below.
>
> ```
> esphome:
>   name: garage3
>   friendly_name: garage3
>   platformio_options:
>     platform: [...]

also with 2024.12.0

tcb211, 2025-02-26:
As of 2025.2.0, issue persisted, so I dropped the ```arduino_nano_esp32```  (even tho that is the board I am using) and replaced with ```esp32-s3-devkitc-1``` in a simpler YAML, which works.
``` 
esphome:
  name: garage3
  friendly_name: garage3

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
```
Replacing ```esp32-s3-devkitc-1``` with ```arduino_nano_esp32``` in the above YAML results in an unknown board ID error. So clearly something is amiss with ```arduino_nano_esp32``` as a board specification.

ssieb, 2025-02-26:
That board is not supported by the platformio version.  Just use the devkit board.  And don't override the platform versions.

tcb211, 2025-02-26:
[https://docs.platformio.org/en/latest/boards/espressif32/arduino_nano_esp32.html](url)

ssieb, 2025-02-26:
"latest" is not what we're using.