esphome/issues #6482: Wifi Output Power does not get applied

DerGary, 2024-11-24:
### The problem

I tried to change the `output_power` of the `wifi` component but it seems to have no effect, the signal strength was -69dBm before the change and is still the same after changing to the lowest value 8.5 as well as the highest value 20.

### Which version of ESPHome has the issue?

2024.11.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.11.3

### What platform are you using?

ESP32

### Board

diymore 3 Stück ESP32 USB C ESP32 Entwicklungsausschuss ESP32 WROOM 32 ESP32 NodeMcu Development Board 2,4 GHz WLAN WiFi Bluetooth CP2102

### Component causing the issue

wifi

### Example YAML snippet

```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wlan-ssid
  password: !secret wlan-password
  output_power: 8.5dB

sensor:
  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 1s
    entity_category: "diagnostic"
```

### Anything in the logs that might be useful for us?

```txt
not really, here is the output of the wifi component:

[14:04:51][C][wifi:600]: WiFi:
[14:04:51][C][wifi:428]:   Local MAC: XXX
[14:04:51][C][wifi:433]:   SSID: 'XXX'
[14:04:51][C][wifi:436]:   IP Address: XXX
[14:04:51][C][wifi:440]:   BSSID: XXX
[14:04:51][C][wifi:441]:   Hostname: 'XXX'
[14:04:51][C][wifi:443]:   Signal strength: -69 dB ▂▄▆█
[14:04:51][C][wifi:447]:   Channel: 11
[14:04:51][C][wifi:448]:   Subnet: 255.255.255.0
[14:04:51][C][wifi:449]:   Gateway: XXX
[14:04:51][C][wifi:450]:   DNS1: XXX
[14:04:51][C][wifi:451]:   DNS2: 0.0.0.0
```

### Additional information

I'm trying to decrease the wifi power as I believe it might interfere with my motion sensor that is right next to the board and is often triggered false positively.

randybb, 2024-11-24:
It is signal strength that is the esphome device receiving (from AP), check signal strength on your AP.

DerGary, 2024-11-24:
oh you are right, when I look at the values on the AP they change when I change the output_power 🙈 I'm sorry, I thought the change should be visible in both directions. Thanks you for the quick reply, I'm a bit smarter now 🙏

closed 2024-11-24 by DerGary