esphome/issues #6490: Sunrise/sunset time not in the correct timezone on boot

AndreasBVB, 2024-11-27:
### The problem

I realized the next sunrise/sunset time sensor are consistently displaying time in UTC right after boot and correcting themselves after about a minute. This is highly undesirable because automations based on these sensors need to be delayed for an unknown amount of time.

### Which version of ESPHome has the issue?

2024.9.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
sun:
  latitude: 40.486932143332034,°
  longitude: -78.72494325515122°

text_sensor:
  - platform: sun
    id: next_sunset_h
    type: sunset
    format: "%H"
```

ssieb, 2024-11-27:
What is the time source?  The sensors shouldn't have a value until the time is synced, so if it's showing before that I would consider it a bug.

AndreasBVB, 2024-11-27:
Time source is from a highly reliable NTP on LAN:
```yaml
time:
  - platform: sntp
    id: sntp_time
    timezone: America/New_York
    servers:
     - 192.168.1.1
     - time.nist.gov
     - pool.ntp.org
```

After boot the time will quickly sync while the sunset hours remaining UTC. The sunset hour corrects itself after about a minute, way after the NTP time sync.

ssieb, 2024-11-27:
Can you paste serial logs?  The sensor only updates every 60s by default, but it really shouldn't even have a value until the time is valid.

AndreasBVB, 2024-11-27:
Had to trim off a lot of crap, does it still help?
I was expecting info about setting up the SNTP and sun components but didn't see any. Is it because it's WIFI log?
```
[01:08:38][I][app:100]: ESPHome version 2024.9.1 compiled on Nov 27 2024, 01:08:16
[01:08:38][C][wifi:600]: WiFi:
[01:08:38][C][wifi:428]:   Local MAC: AA:AA:AA:AA:AA:AA
[01:08:38][C][wifi:433]:   SSID: [redacted]
[01:08:38][C][wifi:436]:   IP Address: 192.168.1.249
[01:08:38][C][wifi:440]:   BSSID: [redacted]
[01:08:38][C][wifi:441]:   Hostname: 'lamp-piano'
[01:08:38][C][wifi:443]:   Signal strength: -29 dB ▂▄▆█
[01:08:38][C][wifi:447]:   Channel: 11
[01:08:38][C][wifi:448]:   Subnet: 255.255.255.0
[01:08:38][C][wifi:449]:   Gateway: 192.168.1.254
[01:08:38][C][wifi:450]:   DNS1: 192.168.1.254
[01:08:38][C][wifi:451]:   DNS2: 0.0.0.0
[01:08:38][C][logger:185]: Logger:
[01:08:38][C][logger:186]:   Level: DEBUG
[01:08:38][C][logger:188]:   Log Baud Rate: 115200
[01:08:38][C][uptime.sensor:033]: Uptime Sensor 'LAMP PIANO Uptime'
[01:08:38][C][uptime.sensor:033]:   Device Class: 'duration'
[01:08:38][C][uptime.sensor:033]:   State Class: 'total_increasing'
[01:08:38][C][uptime.sensor:033]:   Unit of Measurement: 's'
[01:08:38][C][uptime.sensor:033]:   Accuracy Decimals: 0
[01:08:38][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[01:08:38][C][uptime.sensor:034]:   Type: Seconds
[01:08:38][C][template.sensor:022]: Template Sensor 'LAMP PIANO NEXT SUNSET'
[01:08:38][C][template.sensor:022]:   State Class: ''
[01:08:38][C][template.sensor:022]:   Unit of Measurement: ''
[01:08:38][C][template.sensor:022]:   Accuracy Decimals: 1
[01:08:38][C][template.sensor:023]:   Update Interval: 60.0s
[01:08:38][C][sun.text_sensor:009]: Sun Text Sensor 'next_sunset_h_text'
[01:08:38][C][sun.text_sensor:009]:   Icon: 'mdi:weather-sunset-down'
[01:08:38][C][sntp:048]: SNTP Time:
[01:08:38][C][sntp:049]:   Server 1: '192.168.1.1'
[01:08:38][C][sntp:050]:   Server 2: 'time.nist.gov'
[01:08:38][C][sntp:051]:   Server 3: 'pool.ntp.org'
[01:08:38][C][sntp:052]:   Timezone: 'EST5EDT,M3.2.0,M11.1.0'
[01:08:38][C][mdns:116]: mDNS:
[01:08:38][C][mdns:117]:   Hostname: lamp-piano
[01:08:38][C][esphome.ota:073]: Over-The-Air updates:
[01:08:38][C][esphome.ota:074]:   Address: lamp-piano.local:3232
[01:08:38][C][esphome.ota:075]:   Version: 2
[01:09:05][D][text_sensor:064]: 'next_sunset_h_text': Sending state '16'
```

ssieb, 2024-11-27:
That's not serial logs and doesn't show the problem.  That value looks correct to me.

ssieb, 2024-11-27:
Yes, that's what you need.  Your board doesn't have a USB port?
Your esphome version is also kind of old.

AndreasBVB, 2024-11-27:
It's a ESP32-C3 based switch with very hard to access UART PINs. While looking for boards, I noticed something interesting:
- the time sync will briefly sync the device to UTC time before syncing correctly to local time
- it looks like the next sunset is immediately set after time is synced to UTC
- the next sunset time sensor will wait for the next update (60s) to correct itself to local time
- if I change the update interval of the next sunset sensor to 10s, it doesn't change anything, the correction from UTC time still waits for about 1min
- if I change the update interval to very short: 1s or shorter, the device won't connect to HomeAssistant API or web, with error messages:
```
WARNING Can't connect to ESPHome API for lamp-piano @ 192.168.1.249: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.249', port=6053))]: [Errno 111] Connect call failed ('192.168.1.249', 6053) (SocketAPIError)
```

ssieb, 2024-11-27:
Can you show the logs?  I don't see that.  Using ntp, it syncs to the timezone and the sunset shows correctly.

AndreasBVB, 2024-11-27:
OK here we go, looks like the next sunrise/sunset sensor got set way before SNTP time sync. Maybe this is not intended?
```
[13:02:24]ESP-ROM:esp32c3-api1-20210207
[13:02:24]Build:Feb  7 2021
[13:02:24]rst:0x15 (USB_UART_CHIP_RESET),boot:0xf (SPI_FAST_FLASH_BOOT)
[13:02:24]Saved PC:0x4205cbf4
[13:02:24]SPIWP:0xee
[13:02:24]mode:DIO, clock div:1
[13:02:24]load:0x3fcd6100,len:0x38c
[13:02:24]load:0x403ce000,len:0x6ac
[13:02:24]load:0x403d0000,len:0x24e4
[13:02:24]entry 0x403ce000
[13:02:25][I][logger:171]: Log initialized
[13:02:25][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[13:02:25][D][esp32.preferences:114]: Saving 1 preferences to flash...
[13:02:25][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[13:02:25][I][app:029]: Running through setup()...
[13:02:25][D][text_sensor:064]: 'DEBUG SUN MAC': Sending state 'AA:AA:AA:AA:AA:AA'
[13:02:25][C][wifi:048]: Setting up WiFi...
[13:02:25][C][wifi:061]: Starting WiFi...
[13:02:25][C][wifi:062]:   Local MAC: AA:AA:AA:AA:AA:AA
[13:02:25][D][wifi:760]: Loaded saved fast_connect wifi settings
[13:02:25][I][wifi:313]: WiFi Connecting to 'SomeSSID'...
[13:02:25][D][sensor:094]: 'DEBUG SUN Uptime': Sending state 0.24400 s with 0 decimals of accuracy
[13:02:25][D][text_sensor:064]: 'NEXT SUNSET TIME': Sending state '21:57:34'
[13:02:25][W][component:237]: Component sun.text_sensor took a long time for an operation (603 ms).
[13:02:25][W][component:238]: Components should block for at most 30 ms.
[13:02:25][D][text_sensor:064]: 'NEXT SUNRISE TIME': Sending state '12:20:28'
[13:02:25][W][component:157]: Component wifi set Warning flag: associating to network
[13:02:25][I][wifi:617]: WiFi Connected!
[13:02:25][C][wifi:428]:   Local MAC: AA:AA:AA:AA:AA:AA
[13:02:25][C][wifi:433]:   SSID: [redacted]
[13:02:25][C][wifi:436]:   IP Address: 192.168.1.243
[13:02:25][C][wifi:440]:   BSSID: [redacted]
[13:02:25][C][wifi:441]:   Hostname: 'debug-sun'
[13:02:25][C][wifi:443]:   Signal strength: -48 dB ▂▄▆█
[13:02:25][C][wifi:447]:   Channel: 11
[13:02:25][C][wifi:448]:   Subnet: 255.255.255.0
[13:02:25][C][wifi:449]:   Gateway: 192.168.1.254
[13:02:25][C][wifi:450]:   DNS1: 192.168.1.254
[13:02:25][C][wifi:451]:   DNS2: 0.0.0.0
[13:02:25][D][wifi:626]: Disabling AP...
[13:02:25][C][web_server:100]: Setting up web server...
[13:02:25][C][sntp:023]: Setting up SNTP...
[13:02:25][C][api:026]: Setting up Home Assistant API server...
[13:02:25][I][app:062]: setup() finished successfully!
[13:02:25][D][text_sensor:064]: 'DEBUG SUN IP': Sending state '192.168.1.243'
[13:02:25][W][component:170]: Component wifi cleared Warning flag
[13:02:25][D][sntp:073]: Synchronized time: 2024-11-27 13:02:25
[13:02:25][W][component:157]: Component api set Warning flag: unspecified
[13:02:25][I][app:100]: ESPHome version 2024.11.1 compiled on Nov 27 2024, 13:00:41
[13:02:25][C][wifi:600]: WiFi:
[13:02:25][C][wifi:428]:   Local MAC: AA:AA:AA:AA:AA:AA
[13:02:25][C][wifi:433]:   SSID: [redacted]
[13:02:25][C][wifi:436]:   IP Address: 192.168.1.243
[13:02:25][C][wifi:440]:   BSSID: [redacted]
[13:02:25][C][wifi:441]:   Hostname: 'debug-sun'
[13:02:25][C][wifi:443]:   Signal strength: -48 dB ▂▄▆█
[13:02:25][C][wifi:447]:   Channel: 11
[13:02:25][C][wifi:448]:   Subnet: 255.255.255.0
[13:02:25][C][wifi:449]:   Gateway: 192.168.1.254
[13:02:25][C][wifi:450]:   DNS1: 192.168.1.254
[13:02:25][C][wifi:451]:   DNS2: 0.0.0.0
[13:02:25][C][logger:185]: Logger:
[13:02:25][C][logger:186]:   Level: DEBUG
[13:02:25][C][logger:188]:   Log Baud Rate: 115200
[13:02:25][C][logger:189]:   Hardware UART: USB_CDC
[13:02:25][C][uptime.sensor:033]: Uptime Sensor 'DEBUG SUN Uptime'
[13:02:25][C][uptime.sensor:033]:   Device Class: 'duration'
[13:02:25][C][uptime.sensor:033]:   State Class: 'total_increasing'
[... 99 lines cut ...]
    #update_interval: 10s

sensor:
  - platform: uptime
    update_interval: 600s
    name: ${device_upper} Uptime
  
button:
  - platform: restart
    name: ${device_upper} RESTART
    entity_category: config

time:
  - platform: sntp
    id: sntp_time
    timezone: America/New_York
    servers:
     - time.nist.gov
     - pool.ntp.org

sun:
  latitude: 39.8°
  longitude: -80.2°

```

ssieb, 2024-11-27:
Can you change the sunrise and sunset format to `"%c"` instead, to see what date it has?

AndreasBVB, 2024-11-27:
Correct date and time, only in UTC (wrong time zone):
```
[13:31:40]ESP-ROM:esp32c3-api1-20210207
[13:31:40]Build:Feb  7 2021
[13:31:40]rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
[13:31:40]Saved PC:0x403820cc
[13:31:40]SPIWP:0xee
[13:31:40]mode:DIO, clock div:1
[13:31:40]load:0x3fcd6100,len:0x38c
[13:31:40]load:0x403ce000,len:0x6ac
[13:31:40]load:0x403d0000,len:0x24e4
[13:31:40]entry 0x403ce000
[13:31:41][I][logger:171]: Log initialized
[13:31:41][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[13:31:41][D][esp32.preferences:114]: Saving 1 preferences to flash...
[13:31:41][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[13:31:41][I][app:029]: Running through setup()...
[13:31:41][D][text_sensor:064]: 'DEBUG SUN MAC': Sending state 'AA:AA:AA:AA:AA:AA'
[13:31:41][C][wifi:048]: Setting up WiFi...
[13:31:41][C][wifi:061]: Starting WiFi...
[13:31:41][C][wifi:062]:   Local MAC: AA:AA:AA:AA:AA:AA
[13:31:41][I][wifi:313]: WiFi Connecting to 'SomeSSID'...
[13:31:41][D][sensor:094]: 'DEBUG SUN Uptime': Sending state 0.32200 s with 0 decimals of accuracy
[13:31:41][D][text_sensor:064]: 'NEXT SUNRISE TIME': Sending state 'Thu Nov 28 12:20:28 2024'
[13:31:41][D][text_sensor:064]: 'NEXT SUNSET TIME': Sending state 'Wed Nov 27 21:57:34 2024'
[13:31:41][W][component:157]: Component wifi set Warning flag: associating to network
[13:31:44][I][wifi:617]: WiFi Connected!
[13:31:44][C][wifi:428]:   Local MAC: AA:AA:AA:AA:AA:AA
[13:31:44][C][wifi:433]:   SSID: [redacted]
[13:31:44][C][wifi:436]:   IP Address: 192.168.1.243
[13:31:44][C][wifi:440]:   BSSID: [redacted]
[13:31:44][C][wifi:441]:   Hostname: 'debug-sun'
[13:31:44][C][wifi:443]:   Signal strength: -49 dB ▂▄▆█
[13:31:44][C][wifi:447]:   Channel: 11
[13:31:44][C][wifi:448]:   Subnet: 255.255.255.0
[13:31:44][C][wifi:449]:   Gateway: 192.168.1.254
[13:31:44][C][wifi:450]:   DNS1: 192.168.1.254
[13:31:44][C][wifi:451]:   DNS2: 0.0.0.0
[13:31:44][D][wifi:626]: Disabling AP...
[13:31:44][D][wifi:776]: Saved fast_connect wifi settings
[13:31:44][C][web_server:100]: Setting up web server...
[13:31:44][C][sntp:023]: Setting up SNTP...
[13:31:44][C][api:026]: Setting up Home Assistant API server...
[13:31:44][I][app:062]: setup() finished successfully!
[13:31:44][D][esp32.preferences:114]: Saving 1 preferences to flash...
[13:31:44][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[13:31:44][D][text_sensor:064]: 'DEBUG SUN IP': Sending state '192.168.1.243'
[13:31:44][W][component:170]: Component wifi cleared Warning flag
[13:31:44][D][sntp:073]: Synchronized time: 2024-11-27 13:31:44
[13:31:44][W][component:157]: Component api set Warning flag: unspecified
[13:31:44][I][app:100]: ESPHome version 2024.11.1 compiled on Nov 27 2024, 13:31:26
[13:31:44][C][wifi:600]: WiFi:
[13:31:44][C][wifi:428]:   Local MAC: AA:AA:AA:AA:AA:AA
[13:31:44][C][wifi:433]:   SSID: [redacted]
[13:31:44][C][wifi:436]:   IP Address: 192.168.1.243
[13:31:44][C][wifi:440]:   BSSID: [redacted]
[13:31:44][C][wifi:441]:   Hostname: 'debug-sun'
[13:31:44][C][wifi:443]:   Signal strength: -49 dB ▂▄▆█
[13:31:44][C][wifi:447]:   Channel: 11
[13:31:44][C][wifi:448]:   Subnet: 255.255.255.0
[13:31:44][C][wifi:449]:   Gateway: 192.168.1.254
[13:31:44][C][wifi:450]:   DNS1: 192.168.1.254
[13:31:44][C][wifi:451]:   DNS2: 0.0.0.0
[13:31:44][C][logger:185]: Logger:
[13:31:44][C][logger:186]:   Level: DEBUG
[13:31:44][C][logger:188]:   Log Baud Rate: 115200
[13:31:44][C][logger:189]:   Hardware UART: USB_CDC
[13:31:44][C][uptime.sensor:033]: Uptime Sensor 'DEBUG SUN Uptime'
[13:31:44][C][uptime.sensor:033]:   Device Class: 'duration'
[13:31:44][C][uptime.sensor:033]:   State Class: 'total_increasing'
[... 68 lines cut ...]
[13:31:46][C][web_server:154]:   Address: debug-sun.local:80
[13:31:46][C][sntp:048]: SNTP Time:
[13:31:46][C][sntp:049]:   Server 1: 'time.nist.gov'
[13:31:46][C][sntp:050]:   Server 2: 'pool.ntp.org'
[13:31:46][C][sntp:051]:   Server 3: ''
[13:31:46][C][sntp:052]:   Timezone: 'EST5EDT,M3.2.0,M11.1.0'
[13:31:46][C][mdns:116]: mDNS:
[13:31:46][C][mdns:117]:   Hostname: debug-sun
[13:31:46][C][esphome.ota:073]: Over-The-Air updates:
[13:31:46][C][esphome.ota:074]:   Address: debug-sun.local:3232
[13:31:46][C][esphome.ota:075]:   Version: 2
[13:31:46][C][safe_mode:018]: Safe Mode:
[13:31:46][C][safe_mode:020]:   Boot considered successful after 60 seconds
[13:31:46][C][safe_mode:021]:   Invoke after 10 boot attempts
[13:31:46][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[13:31:46][C][api:140]: API Server:
[13:31:46][C][api:141]:   Address: debug-sun.local:6053
[13:31:46][C][api:145]:   Using noise encryption: NO
[13:31:46][C][wifi_info:010]: WifiInfo IPAddress 'DEBUG SUN IP'
[13:32:29][D][text_sensor:064]: 'NEXT SUNRISE TIME': Sending state 'Thu Nov 28 07:20:28 2024'
[13:32:33][D][text_sensor:064]: 'NEXT SUNSET TIME': Sending state 'Wed Nov 27 16:57:34 2024'
[13:32:41][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[13:32:41][D][esp32.preferences:114]: Saving 1 preferences to flash...
[13:32:41][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```

ssieb, 2024-11-27:
How is that possible?  Where is it getting the time from?

AndreasBVB, 2024-11-27:
That I can't answer... I presented the full yaml 😐

ssieb, 2024-11-27:
How are you rebooting it?  What happens if you power cycle it?

AndreasBVB, 2024-11-27:
Great catch! I always used the "soft reboot" button, so the UTC time gets saved in the system maybe? I tried the EN button and unplugging and the two sensors become unavailable after "hard reboot". So maybe holding the sensors from presenting values before time sync will solve this problem?

ssieb, 2024-11-27:
Somehow a soft reboot is holding the time, but the timezone isn't available until the ntp component initializes.  So there's no bug here, just a small glitch if you soft reboot.  I'm very curious how the time survives like that, but it seems like this would be a rare situation anyway.

closed 2024-11-27 by ssieb