esphome/issues #6494: [modbus_controller] `max_cmd_retries` works only once when a server becomes offline

Sleeper85, 2024-11-27:
### The problem

Hi @martgras @0x3333

I'm testing a client modbus code that queries three different servers. If server `241` becomes offline, after 6 requests, a `1min pause` (3s*20) is observed and then the client sends new requests indefinitely without respecting the `max_cmd_retries` parameters.

Server 1 information continues to be retrieved but server 2 information is lost.

If I reconnect `server 241` to the RS485 bus everything works correctly again.

This problem exists with esphome `2024.11.1` but also with `2024.10.3` (before the PR of @0x3333), I have not tested other esphome versions.

The RS485 bus

```
R120ohm---(server 2)-----(server 1)-----(client)-----(server 241)---R120ohm
```

### Which version of ESPHome has the issue?

2024.11.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

esp32doit-devkit-v1

### Component causing the issue

modbus_controller

### Example YAML snippet

```yaml
modbus_controller:

  - id: modbus1
    modbus_id: modbus_client
    address: 241
    update_interval: 3s
    command_throttle: 5ms
    max_cmd_retries: 3
    offline_skip_updates: 20

  - id: modbus2
    modbus_id: modbus_client
    address: 1
    update_interval: 3s
    command_throttle: 5ms
    max_cmd_retries: 3
    offline_skip_updates: 20

  - id: modbus3
    modbus_id: modbus_client
    address: 2
    update_interval: 3s
    command_throttle: 5ms
    max_cmd_retries: 3
    offline_skip_updates: 20
```

### Anything in the logs that might be useful for us?

[... 64 lines cut ...]
[16:24:43][D][modbus_controller:038]: Modbus command to device=241 register=0x08 no response received - removed from send queue
[16:24:44][D][modbus_controller:038]: Modbus command to device=241 register=0x1A no response received - removed from send queue
[16:24:46][D][modbus_controller:038]: Modbus command to device=241 register=0x1B no response received - removed from send queue
[16:24:47][D][modbus_controller:038]: Modbus command to device=241 register=0x01 no response received - removed from send queue
[16:24:48][D][modbus_controller:038]: Modbus command to device=241 register=0x05 no response received - removed from send queue
[16:24:49][D][modbus_controller:038]: Modbus command to device=241 register=0x06 no response received - removed from send queue
[16:24:50][D][modbus_controller:038]: Modbus command to device=241 register=0x07 no response received - removed from send queue
[16:24:51][D][modbus_controller:038]: Modbus command to device=241 register=0x08 no response received - removed from send queue
[16:24:52][D][modbus_controller:038]: Modbus command to device=241 register=0x1A no response received - removed from send queue
[16:24:54][D][modbus_controller:038]: Modbus command to device=241 register=0x1B no response received - removed from send queue
[16:24:55][D][modbus_controller:038]: Modbus command to device=241 register=0x01 no response received - removed from send queue
[16:24:56][D][modbus_controller:038]: Modbus command to device=241 register=0x05 no response received - removed from send queue
[16:24:57][D][modbus_controller:038]: Modbus command to device=241 register=0x06 no response received - removed from send queue
[16:24:58][D][modbus_controller:038]: Modbus command to device=241 register=0x07 no response received - removed from send queue
[16:24:59][D][modbus_controller:038]: Modbus command to device=241 register=0x08 no response received - removed from send queue
[16:25:00][D][modbus_controller:038]: Modbus command to device=241 register=0x1A no response received - removed from send queue
[16:25:01][D][modbus_controller:038]: Modbus command to device=241 register=0x01 no response received - removed from send queue
[16:25:02][D][modbus_controller:038]: Modbus command to device=241 register=0x1B no response received - removed from send queue
[16:25:03][D][modbus_controller:038]: Modbus command to device=241 register=0x05 no response received - removed from send queue
[16:25:04][D][modbus_controller:038]: Modbus command to device=241 register=0x06 no response received - removed from send queue
[16:25:06][D][modbus_controller:038]: Modbus command to device=241 register=0x07 no response received - removed from send queue
[16:25:07][D][modbus_controller:038]: Modbus command to device=241 register=0x01 no response received - removed from send queue
[16:25:08][D][modbus_controller:038]: Modbus command to device=241 register=0x08 no response received - removed from send queue
...
```

0x3333, 2024-11-27:
I need all your yaml file, looks like you have several registers in this server...

Sleeper85, 2024-11-27:
I could provide you with more logs tomorrow. In the meantime, here are all the YAMLs used.

[YamBMS 1.5.2 beta1](https://drive.google.com/file/d/1Y-gl4UdeWBCzqf7FwF28YxG7GiSroUN5/view?usp=drivesdk)

Sleeper85, 2024-11-29:
> I need all your yaml file, looks like you have several registers in this server...
> 
> Edit 1: If you could post all logs (without redaction), it would help to check the timings. Edit 2: A verbose log would help a lot.

Attached is an export of the full VERBOSE logs without filtering.

[logs_device_241_disconnected_from_the_RS485_bus_2024.11.19.zip](https://github.com/user-attachments/files/17958282/logs_device_241_disconnected_from_the_RS485_bus_2024.11.19.zip)

0x3333, 2024-12-06:
This log doesn't show the behavior you reported. Also, I see a lot of duplicate commands, which means that you are sending too fast, or you need to decrease the `send_wait_time` in Modbus component. As soon as I have more time, I'll check you log thoroughly

Sleeper85, 2024-12-06:
> This log doesn't show the behavior you reported. Also, I see a lot of duplicate commands, which means that you are sending too fast, or you need to decrease the `send_wait_time` in Modbus component. As soon as I have more time, I'll check you log thoroughly

If you search for `device=241` you can see the behavior I described. When node 241 is disconnected, after the 4th send without response you can see a `1min pause` (3s [update_interval] * 20 [offline_skip_updates]) is observed and then the client sends new requests indefinitely without respecting the `max_cmd_retries` parameters.

```
[11:09:01][W][modbus_controller:027]: Modbus device=241 set offline
[11:09:01][D][modbus_controller:040]: Modbus command to device=241 register=0x07 no response received - removed from send queue
[11:09:03][D][modbus_controller:040]: Modbus command to device=241 register=0x08 no response received - removed from send queue
[11:09:04][D][modbus_controller:040]: Modbus command to device=241 register=0x1A no response received - removed from send queue
[11:09:05][D][modbus_controller:040]: Modbus command to device=241 register=0x1B no response received - removed from send queue

( 1min pause )

[11:10:03][D][modbus_controller:040]: Modbus command to device=241 register=0x01 no response received - removed from send queue
[11:10:04][D][modbus_controller:040]: Modbus command to device=241 register=0x05 no response received - removed from send queue
[11:10:06][D][modbus_controller:040]: Modbus command to device=241 register=0x06 no response received - removed from send queue
...
[11:14:58][D][modbus_controller:040]: Modbus command to device=241 register=0x1A no response received - removed from send queue
```

Sleeper85, 2024-12-06:
@0x3333 

> **send_wait_time (Optional)**: Time in milliseconds before the next ModBUS command is sent if an answer from a previous command is pending. Defaults to 250 ms. If multiple ModBUS devices are attached to the same bus increasing this value can help avoiding to to overlapping reads. When two devices are sending a command at the same time the response read from UART can’t be assigned to the proper design. This value defines the maximum queuing time for a command before it is send anyways.

Regarding `send_wait_time` I'm currently using the default value of `250ms`.

If I read the docs it seems that I should increase it?

You are talking about decreasing this value, is this a mistake?

Regarding sending commands, I use `update_interval: 3s` and `command_throttle: 5ms` is that too fast?

The default value of `command_throttle` is `0ms`.

Sleeper85, 2024-12-06:
Please note that when `device 241 is connected` to the bus I don't see a line like the example below, this only happens when `device 241 is disconnected`.

```
[11:14:59][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=27 count=2
```

Sleeper85, 2024-12-06:
@0x3333 

New test with `esphome 2024.11.3` and new discovery. Device `241` is the one with the `120Ohm` end of bus resistor, during my disconnection test of this device `241` the end of bus resistor is removed and it seems that the above problem is related to that.

To verify this hypothesis I added a device 3 in the `client` YAML, this device 3 does not exist on the bus, this is how I carried out the following tests:

Here is the result, it's already better:

```
[14:00:55][D][modbus_controller:040]: Modbus command to device=3 register=0x1B no response received - removed from send queue
[14:02:13][D][modbus_controller:040]: Modbus command to device=3 register=0x01 no response received - removed from send queue
[14:03:31][D][modbus_controller:040]: Modbus command to device=3 register=0x06 no response received - removed from send queue
[14:04:31][D][modbus_controller:040]: Modbus command to device=3 register=0x07 no response received - removed from send queue
[14:05:22][D][modbus_controller:040]: Modbus command to device=3 register=0x08 no response received - removed from send queue
[14:06:13][D][modbus_controller:040]: Modbus command to device=3 register=0x1B no response received - removed from send queue
[14:07:01][D][modbus_controller:040]: Modbus command to device=3 register=0x01 no response received - removed from send queue
[14:07:37][D][modbus_controller:040]: Modbus command to device=3 register=0x06 no response received - removed from send queue
[14:08:49][D][modbus_controller:040]: Modbus command to device=3 register=0x07 no response received - removed from send queue
[14:09:16][D][modbus_controller:040]: Modbus command to device=3 register=0x08 no response received - removed from send queue
```

On the other hand and unfortunately this produces a new problem...

The `device 1` which is actually connected to the bus goes from `online` to `offline` continuously.

```
[14:46:56][W][modbus_controller:027]: Modbus device=1 set offline
[14:46:56][D][main:098]: BMS 1 modbus server goes offline !
[14:46:57][W][modbus_controller:065]: Modbus device=1 back online
[14:46:57][D][main:092]: BMS 1 modbus server back online !
[14:46:59][W][modbus_controller:027]: Modbus device=1 set offline
[14:46:59][D][main:098]: BMS 1 modbus server goes offline !
[14:47:00][W][modbus_controller:065]: Modbus device=1 back online
[14:47:00][D][main:092]: BMS 1 modbus server back online !
[14:47:02][W][modbus_controller:027]: Modbus device=1 set offline
[14:47:02][D][main:098]: BMS 1 modbus server goes offline !
[14:47:03][W][modbus_controller:065]: Modbus device=1 back online
[14:47:03][D][main:092]: BMS 1 modbus server back online !
[14:47:05][W][modbus_controller:027]: Modbus device=1 set offline
[14:47:05][D][main:098]: BMS 1 modbus server goes offline !
[14:47:06][W][modbus_controller:065]: Modbus device=1 back online
[14:47:06][D][main:092]: BMS 1 modbus server back online !
[14:47:08][W][modbus_controller:027]: Modbus device=1 set offline
[14:47:08][D][main:098]: BMS 1 modbus server goes offline !
[14:47:09][W][modbus_controller:065]: Modbus device=1 back online
[14:47:09][D][main:092]: BMS 1 modbus server back online !

```

And we can see duplicated commands

```
[14:48:24][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=1 count=6
[14:48:24][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=6 count=2
[14:48:24][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=7 count=2
[14:48:24][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=8 count=20
[14:48:24][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=27 count=2
[14:48:27][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=27 count=2
[14:48:27][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=1 count=6
[14:48:27][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=6 count=2
[14:48:27][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=7 count=2
[14:48:27][W][modbus_controller:186]: Duplicate modbus command found: type=0x4 address=8 count=20

```

If I recompile the code without the `device 3` everything works correctly.

Attached is an export of the full logs.

[logs_bms1_offline_back_online.zip](https://github.com/user-attachments/files/18039181/logs_bms1_offline_back_online.zip)

Sleeper85, 2024-12-06:
If I add this option `send_wait_time: 500ms` the problem occurs with `device 1` and also with `device 2` but less often.

```
[15:31:37][W][modbus_controller:027]: Modbus device=1 set offline
[15:31:37][D][main:098]: BMS 1 modbus server goes offline !
[15:31:37][D][binary_sensor:036]: 'yambms-node1 BMS 1 Online Status': Sending state OFF
[15:31:48][W][modbus_controller:027]: Modbus device=2 set offline
[15:31:48][D][main:102]: BMS 2 modbus server goes offline !
[15:31:48][D][binary_sensor:036]: 'yambms-node1 BMS 2 Online Status': Sending state OFF
[15:32:46][W][modbus_controller:065]: Modbus device=1 back online
[15:32:46][D][main:092]: BMS 1 modbus server back online !
[15:32:48][W][modbus_controller:027]: Modbus device=1 set offline
[15:32:48][D][main:098]: BMS 1 modbus server goes offline !
[15:33:10][W][modbus_controller:065]: Modbus device=2 back online
[15:33:10][D][main:096]: BMS 2 modbus server back online !
[15:33:12][W][modbus_controller:027]: Modbus device=2 set offline
[15:33:12][D][main:102]: BMS 2 modbus server goes offline !
[15:34:34][W][modbus_controller:065]: Modbus device=1 back online
[15:34:34][D][main:092]: BMS 1 modbus server back online !
[15:34:36][W][modbus_controller:027]: Modbus device=1 set offline
[15:34:36][D][main:098]: BMS 1 modbus server goes offline !
[15:35:34][W][modbus_controller:065]: Modbus device=2 back online
[15:35:34][D][main:096]: BMS 2 modbus server back online !
[15:35:36][W][modbus_controller:027]: Modbus device=2 set offline
[15:35:36][D][main:102]: BMS 2 modbus server goes offline !
[15:36:13][W][modbus_controller:065]: Modbus device=2 back online
[15:36:13][D][main:096]: BMS 2 modbus server back online !
[15:36:13][D][binary_sensor:036]: 'yambms-node1 BMS 2 Online Status': Sending state ON
[15:36:22][W][modbus_controller:065]: Modbus device=1 back online
[15:36:22][D][main:092]: BMS 1 modbus server back online !
[15:36:25][W][modbus_controller:027]: Modbus device=1 set offline
[15:36:25][D][main:098]: BMS 1 modbus server goes offline !
[15:36:30][W][modbus_controller:027]: Modbus device=2 set offline
[15:36:30][D][main:102]: BMS 2 modbus server goes offline !
[15:36:30][D][binary_sensor:036]: 'yambms-node1 BMS 2 Online Status': Sending state OFF
```

Sleeper85, 2024-12-06:
If I increase the `command_throttle` to `50ms` here is what it gives...

In my project this kind of behavior is not possible, if a node has a problem or is disconnected this cannot create such a problem on the RS485 network...

Not so easy to create a RS485 client-server it seems...

Do you have any idea?

```
[15:54:19][W][modbus_controller:027]: Modbus device=3 set offline
[15:54:19][D][main:102]: BMS 3 modbus server goes offline !
[15:54:19][D][binary_sensor:036]: 'yambms-node1 BMS 3 Online Status': Sending state OFF
[15:54:59][W][modbus_controller:065]: Modbus device=1 back online
[15:54:59][D][main:092]: BMS 1 modbus server back online !
[15:54:59][D][binary_sensor:036]: 'yambms-node1 BMS 1 Online Status': Sending state ON
[15:55:09][W][modbus_controller:065]: Modbus device=2 back online
[15:55:09][D][main:096]: BMS 2 modbus server back online !
[15:55:09][D][binary_sensor:036]: 'yambms-node1 BMS 2 Online Status': Sending state ON
[15:55:26][W][modbus_controller:027]: Modbus device=1 set offline
[15:55:26][D][main:098]: BMS 1 modbus server goes offline !
[15:55:26][D][binary_sensor:036]: 'yambms-node1 BMS 1 Online Status': Sending state OFF
[15:55:29][W][modbus_controller:027]: Modbus device=2 set offline
[15:55:29][D][main:102]: BMS 2 modbus server goes offline !
[15:55:29][D][binary_sensor:036]: 'yambms-node1 BMS 2 Online Status': Sending state OFF
[15:57:00][W][modbus_controller:065]: Modbus device=1 back online
[15:57:00][D][main:092]: BMS 1 modbus server back online !
[15:57:03][W][modbus_controller:027]: Modbus device=1 set offline
[15:57:03][D][main:098]: BMS 1 modbus server goes offline !
[15:58:38][W][modbus_controller:065]: Modbus device=1 back online
[15:58:38][D][main:092]: BMS 1 modbus server back online !
[15:58:41][W][modbus_controller:027]: Modbus device=1 set offline
[15:58:41][D][main:098]: BMS 1 modbus server goes offline !
[15:58:43][W][modbus_controller:065]: Modbus device=1 back online
[15:58:43][D][main:092]: BMS 1 modbus server back online !
[15:58:45][W][modbus_controller:027]: Modbus device=1 set offline
[15:58:45][D][main:098]: BMS 1 modbus server goes offline !
[15:59:51][W][modbus_controller:065]: Modbus device=1 back online
[15:59:51][D][main:092]: BMS 1 modbus server back online !
[15:59:53][W][modbus_controller:027]: Modbus device=1 set offline
[15:59:53][D][main:098]: BMS 1 modbus server goes offline !
[16:00:15][W][modbus_controller:065]: Modbus device=2 back online
[16:00:15][D][main:096]: BMS 2 modbus server back online !
[16:00:17][W][modbus_controller:027]: Modbus device=2 set offline
[16:00:17][D][main:102]: BMS 2 modbus server goes offline !
[16:01:50][W][modbus_controller:065]: Modbus device=2 back online
[16:01:50][D][main:096]: BMS 2 modbus server back online !
[16:02:04][W][modbus_controller:027]: Modbus device=2 set offline
[16:02:04][D][main:102]: BMS 2 modbus server goes offline !
```

Sleeper85, 2024-12-06:
With the options `send_wait_time: 500ms` and `command_throttle: 50ms` and removing the `node 3` client from the code the `offline / online` problem continues.

We can say that increasing these times has an impact on the response time of the server (device 1 et 2) which is then declared `offline`.

With the default values ​​for `send_wait_time` and `command_throttle` without the `node 3` there are no more problems.

With the default values ​​for `send_wait_time` and `command_throttle` with the `node 3` (which does not exist so does not respond) `device 1 and 2` switch from `online` to `offline` continuously. Strangely, `device 241` is not impacted (it still works with esphome 2024.11.1, I don't know if that matters).

In short, I don't understand anything about modbus client-server operation anymore...

```
[17:10:38][W][modbus_controller:027]: Modbus device=1 set offline
[17:10:38][D][modbus_controller:040]: Modbus command to device=1 register=0x08 no response received - removed from send queue
[17:10:39][W][modbus_controller:065]: Modbus device=1 back online
[17:10:39][D][modbus_controller:040]: Modbus command to device=3 register=0x08 no response received - removed from send queue
[17:10:41][D][modbus_controller:040]: Modbus command to device=3 register=0x06 no response received - removed from send queue
[17:10:42][D][modbus_controller:040]: Modbus command to device=3 register=0x1B no response received - removed from send queue
[17:10:44][W][modbus_controller:027]: Modbus device=1 set offline
[17:10:44][D][modbus_controller:040]: Modbus command to device=1 register=0x08 no response received - removed from send queue
[17:10:44][W][modbus_controller:065]: Modbus device=1 back online
[17:10:47][W][modbus_controller:027]: Modbus device=1 set offline
[17:10:47][D][modbus_controller:040]: Modbus command to device=1 register=0x07 no response received - removed from send queue
[17:10:47][W][modbus_controller:065]: Modbus device=1 back online
[17:10:49][W][modbus_controller:027]: Modbus device=1 set offline
[17:10:49][D][modbus_controller:040]: Modbus command to device=1 register=0x01 no response received - removed from send queue
[17:10:49][D][modbus_controller:040]: Modbus command to device=3 register=0x07 no response received - removed from send queue
[17:10:50][W][modbus_controller:065]: Modbus device=1 back online
[17:10:52][W][modbus_controller:027]: Modbus device=1 set offline
[17:10:52][D][modbus_controller:040]: Modbus command to device=1 register=0x01 no response received - removed from send queue
[17:10:53][W][modbus_controller:065]: Modbus device=1 back online
[17:10:53][D][modbus_controller:040]: Modbus command to device=3 register=0x08 no response received - removed from send queue
[17:10:56][W][modbus_controller:027]: Modbus device=1 set offline
[17:10:56][D][modbus_controller:040]: Modbus command to device=1 register=0x06 no response received - removed from send queue
[17:10:57][W][modbus_controller:065]: Modbus device=1 back online
[17:10:57][D][modbus_controller:040]: Modbus command to device=3 register=0x06 no response received - removed from send queue
[17:10:58][D][modbus_controller:040]: Modbus command to device=3 register=0x1B no response received - removed from send queue
[17:10:59][D][modbus_controller:040]: Modbus command to device=3 register=0x01 no response received - removed from send queue
[17:11:00][W][modbus_controller:065]: Modbus device=2 back online
[17:11:02][W][modbus_controller:027]: Modbus device=1 set offline
[17:11:02][D][modbus_controller:040]: Modbus command to device=1 register=0x1B no response received - removed from send queue
[17:11:05][W][modbus_controller:027]: Modbus device=2 set offline
[17:11:05][D][modbus_controller:040]: Modbus command to device=2 register=0x07 no response received - removed from send queue
[17:11:06][W][modbus_controller:065]: Modbus device=2 back online
[17:11:06][D][modbus_controller:040]: Modbus command to device=3 register=0x07 no response received - removed from send queue
[17:11:07][D][modbus_controller:040]: Modbus command to device=3 register=0x08 no response received - removed from send queue
[17:11:08][D][modbus_controller:040]: Modbus command to device=3 register=0x06 no response received - removed from send queue
[17:11:11][W][modbus_controller:027]: Modbus device=2 set offline
[17:11:11][D][modbus_controller:040]: Modbus command to device=2 register=0x07 no response received - removed from send queue
[17:11:12][W][modbus_controller:065]: Modbus device=2 back online
[17:11:14][W][modbus_controller:027]: Modbus device=2 set offline
[17:11:14][D][modbus_controller:040]: Modbus command to device=2 register=0x06 no response received - removed from send queue
[17:11:14][W][modbus_controller:065]: Modbus device=2 back online
[17:11:15][D][modbus_controller:040]: Modbus command to device=3 register=0x1B no response received - removed from send queue
[17:11:17][W][modbus_controller:027]: Modbus device=2 set offline
[17:11:17][D][modbus_controller:040]: Modbus command to device=2 register=0x06 no response received - removed from send queue
[17:11:18][W][modbus_controller:065]: Modbus device=2 back online
[17:11:19][W][modbus_controller:027]: Modbus device=2 set offline
[17:11:19][D][modbus_controller:040]: Modbus command to device=2 register=0x01 no response received - removed from send queue
[17:11:19][D][modbus_controller:040]: Modbus command to device=3 register=0x07 no response received - removed from send queue
[... cut ...] s_controller:065]: Modbus device=2 back online
[17:11:22][W][modbus_controller:027]: Modbus device=2 set offline
[17:11:22][D][modbus_controller:040]: Modbus command to device=2 register=0x01 no response received - removed from send queue
[17:11:23][W][modbus_controller:065]: Modbus device=2 back online
[17:11:23][D][modbus_controller:040]: Modbus command to device=3 register=0x06 no response received - removed from send queue
[17:11:26][W][modbus_controller:027]: Modbus device=2 set offline
```

Sleeper85, 2024-12-08:
Hi @0x3333 

On the **modbus_server** side, are these kinds of warnings normal when a server receives requests intended for another server ?

```
[12:02:43][W][modbus:154]: Got Modbus frame from unknown address 0x02! 
[12:02:43][W][modbus:124]: Modbus CRC Check failed! F9F2!=01
[12:02:43][W][modbus:124]: Modbus CRC Check failed! 20!=00
[12:02:43][W][modbus:124]: Modbus CRC Check failed! 94FB!=2D0
[12:02:43][W][modbus:124]: Modbus CRC Check failed! 9EDB!=04
[12:02:43][W][modbus:124]: Modbus CRC Check failed! 98FA!=402
[12:02:43][W][modbus:124]: Modbus CRC Check failed! 130!=00
[12:02:43][W][modbus:124]: Modbus CRC Check failed! FE52!=04
[12:02:43][W][modbus:124]: Modbus CRC Check failed! 3CC3!=01
[12:02:43][W][modbus:124]: Modbus CRC Check failed! 4800!=00
```

Sleeper85, 2024-12-15:
Bug found and fixed in my code, I numbered the registers from 1 to 27 (client and server) and the 32-bit registers were overlapping which caused multiple requests to be sent from the **modbus_controller** instead of just one.

However when a node goes **offline** there is still a problem because **offline_skip_updates** only works once.
I will open a new issue about this.

closed 2024-12-15 by Sleeper85