esphome/issues #6495: error: esphome/components/lvgl/lvgl_proxy.h: No such file or directory - compilation terminated

rdone2018, 2024-11-28:
### The problem

Not possible to compile updated firmware.

### Which version of ESPHome has the issue?

2024.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.11.3 Supervisor 2024.11.4 Operating System 13.2 Frontend 20241106.2

### What platform are you using?

ESP32

### Board

Waveshare ESP32-S3 Touch 4.3

### Component causing the issue

LVGL: src/esphome/components/image/image.cpp:1:

### Example YAML snippet

```yaml
esphome:
  name: esp-32-touchlcd1
  friendly_name: ESP-32-TouchLCD1
  #build_path: "../config/esphome/esp-32-touchlcd1_build"  
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
    board_build.flash_mode: dio
    board_upload.maximum_ram_size: 524288
    board_upload.maximum_size: 80077216

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 8MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: y
      CONFIG_ESP32S3_DATA_CACHE_64KB: y
      CONFIG_SPIRAM_FETCH_INSTRUCTIONS: y
      CONFIG_SPIRAM_RODATA: y

display:
  - platform: rpi_dpi_rgb
    auto_clear_enabled: false
    color_order: RGB
    pclk_frequency: 16MHz
    dimensions:
      width: 800
      height: 480
    de_pin:
      number: 5
    hsync_pin:
      number: 46
      ignore_strapping_warning: true
    vsync_pin:
      number: 3
      ignore_strapping_warning: true
    pclk_pin: 7
    hsync_back_porch: 30
    hsync_front_porch: 210
    hsync_pulse_width: 30
    vsync_back_porch: 4
    vsync_front_porch: 4
    vsync_pulse_width: 4
    data_pins:
      red:
        - 1         #r3
        - 2         #r4
        - 42        #r5
        - 41        #r6
        - 40        #r7
      blue:
        - 14        #b3
        - 38        #b4
        - 18        #b5
        - 17        #b6
        - 10        #b7
      green:
        - 39        #g2
        - 0         #g3
        - 45        #g4
        - 48        #g5
        - 47        #g6
        - 21        #g7

image:
  - file: http://10.168.111.254:8123/local/Gate_full.png
    id: my_image1
    type: RGB565
    use_transparency: false

http_request:

online_image:
  - url: http://10.168.111.254:8123/local/snapshot.png
    format: png
    id: my_image4
    type: RGB565
#    resize: 220x165
    update_interval: 5s
    use_transparency: false
    on_download_finished:
      component.update: my_display

lvgl:
  displays:
    - display_id: my_display
  touchscreens:
    - my_touch
  buffer_size: 25%
#  bg_color: my_black
#  bg_image_src: my_image1
  bg_color: 0x23233D
  bg_grad_color: 0x2B3E4D
  bg_grad_dir: VER   
  pages:
    - id: clock_page
      widgets:
        - obj: # Image
#            height: size_content
            width: 240 #240
            height: 230 #220
            align: BOTTOM_MID
            pad_all: 0
            border_width: 0
            bg_color: 0x000000
            bg_opa: 85%
```

### Anything in the logs that might be useful for us?

```txt
Processing esp-32-touchlcd1 (board: esp32-s3-devkitc-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 512KB RAM, 76.37MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ArduinoJson @ 6.18.5
|-- pngle @ 1.0.2
|-- lvgl @ 8.4.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/esp-32-touchlcd1/src/esphome/components/image/image.o
Compiling .pioenvs/esp-32-touchlcd1/src/esphome/components/json/json_util.o
Compiling .pioenvs/esp-32-touchlcd1/src/esphome/components/key_provider/key_provider.o
Compiling .pioenvs/esp-32-touchlcd1/src/esphome/components/logger/logger.o
In file included from src/esphome/components/image/image.cpp:1:
src/esphome/components/image/image.h:6:10: fatal error: esphome/components/lvgl/lvgl_proxy.h: No such file or directory
 #include "esphome/components/lvgl/lvgl_proxy.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp-32-touchlcd1/src/esphome/components/image/image.o] Error 1
========================= [FAILED] Took 16.23 seconds =========================
```

closed 2024-11-30 by clydebarrow

rdone2018, 2024-12-02:
Yaml is only partial - complete is correct - previous version works fine. Now I missing "lvgl_prxy.h"

clydebarrow, 2024-12-02:
>  Point is that it is missing "lvgl_proxy.h"

The point is that without seeing your complete yaml, nobody can help you.

krbaker, 2024-12-02:
I found this issue because I was seeing the same error.  I was following https://community.home-assistant.io/t/esp32-s3-4-3inch-capacitive-touch-display-from-waveshare/658279/45 which looks similar enough to your code for me to post this hunch.

pulling in the lvgl lib is no longer needed I think (and pulling it back to an old version on that pr I suspect)
external_components:
  - source: github://pr#6363
    refresh: 0min
    components: [lvgl]
    
 I don't see this in your code but its not all there.  Hope this helps, but yeah, post all your code.