esphome/issues #6496: Home Assistant Add-on Sidebar label is very long

bphillips09, 2024-11-28:
### The problem

The Home Assistant Add-on Sidebar label is much longer than before ('ESPHome' -> 'ESPHome Device Compiler'). It causes the Sidebar to extend to fit the label length.

It also creates confusion for a few reasons:
1. It does not link directly to a compiler, but to a page that requires 2-to-3 clicks to have compile functionality. 
2. The label suggests that it is a "_Device_ Compiler", not a "_Firmware_ Compiler"
3. Many tasks can be accomplished on the same page, i.e. viewing logs or selecting devices, etc. and not just compilation

[image]

### Which version of ESPHome has the issue?

2024.11.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.11.2

### What platform are you using?

ESP32

ssieb, 2024-11-28:
https://github.com/esphome/home-assistant-addon/pull/167

ssieb, 2024-11-28:
Update the addon.

closed 2024-11-28 by ssieb