esphome/issues #6497: Audio Playback Not Working After ESPHome Update (2024.10.3 → 2024.11.X) on M5Stack Atom Echo

muik, 2024-11-28:
### The problem

After updating ESPHome from version 2024.10.3 to 2024.11.2, audio playback on my M5Stack Atom Echo is not working. I am using the [m5stack-atom-echo.yaml](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml) configuration and had previously added code to play a sound when a wake word is detected. This feature worked perfectly before the update.

Here’s the code snippet I used for audio playback:
```
micro_wake_word:
  on_wake_word_detected:
    - lambda: id(echo_speaker).play(id(timer_finished_wave_file), sizeof(id(timer_finished_wave_file)));
    - delay: 1s
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
```

Steps to reproduce:

Use M5Stack Atom Echo with the [m5stack-atom-echo.yaml](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml) configuration.
Add code to play a sound on wake word detection.
Update ESPHome to version 2024.11.2.
Attempt to play the sound.
Expected behavior: The sound should play as it did on version 2024.10.3.

Actual behavior: No sound is played.

I’ve confirmed that the wake word detection is working and the rest of the device's functionality appears unaffected. It seems to be an issue specific to audio playback.

Environment:

Device: M5Stack Atom Echo
ESPHome Version: 2024.11.2
Please let me know if additional information or logs are required!

### Which version of ESPHome has the issue?

2024.11.X

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

M5Stack Atom Echo

muik, 2024-11-28:
Tagging @kahrendt as this issue might be related to your expertise.

GrumpyMeow, 2024-11-30:
Aha.. It's not just me having this issue. I'm on 2024.11.1.
I just try to play a sound when i press a button. But nothing... Not the/a beep.

kahrendt, 2024-11-30:
Would you please try the following modified yaml action to help my diagnose the problem?

```
micro_wake_word:
  on_wake_word_detected:
    - lambda: id(echo_speaker).start();
    - delay: 50ms
    - lambda: id(echo_speaker).play(id(timer_finished_wave_file), sizeof(id(timer_finished_wave_file)));
    - delay: 1s
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
```

My best guess is that the speaker's play function doesn't start up fast enough for the data to be written.

mateuszlukasik-PL, 2024-12-01:
> My best guess is that the speaker's play function doesn't start up fast enough for the data to be written.

@kahrendt look here on my latest comment :
[https://github.com/esphome/issues/issues/6222#issuecomment-2509771506](url)

muik, 2024-12-02:
> Would you please try the following modified yaml action to help my diagnose the problem?
>
> ```
> micro_wake_word:
>   on_wake_word_detected:
>     - lambda: id(echo_speaker).start();
>     - delay: 50ms [...]

I tried it, and the sound works well. I’ll also attach the logs in case they might assist with the diagnosis. Thank you!
```
...
[10:00:54][D][micro_wake_word:123]: Wake Word 'okay_nabu' Detected
[10:00:54][D][micro_wake_word:195]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[10:00:54][D][micro_wake_word:129]: Stopping Microphone
[10:00:54][D][micro_wake_word:195]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[10:00:54][D][esp-idf:000]: I (310228) I2S: DMA queue destroyed
[10:00:54]
[10:00:54][D][micro_wake_word:195]: State changed from STOPPING_MICROPHONE to IDLE
[10:00:54][D][ring_buffer:034][speaker_task]: Created ring buffer with size 16000
[10:00:54][D][esp-idf:000][speaker_task]: I (310259) I2S: queue free spaces: 5
[10:00:54]
[10:00:54][D][esp-idf:000][speaker_task]: I (310269) I2S: DMA Malloc info, datalen=blocksize=960, dma_buf_count=4
[10:00:54]
[10:00:54][D][esp-idf:000][speaker_task]: I (310279) I2S: DMA Malloc info, datalen=blocksize=480, dma_buf_count=4
[10:00:54]
[10:00:54][D][i2s_audio.speaker:113]: Started Speaker
[10:00:54][W][component:170]: Component i2s_audio.speaker cleared Warning flag
[10:00:55][D][voice_assistant:512]: State changed from IDLE to START_MICROPHONE
...
```

GrumpyMeow, 2024-12-02:
After adding `start` and `- delay: 50ms` i do get sound.

So the snippet below does work...
```
- platform: template
    name: Trigger
    on_press:
          - lambda: id(echo_speaker).start();
          - delay: 50ms
          - lambda: id(echo_speaker).play(id(timer_finished_wave_file), sizeof(id(timer_finished_wave_file)));
          - wait_until:
              not:
                speaker.is_playing:
```

kahrendt, 2024-12-24:
Instead of guessing the delay duration, you could replace the delay line with: 
```
          - wait_until:
              condition:
                speaker.is_playing:
```

The long term fix will be using the speaker media player (not finished yet) instead of sending audio directly to the speaker component. I'm working on a slimmed down version of the media player suitable for running on devices like an Echo without psram.

muik, 2024-12-26:
Thank you! It works now.

closed 2024-12-26 by muik