esphome/issues #6498: RGB565 with Alpha-Data since #7631

lubeda, 2024-11-28:
### The problem

Hi,
since #7631 (@clydebarrow) images with type RGB565 support transparency in an "weird" (sorry i'm no nativ english speaker and don't if this wording is offending) way. There is a distinction between the type RGB24 (RGB without transparancy) and RGBA (with transparency) But RGB565 has a kind of hidden, implict way of handling the transparancy.
Also RGB565 is for saving memory introducing transparancy to it increases memory requirement to 33%.

### Which version of ESPHome has the issue?

2024.11.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

all

### What platform are you using?

ESP8266

### Board

UIlanzi TC001

### Component causing the issue

image and animation

### Example YAML snippet

```yaml
not relevant
```

### Anything in the logs that might be useful for us?

```txt
I think when converting to RGB565 the transparency of an image should be considered but but for storing and using in esphome the should be two image types RGB565 and RGB565A
```

### Additional information

see discussion here: https://github.com/lubeda/EspHoMaTriXv2/issues/255

clydebarrow, 2024-11-28:
Yes, there does need to be a way of specifying either the original chroma-key transparency or transparency with an alpha channel. Probably reverting the default to chroma-key and adding an alternative value for `transparency:` to add the alpha channel would be best. Stand by.

clydebarrow, 2024-11-28:
> i also think that in the cpp files (image and animation) the flag `transparent_` isn't initialized correct, at least for RGB565.

I believe you are right, the property definition should have a default initialiser.

clydebarrow, 2024-12-03:
The PR linked above addresses this issue. Please test with it.

closed 2025-01-13 by jesserockz