esphome/issues #6499: ESPHome (2024.11.2) OTA fails with use_address

prem-anhed, 2024-11-28:
### The problem

ERROR Error resolving IP address of 192.168.103.65. Is it connected to WiFi? 
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips) 
ERROR Error resolving IP address: Int or String expected

OTA has worked with this configuration with versions prior to 2024.11.0

API works great as for example show LOGS

***********
esp32:
  board: esp32dev
  framework:
    type: arduino

api:
  port: 6053
  encryption:
    key: "1i+WrMhJwxSqOq8abbIT/2RTklkJzg5O5WtMqZHFb48="
    
ota:
  platform: esphome
  password: "cc4bf018b9f818a9f65d26d7763e5d09"
  port: 3232
    
wifi:
  ssid: XXXX
  password: YYYYYYYY
  manual_ip:
    static_ip: 192.168.2.5
    gateway: 192.168.2.1
    subnet: 255.255.255.0
  use_address: 192.168.103.65

### Which version of ESPHome has the issue?

core-2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino

api:
  port: 6053
  encryption:
    key: "1i+WrMhJwxSqOq8abbIT/2RTklkJzg5O5WtMqZHFb48="
    
ota:
  platform: esphome
  password: "cc4bf018b9f818a9f65d26d7763e5d09"
  port: 3232
    
wifi:
  ssid: XXXX
  password: YYYYYYYY
  manual_ip:
    static_ip: 192.168.2.5
    gateway: 192.168.2.1
    subnet: 255.255.255.0
  use_address: 192.168.103.65
```

### Anything in the logs that might be useful for us?

```txt
ERROR Error resolving IP address of 192.168.103.65. Is it connected to WiFi? 
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips) 
ERROR Error resolving IP address: Int or String expected
```

randybb, 2024-11-28:
a similar problem like in #6465

jesserockz, 2024-11-28:
Duplicate of https://github.com/esphome/issues/issues/6465

closed 2024-12-01 by frenck