esphome/issues #6502: Build fails for esp-idf on lolin_d32 board

steveatkoan, 2024-11-28:
### The problem

I made a bluetooth proxy mostly copy/pasting the config from here https://esphome.io/components/bluetooth_proxy.html. It built fine initially. A later update of ESPHome broke the build 

### Which version of ESPHome has the issue?

2024.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.11

### What platform are you using?

ESP32-IDF

### Board

lolin_d32

### Example YAML snippet

```yaml
esphome:
  name: esphome-btproxy

esp32:
  board: lolin_d32
  framework:
    type: esp-idf
    # see https://esphome.io/components/bluetooth_proxy.html

# Enable logging
logger:

# Enable Home Assistant API
api:
  # password: !secret api_password
  encryption: 
    key: !secret api_encryption_key

ota:
  platform: esphome
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.11.1
INFO Reading configuration /config/esphome/esphome-btproxy.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-btproxy (board: lolin_d32; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5") 
-- Component directory /data/cache/platformio/packages/framework-espidf/components/bt does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/esphome-btproxy/.pioenvs/esphome-btproxy/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:236 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 11.47 seconds =========================
```

Floriszz, 2024-12-30:
I had the same issue when building from a Windows 10 PC. While my Home Assistant is on a Raspberry PI 3b, it is not powerfull enough to build ESPhome and crashes, I'm forced to build on PC.
I took a clean Windows 10 install.
Then installed Python 3.12.8 (the newest 3.13.x does not work with ESPhome), wheel and esphome. (just as in the ESPhome manual for ' manual install ESPhome' ) https://esphome.io/guides/installing_esphome.html
Then download and install esp-idf ; https://dl.espressif.com/dl/esp-idf/ (I took the top one, which resulted in downloading the 1,9Gb anyway). All default settings.
It will install a load of tools including PIP and Setuptools(if you don't have the latest jet). End config for Command prompt.
Then installed the yaml with ESPhome on the ESP32. (using USB).
During the building process ESP-IDF installs a load of Python tools the first time automatically.

I could not use the backup WIFI Access Point mode. It just won't connect, while the ESP32 will keep rebooting.
But I could add the ESP32, through the automatic bluetooth connection in HomeAssistant and add the API key in a popup and provide the client WIFI (SSID and key) in another popup through HomeAssistant.

ssieb, 2024-12-30:
@steveatkoan re-install the addon.  Platformio sometimes gets itself messed up somehow.
Or try upgrading to the latest esphome version first because it uses newer platform versions, so it might avoid this issue as well.

steveatkoan, 2024-12-30:
Thank you
I had tried with several versions of ESPHome. Most recent seems to work because it updates esp idf 

possible cause is that I have made my home assistant config directory a git repo as a way of backing up and logging a history of changes. ESPHome is a subdirectory of this. This may be messing up the esp-idf install. See the log snippet below.

so it may continue to have problems with minor updates that do not update esp idf?

‘’’
Platform Manager: Installing git+https://github.com/pioarduino/platform-espressif32.git#51.03.07
INFO Installing git+https://github.com/pioarduino/platform-espressif32.git#51.03.07
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-sq3cue8x'...
Note: switching to 'ec17486ffa451374e108b72be07e9c8fa83338e8'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Platform Manager: espressif32@51.3.6+sha.ec17486 has been installed!
‘’’

ssieb, 2024-12-30:
That's a normal message.
esphome doesn't use the config directory for build files any more, so you should be fine backing it up that way.

steveatkoan, 2024-12-31:
Thanks for your advice. 
I tried a second install and it worked fine.
closing this issue for now as the problem appears resolved by 2024.12.2

closed 2024-12-31 by steveatkoan