esphome/issues #6503: Tuya MCU: `Initialization failed. Current init_state: 4`

felipecrs, 2024-11-29:
### The problem

Hi there. I flashed my Smart Garage Door Opener with Tuya but now I cannot make it work.

Device model: [DY-CK400A](https://community.home-assistant.io/t/garage-door-opener-dy-ck400a/398393)

Chip model: TYWE1S (ESP8266 2MB)

It uses Tuya MCU, and pretty much everything is working. Except that it doesn't properly recognize the binary sensor state on initialization. It only recognizes the state after it changes for the first time.

It would be really great if this could be fixed, otherwise I cannot rely 100% on the sensor state (in case of a power outage, for example, the sensor would restore with a wrong state).

```yaml
esp8266:
  board: esp_wroom_02

uart:
  rx_pin: GPIO13
  tx_pin: GPIO15
  baud_rate: 9600
  debug:

tuya:

binary_sensor:
  - platform: tuya
    id: contact
    sensor_datapoint: 101

switch:
  - platform: tuya
    id: relay
    switch_datapoint: 1
    restore_mode: DISABLED

cover:
  - platform: template
    device_class: garage
    name: ""
    lambda: "return id(contact).state ? COVER_OPEN : COVER_CLOSED;"
    open_action:
      then:
        switch.turn_on: relay
    close_action:
      then:
        switch.turn_off: relay
```

### Which version of ESPHome has the issue?

2024.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

TYWE1S

### Component causing the issue

tuya

### Anything in the logs that might be useful for us?

```txt
[21:50:29][C][tuya:041]: Tuya:
[21:50:29][C][tuya:044]:   Initialization failed. Current init_state: 4
[21:50:29][C][tuya:049]:   If no further output is received, confirm that this is a supported Tuya device.
[21:50:32][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[21:50:32][D][uart_debug:114]: <<< 55:AA:03:00:00:01:01:04
```

And then if I jumper the sensor once (to force a new state):

```
[21:52:22][D][tuya:335]: Datapoint 1 update to OFF
[21:52:22][D][switch:055]: 'relay': Sending state OFF
[21:52:22][D][tuya:335]: Datapoint 101 update to OFF
[21:52:22][D][binary_sensor:036]: 'contact': Sending state OFF
[21:52:22][D][uart_debug:114]: <<< 55:AA:03:07:00:05:01:01:00:01:00:11:55:AA:03:07:00:05:65:01:00:01:00:75
[21:52:23][C][tuya:041]: Tuya:
[21:52:23][C][tuya:056]:   Datapoint 1: switch (value: OFF)
[21:52:23][C][tuya:056]:   Datapoint 101: switch (value: OFF)
[21:52:23][C][tuya:074]:   Product: '{"p":"qig0zontbimfi8pn","v":"1.0.0","m":0}'
[21:52:23][D][tuya:335]: Datapoint 1 update to ON
[21:52:23][D][switch:055]: 'relay': Sending state ON
[21:52:23][D][tuya:335]: Datapoint 101 update to ON
[21:52:23][D][binary_sensor:036]: 'contact': Sending state ON
[21:52:23][D][cover:170]: 'Portão do terreiro' - Publishing:
[21:52:23][D][cover:176]:   State: OPEN
[21:52:23][D][cover:186]:   Current Operation: IDLE
[21:52:23][D][uart_debug:114]: <<< 55:AA:03:07:00:05:01:01:00:01:01:12:55:AA:03:07:00:05:65:01:00:01:01:76
```
```

### Additional information

I would deeply appreciate any help. Thank you for making all this possible.

Cossid, 2024-12-03:
There isn't really much ESPHome can do differently here, the issue is that until the TuyaMCU receives a state change, it also doesn't send any status back about its state.  ESPHome asks, but it just doesn't have anything to answer yet.  This is mostly a quirk on the design of some TuyaMCUs and not really anything ESPHome can improve upon.

You'll have to add a [Tuya Switch](https://esphome.io/components/switch/tuya.html) at DPID 1 and a [Tuya Binary Sensor](https://esphome.io/components/binary_sensor/tuya.html) at DPID 101

You may need to add the `[datapoint_hidden](https://esphome.io/components/number/tuya.html#hidden-datapoints)` field, but I'm not sure if it exists on switches or binary sensors, so you may also need to create internal [Tuya Number](https://esphome.io/components/number/tuya.html) components at the same datapoints and just not expose them, so they try to read on load.

After you have those defined, ESPHome might be able to directly ask for their state at startup.

felipecrs, 2024-12-04:
Wow. I am very grateful and happy to read your message. Thanks for taking your time to share your knowledge with me.

I'm afraid it didn't work though.

```yaml
number:
  - platform: tuya
    id: contact_number
    number_datapoint: 101
    min_value: 0
    max_value: 1
    step: 1
    datapoint_hidden:
      datapoint_type: int
      initial_value: 1
```

Which did not change anything:

```
[20:43:29][C][tuya.number:079]: Tuya Number 'contact_number'
[20:43:29][C][tuya.number:080]:   Number has datapoint ID 101
[20:43:29][C][tuya.number:082]:   Datapoint type is 2
[20:43:29][C][tuya.number:088]:   Initial Value: 1.000000
[20:43:29][C][tuya.number:091]:   Restore Value: NO
[20:43:29][C][tuya.binary_sensor:017]: Tuya Binary Sensor:
[20:43:29][C][tuya.binary_sensor:018]:   Binary Sensor has datapoint ID 101
[20:43:29][C][tuya.switch:068]: Tuya Switch 'relay'
[20:43:29][C][tuya.switch:090]:   Restore Mode: always OFF
[20:43:29][C][tuya.switch:024]:   Switch has datapoint ID 1
[20:43:29][C][tuya:041]: Tuya:
[20:43:29][C][tuya:044]:   Initialization failed. Current init_state: 4
[20:43:29][C][tuya:049]:   If no further output is received, confirm that this is a supported Tuya device.
[20:43:31][D][api:103]: Accepted 192.168.1.10
[20:43:31][W][component:237]: Component api took a long time for an operation (289 ms).
[20:43:31][W][component:238]: Components should block for at most 30 ms.
[20:43:32][D][api.connection:1446]: Home Assistant 2024.12.0b5 (192.168.1.10): Connected successfully
```

Then I mapped it to a switch, so that I could write values to it by myself to check if it forces the sensor to update:

```yaml
switch:
  - platform: tuya
    id: contact_fixer
    switch_datapoint: 101
```

But toggling this switch, unfortunately, doesn't trigger the MCU to send the sensor status back:

```
20:57:14 | [D] | [switch:016] | 'contact_fixer' Turning OFF.
20:57:14 | [D] | [tuya:623] | Setting datapoint 101 to 0
20:57:14 | [W] | [tuya:626] | Setting unknown datapoint 101
20:57:14 | [D] | [switch:055] | 'contact_fixer': Sending state OFF
20:57:16 | [E] | [tuya:464] | Initialization failed at init_state 4
```

The only thing which indeed updates the sensor is to physically make it change status once.

That's very odd, and I wonder how the original firmware handled it.

Cossid, 2024-12-04:
It's possible there is some other status request ESPHome just doesn't know about, but without the code of the MCU, there is really no way for us to know.

There is one additional hack you could try, and that is adding a Tuya Number to DPID 7 and setting it to 0, and see if it gives you anything back then.  That would be setting a countdown timer to 0.  I don't know if that would actually trigger a door state change or not though.

felipecrs, 2024-12-04:
Thank you, I've played with it.

Setting it will indeed get rid of the `init_state: 4` issue and the MCU will report the state of DP7 back.

```
[21:53:46][C][tuya.number:079]: Tuya Number 'Test Number'
[21:53:46][C][tuya.number:080]:   Number has datapoint ID 7
[21:53:46][C][tuya.number:082]:   Datapoint type is 2
[21:53:46][C][tuya.number:088]:   Initial Value: 0.000000
[21:53:46][C][tuya.number:091]:   Restore Value: NO
[21:53:46][C][tuya.binary_sensor:017]: Tuya Binary Sensor:
[21:53:46][C][tuya.binary_sensor:018]:   Binary Sensor has datapoint ID 101
[21:53:46][C][tuya.switch:068]: Tuya Switch 'Test Switch'
[21:53:46][C][tuya.switch:090]:   Restore Mode: always OFF
[21:53:46][C][tuya.switch:024]:   Switch has datapoint ID 1
[21:53:46][C][tuya:041]: Tuya:
[21:53:46][C][tuya:058]:   Datapoint 7: int value (value: 65367)
[21:53:46][C][tuya:074]:   Product: '{"p":"qig0zontbimfi8pn","v":"1.0.0","m":0}'
```

This DP refers to a countdown, and when it finishes, it will open or close the gate (i.e. toggle DP1 switch).

Setting it to `0` BTW triggers the countdown to start from `65535`, and then it reports ever time 10s has passed, with the remaining time.

Unfortunately, it doesn't help to trigger the report of DP101 state.

Cossid, 2024-12-04:
Yeah, that's disappointing.  I was hoping setting it to 0 would be how they cancel the countdown, but apparently not.  Now I'm curious how they do, because it supposedly doesn't support negative values.

That being said, you can try reading https://images.tuyacn.com/smart/aircondition/Guide-to-Interworking-with-the-Tuya-MCU.pdf and see if there is a status call ESPHome isn't already making, I'm pretty sure they do the 0x08 command to get device status and waits for the 0x07 response at boot.

I did confirm that Tuya's schema only communicates DPIDs of 1,7,101 for this device, so there aren't any we've missed.

You could still have a tuya number backing 101, and have it restore state, and make a template binary sensor that uses that value, but that is also only somewhat reliable.

felipecrs, 2024-12-04:
> Now I'm curious how they do, because it supposedly doesn't support negative values.

Oh, it's simple: there is no way to cancel the countdown. It will only stop counting once it reaches zero by itself. :)

---

> You could still have a tuya number backing 101, and have it restore state, and make a template binary sensor that uses that value, but that is also only somewhat reliable.

Yeah, I don't like that as it would have the potential wrongly marking my house as closed (while in fact it could be open).

What I have done instead is used a global variable to always make my house look like opened on initialization:

```yaml
globals:
  # workaround for the issue where the sensor is not updated on initialization
  # it's safer to assume it's open rather than closed in such case
  # https://github.com/esphome/issues/issues/6503
  - id: initial_state
    type: bool
    restore_value: no
    initial_value: 'true'

binary_sensor:
  - platform: tuya
    id: mcu_contact
    sensor_datapoint: 101
    on_state:
      if:
        condition:
          lambda: 'return id(initial_state);'
        then:
          lambda: 'id(initial_state) = false;'

  - platform: template
    id: contact
    lambda: 'return id(initial_state) || id(mcu_contact).state;'

cover:
  - platform: template
    device_class: garage
    id: main_cover
    name: ""
    lambda: "return id(contact).state ? COVER_OPEN : COVER_CLOSED;"
```

---

Again, thanks a ton for your help. I will post back here if I ever find something.

felipecrs, 2024-12-04:
I found this in the Tuya documentation:

The Query Status function, but look at this warning:

[image]

Which seems very curious. I wonder if that's why it's not working, and I also wonder if ESPHome code is not prepared to handle such case.

felipecrs, 2025-04-04:
This is still valid, but I don't have any updates.