esphome/issues #6506: error: esphome/components/lvgl/lvgl_proxy.h: No such file or directory - compilation terminated

kebibg, 2024-11-30:
### The problem

The YAML can't be compiled, before a month they are compiled.

### Which version of ESPHome has the issue?

20224.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core:2024.11.3 Suprevisor:2024.11.4 OS v.13.2 Frontend:20241106.2

### What platform are you using?

ESP32-IDF

### Board

GUITION 4” 480x480 ESP32-S3-4848S040 Smart Display

### Component causing the issue

LVGL

### Example YAML snippet

```yaml
substitutions:
  name: "touch-panel"
  friendly_name: "Touch Panel"
  device_description: "Guition ESP32-S3-4848S040 480*480 Smart Screen"
  project_name: "Guition.ESP32_S3_4848S040"
  project_version: "1.0.0"

  lightbulb: "\U000F0335"
  ceiling_light: "\U000F0769"
  lamp: "\U000F06B5"
  floor_lamp: "\U000F08DD"
  string_lights: "\U000F12BA"
  clock: "\U000F0150"
  ceiling_fan: "\U000F1797"
  light_recessed: "\U000F179B"
  blinds_horizontal: "\U000F1A2B"
  blinds_horizontal_closed: "\U000F1A2C"
  curtains_closed: "\U000F1847"
  curtains: "\U000F1846"
  wallsconce: "\U000F091C"
  bed: "\U000F02E3"
  bed_empty: "\U000F08A0"
  thermometer_high: "\U000F10C2"
  humidity: "\U000F058E" 
  wallsconce_variant: "\U000F091E"
  kiss: "\U000F0C73"
  hvac: "\U000F0D43"

esphome:
  name: esp32-s3-480x480
  friendly_name: ESP32-S3-480x480
  project:
    name: "${project_name}"
    version: "${project_version}"
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-s3-devkitc-1
[... 1065 lines cut ...]
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- lvgl @ 8.4.0
Compiling .pioenvs/esp32-s3-480x480/src/esphome/components/image/image.o
Compiling .pioenvs/esp32-s3-480x480/src/esphome/components/light/addressable_light.o
Compiling .pioenvs/esp32-s3-480x480/src/esphome/components/light/automation.o
Compiling .pioenvs/esp32-s3-480x480/src/esphome/components/light/esp_color_correction.o
In file included from src/esphome/components/image/image.cpp:1:
src/esphome/components/image/image.h:6:10: fatal error: esphome/components/lvgl/lvgl_proxy.h: No such file or directory
 #include "esphome/components/lvgl/lvgl_proxy.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-s3-480x480/src/esphome/components/image/image.o] Error 1
========================= [FAILED] Took 15.32 seconds =========================
```

ssieb, 2024-11-30:
Don't use the external component.  Everything should be merged now and that ref is out of date.

closed 2024-11-30 by ssieb

kebibg, 2024-11-30:
Ok. Thank you. I will test it.

equinoxe12, 2024-11-30:
Same problem for me. 
One month ago YAML can be compiled but now I obtain the eroor "esphome/components/lvgl/lvgl_proxy.h: No such file or directory"
I use GUITION 4” 480x480 ESP32-S3-4848S040 Smart Display with HA Core:2024.11.3 Suprevisor:2024.11.4 OS v.13.2 Frontend:20241106.2 and no external component for LVGL

ssieb, 2024-11-30:
@equinoxe12 paste your config.

equinoxe12, 2024-12-01:
oups,
You right, the problem is the external component, in this config I forget to remove it.
Now it 's ok for me.
Thanks.

kebibg, 2024-12-01:
Yes, after remove the external component and make some corrections all is compilated without errors.
@ssieb thanks again.

Regards

freeky92, 2024-12-01:
> Да, после удаления внешнего компонента и внесения некоторых исправлений все компилируется без ошибок. @ssiebеще раз спасибо.
> 
> С уважением

plz tell me how do u remove all external component ? how to import components like i80, spi, ili9xxx, io_bus

freeky92, 2024-12-01:
plz tell me how do u remove all external component ? how to import components like i80, spi, ili9xxx, io_bus

kebibg, 2024-12-01:
See my file above. Inside have an spi for LCD.abnd i2c for touchscreen. Remove external component because in new version, files from external component are included to main.